MichaIng/DietPi #4031: MedusaPy |  Could not start the web server on port 8081

aermak, 2021-01-02:
- DietPi version | G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 5.4.79-v7+ #1373
- SBC model | RPi 3 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | PyMedusa
- Was the software title installed freshly or updated/migrated? fresh
- Can this issue be replicated on a fresh installation of DietPi? probably
- Bug report ID | b09b501d-d9a2-4a5a-992f-8e1042eab75f

#### Steps to reproduce
Medusa log errors every second:
2021-01-02 11:02:14 INFO	TORNADO :: [1c1d327] Could not start the web server on port 8081. Exception: [Errno 98] Address already in use
2021-01-02 11:02:14 INFO	TORNADO :: [1c1d327] Starting Medusa on http://0.0.0.0:8081/
2021-01-02 11:02:14 INFO	MAIN :: [1c1d327] Starting Medusa [master] using '/mnt/dietpi_userdata/medusa/config.ini'
2021-01-02 11:02:13 INFO	MAIN :: [1c1d327] 12345: Unable to find the show in the database

#### Expected behavior

- ...

#### Actual behaviour

- ...

#### Extra details
root@DietPi:~# dietpi-services status   

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status 

[  OK  ] DietPi-Services | nfs-kernel-server    active (exited) since Sat 2021-01-02 10:37:29 CET; 26min ago
[  OK  ] DietPi-Services | php7.3-fpm           active (running) since Wed 2020-12-23 18:26:20 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | lighttpd             active (running) since Wed 2020-12-23 18:26:21 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | transmission-daemon  active (running) since Sat 2021-01-02 10:34:31 CET; 29min ago
[  OK  ] DietPi-Services | medusa               active (running) since Sat 2021-01-02 11:03:31 CET; 26s ago
[  OK  ] DietPi-Services | cron                 active (running) since Wed 2020-12-23 18:26:22 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | dropbear             active (running) since Wed 2020-12-23 18:26:16 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | pihole-FTL           active (running) since Wed 2020-12-23 18:26:00 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | openvpn              active (exited) since Wed 2020-12-23 18:26:00 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | dietpi-nordvpn       active (running) since Wed 2020-12-23 18:26:16 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | dietpi-ramlog        active (exited) since Wed 2020-12-23 18:25:59 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | dietpi-preboot       active (exited) since Wed 2020-12-23 18:26:00 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | dietpi-boot          active (exited) since Wed 2020-12-23 18:26:16 CET; 1 weeks 2 days ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Wed 2020-12-23 18:26:16 CET; 1 weeks 2 days ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor  inactive (dead)

Joulinar, 2021-01-02:
Hi,

Many thanks to your message. Pls can you check `LISTEN` ports

`ss -tulpn | grep LISTEN`

aermak, 2021-01-02:
tcp     LISTEN   0        128              0.0.0.0:8081           0.0.0.0:*      users:(("python3",pid=27596,fd=10))

Joulinar, 2021-01-02:
There is already a python process running using that port. Can you do

`ps -ef|grep 27596`

aermak, 2021-01-02:
root@DietPi:~# ps -ef|grep 27596
root     24119 15655  0 12:09 pts/0    00:00:00 grep 27596
medusa   27596     1  5  2020 ?        04:38:37 /usr/bin/python3 /mnt/dietpi_userdata/medusa/start.py -q --nolaunch --datadir=/mnt/dietpi_userdata/medusa

Joulinar, 2021-01-02:
Medua is already running on that port. You can kill the process or do a reboot

`medusa 27596 1 5 2020 ? 04:38:37 /usr/bin/python3 /mnt/dietpi_userdata/medusa/start.py -q --nolaunch --datadir=/mnt/dietpi_userdata/medusa`

aermak, 2021-01-02:
it's embarrassing how often that works!
thank you

closed 2021-01-02 by aermak