MichaIng/DietPi #4032: Rebooting Every time i install a software bugs me!

Games-Crack, 2021-01-02:
#### Is your feature request related to a problem? Please describe:
I am always frustrated when I install software on my diet pi with dietpi-software because I can't say no to the reboot!

#### Describe the solution you'd like:
Let the user decide if he/she wants to reboot

#### Describe alternatives you've considered:
Or a "Flag" that you can set to not reboot

Joulinar, 2021-01-02:
Hi,

you can avoid the reboot by using `dietpi-software install <ID>` but I guess @MichaIng can explain the reason for the reboot required 😉

MichaIng, 2021-01-03:
Good news is I just thought the same a few days ago: There is not really a reason for the reboot, probably required in a very few cases only. It's just that it was there as long as I can think of and no one aimed to remove it until now.

Let's remove it. In a few cases we might want to start the respective software service one time manually (for those which are not controlled by `dietpi-services`): VNC servers + XRDP, SSH servers, NoMachine, hostapd, that should be it.

Games-Crack, 2021-01-03:
@Joulinar I know that you sometimes have to reboot ;-) 
but not all the time like @MichaIng said and thank you for the command i didn't know this.

MichaIng, 2021-02-09:
Done: #4095

closed 2021-02-14 by MichaIng