MichaIng/DietPi #4033: syslog.service is not-found inactive dead

jtmoon79, 2021-01-03:
#### Required Information

- DietPi version
        G_DIETPI_VERSION_CORE=6
        G_DIETPI_VERSION_SUB=34
        G_DIETPI_VERSION_RC=3
        G_GITBRANCH='master'
        G_GITOWNER='MichaIng'
- Distro version | `10.7`
- Kernel version | `Linux host1 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux`
- SBC model | _RPi 4 Model B (armv7l)_
- Power supply used | Rasperry Pi branded wall wart
- SDcard used | 7GB Sandisk

#### Additional Information (if applicable)
- Software title | `syslog.service`
- Newly installed DietPi `DietPi_RPi-ARMv6-Buster` on 2020/12/28.
- Can this issue be replicated on a fresh installation of DietPi? Yes.
- Bug report ID | 41a77da2-4444-4b7b-9582-ddddddddddd

#### Steps to reproduce

1. install DietPi `DietPi_RPi-ARMv6-Buster`
2. run various first-install upgrade processes
3. reboot
4. review `syslog.service` using command `systemctl list-units --all`

        UNIT                     LOAD      ACTIVE   SUB     DESCRIPTION
        ● syslog.service         not-found inactive dead    syslog.service

#### Expected behaviour

`syslog.service` should be `loaded active running`.

#### Actual behaviour

`syslog.service` is`not-found inactive dead`.

MichaIng, 2021-01-03:
Many thanks for your report.

Note that `systemctl list-units --all` lists all unit names that are references in any way by other units or targets, regardless if those are present or required in any way, or even for backwards compatibility only.

On modern distributions, in very most cases, systemd is the init system and In all those cases, `systemd-journald` is the main system logging daemon, which you can check the status of via `systemctl status systemd-journald`.

Only when you enable full logging (the word "full" is actually misleading here) via `dietpi-software` or install `rsyslog` manually, the `rsyslog.service` will be installed additionally. `systemd-journald` will then send parts of its system logs to this plain text file system logging daemon, i.e. additionally to `journalctl` (by default memory logging only and stored in binary format) logs are stored in plain text into various files in `/var/log`. `syslog.service` (without the leading "r") came from an ancient system logging daemon that is not used anymore for ages in favour of rsyslog and journald, but `rsyslog.service` installs it as alias.

I also see `systemctl list-units --all` listing `syslog.service` aside of `rsyslog.service` but cannot find a reference in any installed unit. Probably it is even hardcoded in systemd to be used as syslog daemon by journald, when present. Makes sense to check for a well known name as well that can be used as alias by actual alternative syslog daemons.

jtmoon79, 2021-01-04:
I was looking for the log of the `ssh.service` (using `openssh-server` package).  It was not logging to a file in `/var/log`. I mistakenly presumed the `syslog.service` being in a `dead` state  was the culprit.

>  `systemd-journald` is the main system logging daemon, which you can check the status of via `systemctl status systemd-journald`

Run `journalctl -fe` to monitor `ssh.service` is the command I wanted.

> Only when you enable full logging (the word "full" is actually misleading here) via `dietpi-software` ... `systemd-journald` will then send parts of its system logs to this plain text file system logging daemon,

I have _Log System_ set to `DietPi-Ramlog #2`, so log messages are not duplicated to files under `/var/log`.

`DietPi-RAMlog #2: Same as #1, with the added feature of saving logfile contents to disk (/root/logfile_storage/*), before being cleared.`

<br />

Great reply. Thanks!

closed 2021-01-04 by jtmoon79

MichaIng, 2021-01-04:
If OpenSSH server is all you want to monitor, the following would focus it:
```
journalctl -feu ssh
```

Indeed we did and continue to migrate most software services to log to journald, so all logs are at a single place, but easily possible to filter. E.g. if OpenSSH was configured to log to a file, the service around it logs to journald until the binary has started, read the settings file, found the directive to log to file and successfully opened that file writable for logging. So there are cases where the log file is empty (or incomplete), so you need to check system logs to see what happened during service init and daemon start. Debugging is easier when you have everything in chronological order in one place, even including kernel or system errors, that do not come from the daemon itself but affect its function. So you are able to see things in context like:
1. kernel: low voltage warning
2. udev: drive disconnected
3. kernel: I/O error
4. service: Failed to write to database
5. service: Crash
6. syslog: Service xyz stopped with signal