MichaIng/DietPi #4034: Dietpi-explorer | Create folder option

sl002, 2021-01-03:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I have to exit out of dietpi-explorer to create a folder using mkdir. 

#### Describe the solution you'd like:

- Add a mkdir command to the dietpi-explorer menu

#### Describe alternatives you've considered:

- ...

#### Additional context

- [image]

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/+204

MichaIng, 2021-01-03:
Many thanks for your request. Makes sense. A `*- New ----` block to create a new directory or a new file (via `nano`, as we have a nano-based file edit option already).

Actually, I don't like that fact that one needs to select a directory to "paste" copied/moved content into that folder. Similarly I don't like that one needs to select a directory to create a new dir/file inside of it. Wouldn't it be somehow more intuitive if those options would be already below of above (open for suggestions) the directory content list?

sl002, 2021-01-04:
I think that leaving the existing menu in place is still best practise, as if a folder has 1000+ files / folders it can take a pi a fair while to load if you have to go into it. Perhaps a "New" Menu under the I/O operations with the options for 

- New Directory / Folder
- New File (Nano)

You could also have the same options when you are in the directory? i.e....

- Previous Directory
- Root Directory
- New Directory
- New File (Nano)
- {File List}

Or somehow implement a keyboard shortcut to get to a "New" menu?

MichaIng, 2021-01-04:
So basically "New Directory" + "New File" would be available from both, above (I assume best?) the file list and from the directory "context" menu. When shown above the file list, the initially selected entry should probably be previous or root directory so that the amount of arrow key presses to navigate through the directory content stays the same.

A keyboard shortcut is sadly not possible easily since we use whiptail menus which do have hotkey support.