MichaIng/DietPi #4035: Squeezelite (or Alsa) patches

whitepj, 2021-01-04:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I run a headless media player, installed in the loft, delivering audio to the top floor of the house. It runs 24/7. I have tried to minimise power consumption as much as possible by investing in a decent switching PSU, but the amp still draws more power than is necessary. This is a problem, because there are fixes available.

#### Describe the solution you'd like:
DietPi is designed to run on low power SBCs with GPIOs available. These outputs can be utilised to control external power requirements. Fundamentally,when the audio is active, then the GPIO state needs to reflect this -- which can be used to switch on (or off) an external power amplifier.

#### Describe alternatives you've considered:
There appear to be two possible approaches: <br />
1. Patch squeezelite to toggle GPIO output (using libwiringPi.so) 
2. Patch alsa

Option (1) has already been done. I am currently experimenting with the squeezelite binary lifted straight from a Volumio installation. So far, audio works (mostly, though I suspect I have a few codecs missing). This is the quick solution.
Option (2) I investigated a couple of years ago,and thought that this had been (mostly?) done. However, this was mostly above my level of expertise. This would be a more preferable option, since it would work for <i>any</i> audio output.

#### Additional context

n/a. But thank you for something that actually delivers. I've spent nearly a week battling with volumio, before giving up. Had DietPi set up and running in one evening.

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/+205

whitepj, 2021-01-19:
UPDATE:

It appears that this functionality is already done -- at least in part (meaning that it already works in a Raspberry when using an IQAudio DAC, but I have no idea whether it will work with any other SBC).
There is some info within the ALSA documentation (which I can't find at the moment).

Refs:
https://github.com/raspberrypi/linux/pull/1493
https://github.com/raspberrypi/linux/commit/849c262ffe9651f412a50f547ad76831e9b2e060

MichaIng, 2021-01-19:
On other SBCs, sound card specific dtoverlays often need to be build manually. Squeezelite has two build flags to enable RPi-specific GPIO support and generic GPIO support which we do not set currently (as they are new and until now nobody requested it 😉): `-DGPIO -DRPI`

Report back if using the dtoverlay on RPi works for your case, else I can rebuild our Squeezelite packages with those flags to see if that works.

MichaIng, 2021-06-02:
@whitepj 
Coming back to this topic. In your care the ALSA driver seems to wake up the DAC automatically, if I understand it correctly? And how does it set to sleep?

The ALSA documentation is btw here: https://www.alsa-project.org/alsa-doc/alsa-lib/index.html

whitepj, 2021-06-03:
Not sure about this, but I assume it is a combination of the ALSA driver
and the Raspberry driver for the IQAudio sound card.
Logfile entries:
Jun  2 07:00:01 Monsters kernel: [2049956.359810] pcm512x 1-004c: No SCLK,
using BCLK: -2
Jun  2 07:00:01 Monsters kernel: [2049956.362271] snd-rpi-iqaudio-dac
soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
Jun  2 07:20:11 Monsters kernel: [2051166.005795] snd-rpi-iqaudio-dac
soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22

Of some interest: rights to IQAudio boards have now been acquired by
Raspberry Pi Foundation, and all boards except the PiDac Zero (sadly) are
being produced -- at a slightly cheaper price than before.
Never worked out what the first line is telling me, about SCLK. Occurs
frequently, but no audible effect.

MichaIng, 2021-06-03:
I just recognised that IQaudio is RPi Foundation now when checking old requests. There seem to have been a "cosmic controller", but that is not available anymore: https://github.com/MichaIng/DietPi/issues/1463