MichaIng/DietPi #4036: wusb300n support

symbios24, 2021-01-04:
Hello i have a Linksys wusb300n wireless usb adapter that is not working with dietpi
Is it possible that you add the driver for it?
i wanted   to use it as a hotspot 
I think it has a  marvel chipset 88W8362 / 88W8060 
Marvell 88W8360 TopDOG
Thank you

MichaIng, 2021-01-04:
Many thanks for your request.

Does the following work?
```
apt update
apt install firmware-libertas
```

symbios24, 2021-01-05:
No it does not work

MichaIng, 2021-01-05:
Okay, I'm afraid there is no chance to add support for this on a regular basis. I wasn't able to reliably get the information about which chip is inside. The ones you named indeed are not in the list of this generic Marvell 88W8xxx firmware package. Linksys (manufacturer) marked it EOL and does not provide any downloads. Searching through the web, I found some (10-13 years old) instructions to use a Windows XP driver and load it via [this](https://sourceforge.net/projects/ndiswrapper/) tool, which wraps it somehow into a kernel module:
- https://ubuntuforums.org/showthread.php?t=1653471
- https://ubuntuforums.org/showthread.php?t=873027

The problem is now that of course all Windows XP driver download links are broken, aside of such from websites I would never touch any link on. And I doubt that this method, which worked on and before Ubuntu 10 (Linux 2.6.X) still works on current Linux 5.X.

It's sad when functional hardware cannot be used, but I think the attempt to getting this ancient piece running on a modern OS will only cost a lot of time and frustration before it ultimately fails.

closed 2021-06-02 by MichaIng

MichaIng, 2024-09-26:
> Is it there any way I can try the driver that worked on kernel 2.6x ?? Maybe it works ok you never know.

Zero chance. Linux drivers are part of the kernel, and the kernel API changes, hence drivers need to be updated with every major version at least. There are rare exceptions where drivers may work on two major versions, but ruled out that a Linux 2.x driver will work on Linux 6.x.

actraiser30, 2024-09-27:
If the want to try my self in an old dietpi/buster or even older 
How can i install the driver?

MichaIng, 2024-10-03:
```sh
apt install ndiswrapper
```
And then follow the steps from the Ubuntu forum links in my comment above.