MichaIng/DietPi #4037: Error installing chromium

andrewstech, 2021-01-04:
#### Details:
- Date           | Mon Jan  4 22:32:24 GMT 2021
- Bug report     | eab45ba3-0745-449b-ae45-7435fa6cbc57
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 400 (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq --allow-change-held-packages install chromium-browser
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Open dietpi-software from command line
2. Search for chromium 
3. Install chromium 
#### Expected behaviour:

- ... Chromium should Install
#### Actual behaviour:

- ...  gives me an error about packages
#### Extra details:

- ... not sure this is the only issue ive had with the installer today.
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

ravenclaw900, 2021-01-04:
Hello,
What happens if you try to install Chromium manually? `apt install chromium`

Joulinar, 2021-01-04:
`chromium-browser` is not yet available on Raspberry OS 64. But `chromium` should work as alternative

MichaIng, 2021-01-04:
Many thanks for your report.

This is a known limitation of the 64-bit (beta) Raspberry Pi image. The Chromium package with GPU acceleration for RPi has simply not yet been built for arm64: http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/

Installing the `chromium` package from the Debian repository works, but it won't run great, as not GPU acceleration is included. Firefox is usually the better performing choice in such case.

andrewstech, 2021-01-04:
Thanks It worked by doing it manually :)

closed 2021-01-04 by andrewstech

andrewstech, 2021-01-05:
@MichaIng I would say no very well so far. It runs horizontally at an angle so not very useable.
[image]

andrewstech, 2021-01-05:
If it helps here is the logs.

[1281:1281:0105/134331.699742:ERROR:gl_surface_egl.cc(725)] No suitable EGL configs found.
[1281:1281:0105/134331.700229:ERROR:gl_surface_egl.cc(773)] EGL Driver message (Error) eglGetConfigAttrib: Bad config.
[1281:1281:0105/134331.700529:ERROR:gl_context_egl.cc(95)] eglGetConfigAttrib failed with error EGL_BAD_CONFIG
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[1281:1281:0105/134331.705368:ERROR:vaapi_wrapper.cc(541)] vaInitialize failed: unknown libva error
[1281:1281:0105/134331.706021:ERROR:gl_surface_egl.cc(725)] No suitable EGL configs found.
[1281:1281:0105/134331.708455:ERROR:gl_surface_egl.cc(773)] EGL Driver message (Error) eglCreatePbufferSurface: Bad config.
[1281:1281:0105/134331.709216:ERROR:gl_surface_egl.cc(2076)] eglCreatePbufferSurface failed with error EGL_BAD_CONFIG
[1281:1281:0105/134331.709766:ERROR:gpu_info_collector.cc(65)] gl::GLContext::CreateOffscreenGLSurface failed
[1281:1281:0105/134331.710643:ERROR:gpu_info_collector.cc(273)] Could not create surface for info collection.
[1281:1281:0105/134331.714433:ERROR:gpu_init.cc(77)] CollectGraphicsInfo failed.
[1281:1281:0105/134331.735431:ERROR:viz_main_impl.cc(150)] Exiting GPU process due to errors during initialization
root@LAB-BENCH:~#

MichaIng, 2021-01-05:
> No suitable EGL configs found.

That makes sense. Could you try the following:
```
echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --test-type --no-sandbox --temp-profile --user-data-dir --disable-smooth-scrolling --disable-low-res-tiling --enable-low-end-device-mode --num-raster-threads=4 --profiler-timing=0 --disable-composited-antialiasing --use-gl=none"' > /etc/chromium.d/custom_flags
cp /etc/chromium.d/custom_flags /root/.chromium-browser.init # Should not be required, however to be sure 😉
```

andrewstech, 2021-01-05:
Well some great news it works very well.  As for video playback i watched around 10 minutes of YouTube and the payback was perfect at 720p.
Websites load and are very responsive. 

Let me know if there is anything else you wish for me to test.