MichaIng/DietPi #4038: Hostapd not working with RPI 4 8GB running dietpi 64 bit

famewolf, 2021-01-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6 G_DIETPI_VERSION_SUB=34 G_DIETPI_VERSION_RC=3 G_GITBRANCH='master' G_GITOWNER='MichaIng'
- 
- Distro version | debian Buster
- Kernel version | Linux DietPi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux

- SBC model | RPi 4 Model B (aarch64)

- Power supply used | Generic Raspberry Pi 4 power supply from Amazon
- SDcard used | Generic

#### Additional Information (if applicable)
- Software title | Hostapd
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Install the hostapd software
2. After reboot wifi is not enabled in dietpi-config.  Attempting to enable gives error.

#### Expected behaviour
I should be able to connect my phone to the hotspot.
- ...

#### Actual behaviour
On initial install I received an error indicating the dhcp server did not start.
isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
   Active: failed (Result: exit-code) since Mon 2021-01-04 18:46:02 EST; 29ms ago
     Docs: man:systemd-sysv-generator(8)
  Process: 3527 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)

Jan 04 18:46:00 DietPi dhcpd[3539]: bugs on either our web page at www.isc.org or in the README file
Jan 04 18:46:00 DietPi dhcpd[3539]: before submitting a bug.  These pages explain the proper
Jan 04 18:46:00 DietPi dhcpd[3539]: process and the information we find helpful for debugging.
Jan 04 18:46:00 DietPi dhcpd[3539]: 
Jan 04 18:46:00 DietPi dhcpd[3539]: exiting.
Jan 04 18:46:02 DietPi isc-dhcp-server[3527]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ... failed!
Jan 04 18:46:02 DietPi isc-dhcp-server[3527]:  failed!
Jan 04 18:46:02 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Jan 04 18:46:02 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Jan 04 18:46:02 DietPi systemd[1]: Failed to start LSB: DHCP server.
- ...

#### Extra details
After reboot the hotspot is not in the list and checking its status via systemctl gives:

hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2021-01-04 18:53:49 EST; 1s ago
  Process: 2839 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)

- ...

ravenclaw900, 2021-01-05:
Could you post a full log with `journalctl -u isc-dhcp-server`?

famewolf, 2021-01-05:
Certainly!
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : hostapd
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : isc-dhcp-server
[ SUB1 ] DietPi-Set_hardware > wifimodules (enable)
[ INFO ] DietPi-Set_hardware | Please wait, enabling WiFi modules...
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: iw wireless-tools crda wpasupplicant
[  OK  ] DietPi-Set_hardware | All required APT packages are already installed.
[  OK  ] wifimodules enable | Completed
[  OK  ] DietPi-Config | Dropping connections, please wait...
[ INFO ] DietPi-Config | Restarting network, please wait...
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/dc:a6:32:dc:8e:d9
Sending on   LPF/eth0/dc:a6:32:dc:8e:d9
Sending on   Socket/fallback
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6
DHCPOFFER of 192.168.3.5 from 192.168.3.1
DHCPREQUEST for 192.168.3.5 on eth0 to 255.255.255.255 port 67
DHCPACK of 192.168.3.5 from 192.168.3.1
bound to 192.168.3.5 -- renewal in 42043 seconds.
[ SUB1 ] DietPi-Services > start 
[  OK  ] DietPi-Services | start : isc-dhcp-server
[  OK  ] DietPi-Services | start : nfs-kernel-server
[  OK  ] DietPi-Services | start : mariadb
**[FAILED] DietPi-Services | start : hostapd**
[  OK  ] DietPi-Services | start : cron
[ INFO ] DietPi-Config | Reloading networking data, please wait...
[  OK  ] DietPi-Config | Network restarted

-- Logs begin at Thu 2019-02-14 05:11:58 EST, end at Mon 2021-01-04 22:26:05 EST. --
Jan 04 18:49:10 DietPi systemd[1]: Starting LSB: DHCP server...
Jan 04 18:49:13 DietPi isc-dhcp-server[797]: Launching IPv4 server only.
Jan 04 18:49:13 DietPi dhcpd[806]: Internet Systems Consortium DHCP Server 4.4.1
Jan 04 18:49:13 DietPi dhcpd[806]: Copyright 2004-2018 Internet Systems Consortium.
Jan 04 18:49:13 DietPi dhcpd[806]: All rights reserved.
Jan 04 18:49:13 DietPi dhcpd[806]: For info, please visit https://www.isc.org/software/dhcp/
Jan 04 18:49:13 DietPi dhcpd[810]: Internet Systems Consortium DHCP Server 4.4.1
Jan 04 18:49:13 DietPi dhcpd[810]: Copyright 2004-2018 Internet Systems Consortium.
Jan 04 18:49:13 DietPi dhcpd[810]: All rights reserved.
Jan 04 18:49:13 DietPi dhcpd[810]: For info, please visit https://www.isc.org/software/dhcp/
Jan 04 18:49:13 DietPi dhcpd[810]: Wrote 0 leases to leases file.
Jan 04 18:49:14 DietPi dhcpd[810]: Server starting service.
Jan 04 18:49:16 DietPi isc-dhcp-server[797]: Starting ISC DHCPv4 server: dhcpd.
Jan 04 18:49:16 DietPi systemd[1]: Started LSB: DHCP server.
Jan 04 18:52:46 DietPi systemd[1]: Stopping LSB: DHCP server...
Jan 04 18:52:46 DietPi isc-dhcp-server[2236]: Stopping ISC DHCPv4 server: dhcpd.
Jan 04 18:52:46 DietPi systemd[1]: isc-dhcp-server.service: Succeeded.
Jan 04 18:52:46 DietPi systemd[1]: Stopped LSB: DHCP server.
Jan 04 18:53:02 DietPi systemd[1]: Starting LSB: DHCP server...
Jan 04 18:53:03 DietPi isc-dhcp-server[2427]: Launching IPv4 server only.
Jan 04 18:53:03 DietPi dhcpd[2436]: Internet Systems Consortium DHCP Server 4.4.1
Jan 04 18:53:03 DietPi dhcpd[2436]: Copyright 2004-2018 Internet Systems Consortium.
Jan 04 18:53:03 DietPi dhcpd[2436]: All rights reserved.
Jan 04 18:53:03 DietPi dhcpd[2436]: For info, please visit https://www.isc.org/software/dhcp/
Jan 04 18:53:03 DietPi dhcpd[2439]: Internet Systems Consortium DHCP Server 4.4.1
Jan 04 18:53:03 DietPi dhcpd[2439]: Copyright 2004-2018 Internet Systems Consortium.
Jan 04 18:53:03 DietPi dhcpd[2439]: All rights reserved.
Jan 04 18:53:03 DietPi dhcpd[2439]: For info, please visit https://www.isc.org/software/dhcp/
Jan 04 18:53:03 DietPi dhcpd[2439]: Wrote 0 leases to leases file.
Jan 04 18:53:03 DietPi dhcpd[2439]: Server starting service.
Jan 04 18:53:05 DietPi isc-dhcp-server[2427]: Starting ISC DHCPv4 server: dhcpd.
Jan 04 18:53:05 DietPi systemd[1]: Started LSB: DHCP server.
Jan 04 22:24:09 DietPi systemd[1]: Stopping LSB: DHCP server...
[... 14 lines cut ...]
Jan 04 22:24:21 DietPi dhcpd[17132]: Server starting service.
Jan 04 22:24:23 DietPi isc-dhcp-server[17120]: Starting ISC DHCPv4 server: dhcpd.
Jan 04 22:24:23 DietPi systemd[1]: Started LSB: DHCP server.

and I'll also do the same for hostapd:

Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 124.
Jan 04 22:29:11 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Start request repeated too quickly.
Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 04 22:29:11 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
root@DietPi:~/.pm2/logs# systemctl status hostapd > /tmp/work
root@DietPi:~/.pm2/logs# cat /tmp/work
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-01-04 22:29:11 EST; 3min 23s ago
[... cut ...]
Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 124.
Jan 04 22:29:11 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Start request repeated too quickly.
Jan 04 22:29:11 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 04 22:29:11 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.

famewolf, 2021-01-05:
I even tried a another uninstall then install to get full display:

Setting up hostapd (2:2.7+git20190128+0c1e29f-6+deb10u2) ...                                                                                                             
Created symlink /etc/systemd/system/multi-user.target.wants/hostapd.service → /lib/systemd/system/hostapd.service.                                                       
**Job for hostapd.service failed because the control process exited with error code.**                                                                                       
See "systemctl status hostapd.service" and "journalctl -xe" for details.                                                                                                 
Created symlink /etc/systemd/system/hostapd.service → /dev/null.                                                                                                         
Setting up libisccfg-export163 (1:9.11.5.P4+dfsg-5.1+deb10u2) ...                                                                                                        
Setting up libirs-export161 (1:9.11.5.P4+dfsg-5.1+deb10u2) ...                                                                                                           
Setting up isc-dhcp-server (4.4.1-2) ...                                                                                                                                 
Generating /etc/default/isc-dhcp-server...                                                                                                                               
**Job for isc-dhcp-server.service failed because the control process exited with error code.**                                                                               
See "systemctl status isc-dhcp-server.service" and "journalctl -xe" for details.                                                                                         
invoke-rc.d: initscript isc-dhcp-server, action "start" failed.                                                                                                          
● isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
   Active: **failed (Result: exit-code)** since Mon 2021-01-04 22:43:04 EST; 29ms ago
     Docs: man:systemd-sysv-generator(8)
  **Process: 21972 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)**

Jan 04 22:43:02 DietPi dhcpd[21984]: bugs on either our web page at www.isc.org or in the README file
Jan 04 22:43:02 DietPi dhcpd[21984]: before submitting a bug.  These pages explain the proper
Jan 04 22:43:02 DietPi dhcpd[21984]: process and the information we find helpful for debugging.
Jan 04 22:43:02 DietPi dhcpd[21984]: 
Jan 04 22:43:02 DietPi dhcpd[21984]: exiting.
**Jan 04 22:43:04 DietPi isc-dhcp-server[21972]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ... failed!
Jan 04 22:43:04 DietPi isc-dhcp-server[21972]:  failed!**
**Jan 04 22:43:04 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Jan 04 22:43:04 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.**
**Jan 04 22:43:04 DietPi systemd[1]: Failed to start LSB: DHCP server.**
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10) ...
Processing triggers for systemd (241-7~deb10u5) ...
[  OK  ] DietPi-Software | APT install for: hostapd isc-dhcp-server iptables libnl-3-200
[ SUB1 ] DietPi-Set_hardware > wifimodules (enable)
[ INFO ] DietPi-Set_hardware | Please wait, enabling WiFi modules...
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: iw wireless-tools crda wpasupplicant
[  OK  ] DietPi-Set_hardware | All required APT packages are already installed.
[  OK  ] wifimodules enable | Completed
[  OK  ] DietPi-Software | systemctl unmask hostapd
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : hostapd
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : isc-dhcp-server

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimise and configure software

[ INFO ] DietPi-Software | Applying DietPi optimisations and configurations for RPi 4 Model B (aarch64), please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring WiFi Hotspot: turn your device into a wifi hotspot

[  OK  ] DietPi-Software | /etc/dhcp/dhcpd.conf: backup to /etc/dhcp/dhcpd.conf.bak_040121_2
[ INFO ] DietPi-Software | For a full list of backup items, please see /var/tmp/dietpi/logs/G_BACKUP_FP.db
[  OK  ] DietPi-Software | Comment in /etc/network/interfaces converted to setting: allow-hotplug wlan0
net.ipv4.ip_forward = 1
net.ipv6.conf.all.forwarding = 1
net.ipv6.conf.default.forwarding = 1
[  OK  ] DietPi-Software | Added setting ieee80211n=1 to end of file /etc/hostapd/hostapd.conf
Removed /etc/systemd/system/multi-user.target.wants/dietpi-autostart_custom.service.
[  OK  ] DietPi-Autostart | systemctl enable dietpi-autostart_custom
[  OK  ] DietPi-Autostart | systemctl daemon-reload
[ SUB1 ] DietPi-Services > dietpi_controlled 
[  OK  ] DietPi-Services | dietpi_controlled : isc-dhcp-server
[  OK  ] DietPi-Services | dietpi_controlled : nfs-kernel-server
[  OK  ] DietPi-Services | dietpi_controlled : mariadb
[... cut ...]
[  OK  ] DietPi-Services | dietpi_controlled : cron
[ SUB1 ] DietPi-Set_hardware > headless (disable)
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_HEADLESS=0
[  OK  ] headless disable | Completed
[  OK  ] DietPi-Survey | Purging survey data

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Software | System rebooting in 3 seconds to finalise DietPi-Software installations. Please wait...

famewolf, 2021-01-05:
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2021-01-04 22:49:02 EST; 420ms ago
  Process: 1816 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)
root@DietPi:~# dietpi-config
[FAILED] DietPi-Config | systemctl start hostapd
[FAILED] DietPi-Config | systemctl start hostapd
[FAILED] DietPi-Config | systemctl start hostapd

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Mon Jan  4 22:49:58 EST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```
---------------------------------------------------------------------
[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.

famewolf, 2021-01-05:
I also had it submit a bug report via the automated method in dietpi-config when trying to enable the wifi.

Joulinar, 2021-01-05:
can you try to start `hostapd` manually pls

`/usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf`

famewolf, 2021-01-05:
I can however the issue is it keeps putting the wifi into disabled mode and when I go into dietpi-config to enable it that's when I get the error and it goes back to disabled.

oot@DietPi:~/.pm2/logs# /usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
Could not set channel for kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0

Joulinar, 2021-01-05:
are you sure your wifi adapter `wlan0` is working correctly?

famewolf, 2021-01-05:
wlan0 works fine when I use it to connect to my network.   In fact I only started using eth0 yesterday when I tried to install hostapd since I want some security camera's that write to a nfs share on the pi to connect directly TO the pi so the router rebooting won't cause gaps in recording.

Joulinar, 2021-01-06:
can you post `ip a`

MichaIng, 2021-01-06:
Please also paste your bug report ID, so I can have a look into it: `echo $G_HW_UUID`

famewolf, 2021-01-06:
> echo $G_HW_UUID

a5776cb9-60a0-45e2-a9b8-dc085a3707a6

famewolf, 2021-01-06:
> Please also paste your bug report ID, so I can have a look into it: `echo $G_HW_UUID`

Please clarify.  Do you want me to uninstall hostapd and get wlan0 an ip address or do you just want to see that it exists?

Joulinar, 2021-01-06:
> Please clarify. Do you want me to uninstall hostapd and get wlan0 an ip address or do you just want to see that it exists?

I guess you are referring to my request. Simple post `ip a`

famewolf, 2021-01-06:
> > Please clarify. Do you want me to uninstall hostapd and get wlan0 an ip address or do you just want to see that it exists?
> 
> I guess you are referring to my request. Simple post `ip a`

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:dc:8e:d9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.5/24 brd 192.168.3.255 scope global dynamic eth0
       valid_lft 83486sec preferred_lft 83486sec
    inet6 fe80::dea6:32ff:fedc:8ed9/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:dc:8e:da brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.1/24 brd 192.168.42.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::dea6:32ff:fedc:8eda/64 scope link 
       valid_lft forever preferred_lft forever

Some of my attempt to sign onto it.

● isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
   Active: active (running) since Tue 2021-01-05 16:04:14 EST; 19h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 3839 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=0/SUCCESS)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/isc-dhcp-server.service
           └─3851 /usr/sbin/dhcpd -4 -q -cf /etc/dhcp/dhcpd.conf wlan0

Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPACK on 192.168.42.10 to 28:e3:47:c4:14:6d (hp810-135qe) via wlan0
Jan 06 11:41:34 DietPi dhcpd[3851]: reuse_lease: lease age 44 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.10

Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPREQUEST for 192.168.42.10 (192.168.42.1) from 28:e3:47:c4:14:6d (hp810-135qe) via wlan0
Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPACK on 192.168.42.10 to 28:e3:47:c4:14:6d (hp810-135qe) via wlan0
Jan 06 11:41:34 DietPi dhcpd[3851]: reuse_lease: lease age 44 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.10

Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPREQUEST for 192.168.42.10 (192.168.42.1) from 28:e3:47:c4:14:6d (hp810-135qe) via wlan0
Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPACK on 192.168.42.10 to 28:e3:47:c4:14:6d (hp810-135qe) via wlan0
Jan 06 11:41:34 DietPi dhcpd[3851]: reuse_lease: lease age 44 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.10

Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPREQUEST for 192.168.42.10 (192.168.42.1) from 28:e3:47:c4:14:6d (hp810-135qe) via wlan0
Jan 06 11:41:34 DietPi dhcpd[3851]: DHCPACK on 192.168.42.10 to 28:e3:47:c4:14:6d (hp810-135qe) via wlan0

famewolf, 2021-01-06:
I just successfully signed onto the ap......so whatever was wrong is solved.  I'd still like to know WHAT it was but sorry for wasting yall's time.

Joulinar, 2021-01-06:
Yes wlan0 is up correctly, DHCP service is providing IP addresses. Pls can you share status of `hostapd.service`

famewolf, 2021-01-06:
> Yes wlan0 is up correctly, DHCP service is providing IP addresses. Pls can you share status of `hostapd.service`

 hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-01-05 16:04:17 EST; 19h ago
  Process: 4036 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=0/SUCCESS)
 Main PID: 4056 (hostapd)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/hostapd.service
           └─4056 /usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf

Jan 06 11:42:33 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated
Jan 06 11:42:33 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated
Jan 06 11:47:06 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: associated
Jan 06 11:47:06 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d RADIUS: starting accounting session 714750A6EE0819BC
Jan 06 11:47:06 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d WPA: pairwise key handshake completed (RSN)
Jan 06 11:50:25 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated
Jan 06 11:50:25 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated
Jan 06 11:50:26 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated
Jan 06 11:50:26 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated
Jan 06 11:50:27 DietPi hostapd[4056]: wlan0: STA 28:e3:47:c4:14:6d IEEE 802.11: disassociated

Joulinar, 2021-01-06:
Ok seems to be working as well. Did you test a reboot on your DietPi device

famewolf, 2021-01-06:
> Ok seems to be working as well. Did you test a reboot on your DietPi device

No but I'll have to wait till later to do so as the pi provides the nfs share the security cams use and every time I reboot the pi I have to reboot all of them as well. (modded software in wyzecam thinks nfs share is sdcard..currently doesn't handle auto remounting well).

famewolf, 2021-01-06:
Is there not a webui for hostapd?  The webpage here: https://github.com/billz/raspap-webgui   shows one but I'm not getting one going to the ip address of the wlan0 interface (192.168.42.1).  I can't find any documentation in googling indicating an alternate port #.

Joulinar, 2021-01-06:
raspAP is something else. DietPi install plain `hostapd` without GUI. Adjustments on the HotSpot can be done using `dietpi-config`

https://dietpi.com/docs/software/advanced_networking/#wifi-hotspot

MichaIng, 2021-01-06:
It's build around `hostapd` as well, but a large set of other features is installed around it (aside of the web UI), even a Pi-hole-like dnsmasq-based ad blocking feature 😄.

famewolf, 2021-01-06:
Thanks for the info.  Perhaps now that I have the basics working I'll consider trying it.   I would like to be able to see that all the camera's are connected etc.

MichaIng, 2021-01-06:
Be aware that the RaspAP installer might break your network. Not sure how compatible or intrusive it is with existing network setups and the multiple possible tools used to configure it.

closed 2021-02-15 by MichaIng