MichaIng/DietPi #4040: dietpi dosent want to update/upgrade after electical shutdown

ROTOMEGA, 2021-01-05:
#### Details:
- Date           | Tue Jan  5 08:56:20 GMT 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
1. make dietpi install software
2. unplug while installing software
3.re-install software
#### Expected behaviour:
update just fine
#### Actual behaviour:
dosent want to update
#### Extra details:

- ...
#### Additional logs:
```
Ign:1 https://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:3 https://archive.raspberrypi.org/debian buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 176.126.240.84 443]
Reading package lists...
W: https://archive.raspberrypi.org/debian/dists/buster/InRelease: No system certificates available. Try installing ca-certificates.
W: https://archive.raspberrypi.org/debian/dists/buster/Release: No system certificates available. Try installing ca-certificates.
E: The repository 'https://archive.raspberrypi.org/debian buster Release' no longer has a Release file.
```

Joulinar, 2021-01-05:
Hi,

many thanks for your message. Looks like you are missing `ca-certificates`. We could try to reinstall but not sure if it is working due to certificate errors.

`apt install --reinstall ca-certificates`

MichaIng, 2021-01-05:
That reinstall attempt will most likely fail for the same reason. I guess the package is actually installed, probably the combined `ca-certificates.crt` got corrupted. If that executable is (still) present:
```
update-ca-certificates
```
or
```
update-ca-certificates -f
```

How does that directory look like?
```
ls -al /etc/ssl/certs
```

closed 2021-02-14 by MichaIng