MichaIng/DietPi #4041: FAILED Diet-Pi Update | APT upgrade - command : apt-get -qq upgrade

MeshuggahMischell, 2021-01-05:
#### Details:
- Date           | Tue Jan  5 12:29:17 GMT 2021
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```
root@DietPi:~# dpkg --configure -a
dpkg: error: unable to access the dpkg database directory /var/lib/dpkg: Read-only file system

MeshuggahMischell, 2021-01-05:
just rebooted the Pi isn't responding to any Ping SSH requests whatever ...

Joulinar, 2021-01-05:
Hi,

many thanks for your message. Probably your SD card died. Can you connect a screen to see what happen?

MeshuggahMischell, 2021-01-05:
plugged the Pi to my monitor now, but I didn't got display, have to check if the Pi maybe died.
I unplugged the SD Card and tried it on my windows 10 computer and I was able to access and see the files...
Strange...

Joulinar, 2021-01-05:
you have a spare SD card? and try a fresh install?

MeshuggahMischell, 2021-01-06:
Good morning, I just took a new SD Card installed everything from scratch but run into the same problem again.
As soon as I start installing the upgrade the installation fails and the Pi isn't resonding after a reboot.

[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-Update | APT upgrade, please wait...
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed Jan  6 06:22:18 GMT 2021
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux DietPi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MeshuggahMischell, 2021-01-06:
I am not able to reinstall it, it starts the Upgrade automatically and runs into this problem over and over again....

MeshuggahMischell, 2021-01-06:
─────────────────────────────────────────────────────
 DietPi v6.31.2 : Update available
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.178.2 (eth0)
[  OK  ] DietPi-Login | Checking network connectivity
[  OK  ] DietPi-Login | Checking DNS resolver
[  OK  ] Network time sync | Completed
[FAILED] DietPi-Update | RootFS is currently Read Only (R/O) mounted. Aborting...
[ INFO ] DietPi-Update | DietPi requires RootFS to be Read/Write (R/W) mounted. Please run "dietpi-drive_manager" to re-enable.

When i run "dietpi-drive_manager" I just get the message " RootFS is currently set to "Read Only (R/O)". DietPi-Drive_Manager requires "Read Write (R/W)" access to function."

How can I set it to R/W ?

Joulinar, 2021-01-06:
Hi,
That's basically main issue on your side to have RootFS r/o. Usually this should not be the case. You can try following to mount it r/w
`mount -o remount,rw /`

MeshuggahMischell, 2021-01-06:
You won't belive it, I reinstalled everything 3 times, ran into the same problem 3 times.
Now where you've told me the command to mount the root drive correctly I installed everything the fouth time and the problem doesn't come up again.
But thank you for the command I will keep it in mind if I run into problems during the next Upgrade again.

Joulinar, 2021-01-06:
running into this problem usually indicates problems (like corruption) on the SD card.  But good that it is working now. I will close the issue. Feel free to reopen if needed.

closed 2021-01-06 by Joulinar

MichaIng, 2021-01-06:
> running into this problem usually indicates problems (like corruption) on the SD card.

Seconded, have a regular eye on `dmesg -l emerg,alert,crit,err` for SD card I/O errors. Possible that there are bad blocks (permanently un-writable parts on the SD card), that is attempted to be written to by chance during APT upgrades.