MichaIng/DietPi #4043: mariadb.service fails to start if dietpi_userdata has been moved

zorlaski, 2021-01-08:
#### Details:
- Date           | Thu Jan  7 23:17:22 PST 2021
- Bug report     | 79861b09-14bc-42dc-9917-fc36d4be39fb
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | systemctl restart mariadb
- Exit code      | 1
- Software title | DietPi-Software

I am using 2 external 500GB HDD's connected in RAID 1 by USB cable. 

#### Steps to reproduce:
1.  Clean install, RAID configured with mdadm, MATE DE installed with dietpi-software
2.  mounted the external RAID drive (named "md0") to /home/dietpi/raid1
3.  moved dietpi_userdata to /home/dietpi/raid1
#### Expected behaviour:
-  mariadb.service initializes fine and installation of software (nextcloud) continues
#### Actual behaviour:

- mariadb fails to start, forcing a halt to the installation
#### Extra details:
`dietpi@DietPi:~/raid1$ ls -al /mnt/dietpi_userdata/mysql 
total 110632
drwxr-xr-x 4 mysql  mysql      4096 Jan  7 22:16 .
drwxrwxr-x 7 dietpi dietpi     4096 Jan  7 22:13 ..
-rw-rw---- 1 mysql  mysql     16384 Jan  7 22:16 aria_log.00000001
-rw-rw---- 1 mysql  mysql        52 Jan  7 22:16 aria_log_control
-rw-r--r-- 1 root   root          0 Jan  7 22:16 debian-10.3.flag
-rw-rw---- 1 mysql  mysql       972 Jan  7 22:16 ib_buffer_pool
-rw-rw---- 1 mysql  mysql  12582912 Jan  7 22:16 ibdata1
-rw-rw---- 1 mysql  mysql  50331648 Jan  7 22:16 ib_logfile0
-rw-rw---- 1 mysql  mysql  50331648 Jan  7 22:16 ib_logfile1
drwx------ 2 mysql  mysql      4096 Jan  7 22:16 mysql
drwx------ 2 mysql  mysql      4096 Jan  7 22:16 performance_schema
`
#### Additional logs:
```
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

Joulinar, 2021-01-08:
Hi,

Many thanks for your message. Pls can you provide following.

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

zorlaski, 2021-01-08:
> systemctl restart mariadb.service

same error as before
> journalctl -u mariadb
```
Jan 07 23:17:21 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 07 23:17:22 DietPi mysqld[28528]: 2021-01-07 23:17:22 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 28528 ...
Jan 07 23:17:22 DietPi mysqld[28528]: 2021-01-07 23:17:22 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 32186)
Jan 07 23:17:22 DietPi mysqld[28528]: 2021-01-07 23:17:22 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 07 23:17:22 DietPi mysqld[28528]: [90B blob data]
Jan 07 23:17:22 DietPi mysqld[28528]: 2021-01-07 23:17:22 0 [ERROR] Aborting
Jan 07 23:17:22 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 07 23:17:22 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
```
> cat /var/log/mysql/error.log
file is empty
> readlink /var/lib/mysql
`/mnt/dietpi_userdata/mysql`
> readlink -f /var/lib/mysql
`/home/dietpi/raid1/mysql`

Joulinar, 2021-01-08:
Just to be sure, the following is working and you see the demo file created

```
sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
ls -la /var/lib/mysql/
```

zorlaski, 2021-01-08:
> ```
> sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
> ```
"Sorry, user dietpi is not allowed to execute '/usr/bin/touch /var/lib/mysql/demo.file' as mysql:mysql on DietPi."
(the command succeeds when run as root)
>```
> ls -la /var/lib/mysql/
> ```
```
total 110632
drwxr-xr-x 4 mysql  mysql      4096 Jan  7 22:16 .
drwxrwxr-x 8 dietpi dietpi     4096 Jan  8 00:13 ..
-rw-rw---- 1 mysql  mysql     16384 Jan  7 22:16 aria_log.00000001
-rw-rw---- 1 mysql  mysql        52 Jan  7 22:16 aria_log_control
-rw-r--r-- 1 root   root          0 Jan  7 22:16 debian-10.3.flag
-rw-rw---- 1 mysql  mysql       972 Jan  7 22:16 ib_buffer_pool
-rw-rw---- 1 mysql  mysql  12582912 Jan  7 22:16 ibdata1
-rw-rw---- 1 mysql  mysql  50331648 Jan  7 22:16 ib_logfile0
-rw-rw---- 1 mysql  mysql  50331648 Jan  7 22:16 ib_logfile1
drwx------ 2 mysql  mysql      4096 Jan  7 22:16 mysql
drwx------ 2 mysql  mysql      4096 Jan  7 22:16 performance_schema

```

Joulinar, 2021-01-08:
yes, if possible use user `root` for the investigation.

But strange I don't see the `demo.file` created on `/var/lib/mysql/`. Are you sure command was executed without errors?

zorlaski, 2021-01-08:
oops i forgot to run the ls again. here it is now.
```
drwxr-xr-x 4 mysql  mysql      4096 Jan  8 01:11 .
drwxrwxr-x 8 dietpi dietpi     4096 Jan  8 00:13 ..
-rw-rw---- 1 mysql  mysql     16384 Jan  7 22:16 aria_log.00000001
-rw-rw---- 1 mysql  mysql        52 Jan  7 22:16 aria_log_control
-rw-r--r-- 1 root   root          0 Jan  7 22:16 debian-10.3.flag
-rw-r--r-- 1 mysql  mysql         0 Jan  8 01:11 demo.file
-rw-rw---- 1 mysql  mysql       972 Jan  7 22:16 ib_buffer_pool
-rw-rw---- 1 mysql  mysql  12582912 Jan  7 22:16 ibdata1
-rw-rw---- 1 mysql  mysql  50331648 Jan  7 22:16 ib_logfile0
-rw-rw---- 1 mysql  mysql  50331648 Jan  7 22:16 ib_logfile1
drwx------ 2 mysql  mysql      4096 Jan  7 22:16 mysql
drwx------ 2 mysql  mysql      4096 Jan  7 22:16 performance_schema

````

Joulinar, 2021-01-08:
ok let's try quite simple way to start database manually. If it's working, you should see the command hanging. 

`sudo -u mysql -g mysql /usr/sbin/mysqld`

Just open a 2nd SSH session to verify `cat /var/log/mysql/error.log`

zorlaski, 2021-01-08:
```
dietpi@DietPi:~$ sudo cat /var/log/mysql/error.log
2021-01-08  1:35:16 0 [Note] InnoDB: Using Linux native AIO
2021-01-08  1:35:16 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2021-01-08  1:35:16 0 [Note] InnoDB: Uses event mutexes
2021-01-08  1:35:16 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-01-08  1:35:16 0 [Note] InnoDB: Number of pools: 1
2021-01-08  1:35:16 0 [Note] InnoDB: Using generic crc32 instructions
2021-01-08  1:35:16 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2021-01-08  1:35:16 0 [Note] InnoDB: Completed initialization of buffer pool
2021-01-08  1:35:16 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-01-08  1:35:17 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-01-08  1:35:17 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-01-08  1:35:17 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-01-08  1:35:17 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-01-08  1:35:17 0 [Note] InnoDB: 10.3.27 started; log sequence number 1625452; transaction id 20
2021-01-08  1:35:17 0 [Note] InnoDB: Loading buffer pool(s) from /home/dietpi/raid1/mysql/ib_buffer_pool
2021-01-08  1:35:17 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-01-08  1:35:17 0 [Note] InnoDB: Buffer pool(s) load completed at 210108  1:35:17
2021-01-08  1:35:17 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-08  1:35:17 0 [Note] Reading of all Master_info entries succeeded
2021-01-08  1:35:17 0 [Note] Added new Master_info '' to hash table
2021-01-08  1:35:17 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.27-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
```

Joulinar, 2021-01-08:
ok this seems to be working, strange. Simply kill the process to stop the manuell started database. 

But the real service is not working? As well it's not producing any output on the log file?

zorlaski, 2021-01-08:
here is the current log after killing the manually started database. Starting mariadb with dietpi-services continues to fail.

```
dietpi@DietPi:~$ sudo cat /var/log/mysql/error.log
2021-01-08  1:35:16 0 [Note] InnoDB: Using Linux native AIO
2021-01-08  1:35:16 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2021-01-08  1:35:16 0 [Note] InnoDB: Uses event mutexes
2021-01-08  1:35:16 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-01-08  1:35:16 0 [Note] InnoDB: Number of pools: 1
2021-01-08  1:35:16 0 [Note] InnoDB: Using generic crc32 instructions
2021-01-08  1:35:16 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2021-01-08  1:35:16 0 [Note] InnoDB: Completed initialization of buffer pool
2021-01-08  1:35:16 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-01-08  1:35:17 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-01-08  1:35:17 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-01-08  1:35:17 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-01-08  1:35:17 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-01-08  1:35:17 0 [Note] InnoDB: 10.3.27 started; log sequence number 1625452; transaction id 20
2021-01-08  1:35:17 0 [Note] InnoDB: Loading buffer pool(s) from /home/dietpi/raid1/mysql/ib_buffer_pool
2021-01-08  1:35:17 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-01-08  1:35:17 0 [Note] InnoDB: Buffer pool(s) load completed at 210108  1:35:17
2021-01-08  1:35:17 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-08  1:35:17 0 [Note] Reading of all Master_info entries succeeded
2021-01-08  1:35:17 0 [Note] Added new Master_info '' to hash table
2021-01-08  1:35:17 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.27-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10

Status information:

Current dir: /var/lib/mysql/
Running threads: 5  Cached threads: 0  Stack size: 299008

Key caches:
default
Buffer_size:     134217728
Block_size:           1024
Division_limit:        100
Age_threshold:         300
Partitions:              0
blocks used:             0
not flushed:             0
w_requests:              0
writes:                  0
r_requests:              0
reads:                   0

handler status:
read_key:            0
read_next:           0
read_rnd             0
read_first:          3
write:               0
delete               0
update:              0

Table status:
Opened tables:         17
Open tables:           11
Open files:            25
Open streams:           0

Alarm status:
Active alarms:   0
Max used alarms: 0
Next alarm time: 0

Memory status:
Non-mmapped space allocated from system: 6582272
Number of free chunks:                   19
Number of fastbin blocks:                0
Number of mmapped regions:               20
Space in mmapped regions:                296873984
Maximum total allocated space:           0
Space available in freed fastbin blocks: 0
Total allocated space:                   5692088
Total free space:                        890184
Top-most, releasable space:              27912
Estimated memory (with thread stack):    304951296
Global memory allocated by server:       288827288
Memory allocated by threads:             87520

Events status:
LLA = Last Locked At  LUA = Last Unlocked At
WOC = Waiting On Condition  DL = Data Locked

Event scheduler status:
State      : INITIALIZED
Thread id  : 0
LLA        : n/a:0
LUA        : n/a:0
WOC        : NO
Workers    : 0
Executed   : 0
Data locked: NO

Event queue status:
Element count   : 0
Data locked     : NO
Attempting lock : NO
LLA             : init_queue:140
LUA             : init_queue:150
WOC             : NO
Next activation : never
2021-01-08  1:45:24 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
2021-01-08  1:45:24 0 [Note] Event Scheduler: Purging the queue. 0 events
2021-01-08  1:45:24 0 [Note] InnoDB: FTS optimize thread exiting.
2021-01-08  1:45:24 0 [Note] InnoDB: Starting shutdown...
2021-01-08  1:45:24 0 [Note] InnoDB: Dumping buffer pool(s) to /home/dietpi/raid1/mysql/ib_buffer_pool
2021-01-08  1:45:24 0 [Note] InnoDB: Buffer pool(s) dump completed at 210108  1:45:24
2021-01-08  1:45:25 0 [Note] InnoDB: Shutdown completed; log sequence number 1625461; transaction id 21
2021-01-08  1:45:25 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-01-08  1:45:25 0 [Note] /usr/sbin/mysqld: Shutdown complete
```

Joulinar, 2021-01-08:
that's strange, The database is able to stat manuell but not with the service. As well no error.log messages are created using the service. 

@MichaIng any ideas

MichaIng, 2021-01-08:
The MariaDB systemd unit has some internal restrictions: `systemctl cat mariadb`
```
# Prevent writes to /usr, /boot, and /etc
ProtectSystem=full

# Doesn't yet work properly with SELinux enabled
# NoNewPrivileges=true

PrivateDevices=true

# Prevent accessing /home, /root and /run/user
ProtectHome=true
```
`/home/dietpi/raid1/mysql` access is hence not permitted. Based on the concept of a system service this makes sense. Users home directories are meant to contain files which are relevant for this single UNIX login user only and by default read/write-able for that single user only. MariaDB however is a system-wide running server that can be accessed via port and UNIX socket by every user, of course when being able to authenticate. Based on this concept, granting a system services, that can be theoretically controlled by any user, write access to your login users private home directory, grants the whole system theoretical write access. So much about that permissions/user concept, while of course DietPi on an SBC is usually a single "real" user machine and we introduce that `/mnt/dietpi_userdata` directory which practically contains the relevant private data, accessible for a lot of system services when installing them through `dietpi-software`, but just to give some background 😉.

This means you can:
- either mount your RAID to e.g. a `/mnt/` sub directory, which would match our ideas
- or allow your own preferences, add `ReadWritePaths=/home/dietpi/raid1/mysql` to the systemd unit, via override configuration, e.g. start `dietpi-services` > `mariadb` > `Edit` > Add the above line below the `[Service]` line.

Joulinar, 2021-01-08:
ahh I never checked MariaDB service that deep for home directory protection. good to know.

zorlaski, 2021-01-08:
moving dietpi_userdata to /mnt/raid1 with dietpi-software worked fine,  mariadb starts now. thanks!

closed 2021-01-08 by zorlaski