MichaIng/DietPi #4044: Caddy server

jcw, 2021-01-08:
#### Formal software information
- Software title | Caddy server
- Short description | self-contained web- and proxy server
- Official URL | https://caddyserver.com
- GitHub URL | https://github.com/caddyserver/caddy
- (Official) install documentation | https://caddyserver.com/docs/install

#### Are there similar/alternative software titles available with DietPi-Software?
- Nginx, Apache, etc

#### What makes your requested software better than the above solutions, if available?
- does HTTPS (w/ Let's Encrypt) by default, for multiplel domains
- very simple "Caddyfile" config when used for simple configurations
- a single executable, and a single config file (unless you prefer to split it)
- built in Go, which has its own strong and independent network stack (with Google behind it)

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- a DIY install of caddy is easy, but it interferes with other web servers on ports 80 & 443
- the key benefit (auto-HTTPS) of caddy mainly adds value if it were made the "central" web server
- this is of course not going to satisfy everyone, so "switching to" caddy probably needs to be optional
- but this is already supported in DietPi, with the choice of a "Webserver Preference" in dietpi-software

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Install Caddy (single-exe, just fetch the proper one from github release page)
2. prepare minimal CaddyFile (the usual path is /etc/caddy/, I think)
3. add new "caddy.service" to systemd, see https://caddyserver.com/docs/install#linux-service

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- I can help, in that I'm fairly familiar with Go, and have been using Caddy for some time (v2 only since mid 2020)
- I've set it up for static sites, as PHP proxy (e.g. Nextcloud), and as reverse proxy (e.g. Gitea)
- and FWIW: no "deep" insight into Caddy, moderate networking knowledge, well-versed in Unix stuff

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/
https://feathub.com/MichaIng/DietPi/+100

(related issue: #1622)

gt2416, 2021-11-03:
+1 for this. Caddy automatically manages certificates and setting it up as a reverse proxy is so much easier than using Apache / nginx.

jalsco, 2021-11-30:
Hmmm - I been using Caddy Server on DeitPi Buster since August 2021. It works flawlessly (and it is awesome 😁✨) .  I agree with the other posters that if it were in the autoinstall packages that would be better.  As now it is a mnaul install - which is very straightforward (just follow the steps on the Caddy website for Debian ) .

I was checking in here in case reported problems running caddy on the new Deipi Pi (Debian 11) . I did not see any issues reported - so I will continue with the DietPi upgrade.  Assuming Caddy runs then I will not post again. However if it fails - I will come back and edit this post  and also downgrade by DietPi back to 10-Buster... But I expect it will work !

Xerrion, 2022-01-12:
+1 on this, have been using Caddy for over a year on my dietpi, would be really neat to have this available built-in as an alternative.

eikaramba, 2022-03-19:
+1 too - honestly i find caddy so much easier than nginx/lighttpd

for the others: is it a straight install on the dietpi debian 11? or anything to look out for?

jcw, 2022-03-19:
Yep, should just work - there's no install, it's a single executable.

Meanwhile, I'm about to fall back to Nginx + CertBot for a new server I'm preparing here, given the lack of support in DietPi.
Unfortunate, but not the end of the world. Still, as others point out, Caddy is a breath of fresh air. Wink, wink, @MichaIng :)

gt2416, 2022-03-19:
Nothing special to do. You can build it on diet pi easily. You have to install go first to build it. Works as you would expect