MichaIng/DietPi #4045: Odroid C4 | No serial console

dhuffer, 2021-01-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.34.3
- Distro version | buster
- Kernel version | Linux DietPi 4.9.236-arm64 
- SBC model | Odroid C4 (aarch64)
- Power supply used | 10V 4A Lab PSU
- SDcard used | (EG: Samsung Evo Plus)

#### Steps to reproduce

1. Flash DietPi Image to sd-card
2. Connect uart-console to debug port
3. Boot Odroid C4 from sd-card

#### Expected behaviour

- See bootup log and login prompt on uart-console

#### Actual behaviour

- no bootup log, no login prompt

#### Extra details

- The exact same setup works flawlessly at my other Odroid devices (C2, HC2, HC1, MC1)

MichaIng, 2021-01-08:
I just recognised the same one minute ago, when building the new image for definite HC4 support. I'm currently checking the intended serial device, is it `/dev/ttyS0`, i.e. if you connect via SSH, does `systemctl enable --now serial-getty@ttyS0` makes the login mask appear on the UART console?

dhuffer, 2021-01-08:
I can't confirm that. The last message at the console is "hw id:" and it stays that way, even after the systemctl command. What tests can I do further?

dhuffer, 2021-01-08:
It works! I checked with a second (and different) terminal adapter.

MichaIng, 2021-01-08:
Ah okay great, I just got confused as `/boot/boot.ini` shows it pretty clear. Theoretically, `/dev/ttyS1` + `/dev/ttyS2` can be made accessible at the 40-pin GPIO optionally, but that 4-pin connector is `/dev/ttyS0` by default: https://wiki.odroid.com/odroid-c4/application_note/gpio/uart

dhuffer, 2021-01-08:
Yes, thanks for your effort. I was confused, too. Maybe it's an issue with the compatibility of certain serial adapters, where the older Odroids were not as picky. But that's not an dietpi issue.

Thanks for your great work 🥇

closed 2021-01-08 by dhuffer

MichaIng, 2021-01-08:
Thanks for the kind feedback. I just updated the image to have the serial console enabled automatically + full Odroid HC4 support (SATA).