MichaIng/DietPi #4046: Pi-Hole - Query logs and other lists in the webinterface are empty

jwgn, 2021-01-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.7

- Kernel version | `uname -a`
Linux pihole 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
Raspberry Pi PoE HAT
PoE Switch Ubiquity EdgeSwitch 24 Lite

- SDcard used | (EG: SanDisk ultra)
unknown

#### Additional Information (if applicable)
- Software title
Pi-Hole

- Was the software title installed freshly or updated/migrated?
Updated a few times

- Can this issue be replicated on a fresh installation of DietPi?
Not tried yet.

- Bug report ID
e6e6375d-8e4e-47e8-91f7-500c648ff17f

#### Steps to reproduce

1. Open Pi-Hole Webinterface
2. Click on Query Log / Whitelist / Blacklist etc.

#### Expected behaviour

You should see a list of entries

#### Actual behaviour

No entries at all, cannot edit any lists or view queries

#### Extra details

- ...

Joulinar, 2021-01-09:
Hi,

Did you clear cache and cookies on your browser? Did you restart your system?

jwgn, 2021-01-09:
Hi,

Indeed there seems to be something wrong with my system. It does not work even after resetting Firefox, but it works on my Chrome OS tablet. So I'm investigating further here. Sorry for bothering :)

closed 2021-01-09 by jwgn

MichaIng, 2021-01-09:
Sounds like a client-side issue then? Did you check the browser console? I had a look into your bug report but couldn't see any obvious issue server-wise. I see you use two upstream DNS servers, Unbound, and the other looks like it's pointing to your router? AFAIK Pi-hole does not strictly prioritise those in order or such (correct me if I am wrong), so if you want to have Unbound used for every query by Pi-hole, you need to remove the other entry. In case you specify Pi-hole as the upstream DNS in your router (to pass to DHCP clients), then adding the router as upstream DNS to Pi-hole would create a loop 😉.