MichaIng/DietPi #4048: Unable to update diet-pi

careless-havoc, 2021-01-11:
### Creating a bug report/issue
Pi-hole noob so any help would be appreciated

#### Details:
- Date           | Mon Jan 11 22:29:03 GMT 2021
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...logged into the pihole, ran dietpi-update per instructions on diet-pi's site
2. ...
#### Expected behaviour:

- ...Successfully install update
#### Actual behaviour:

- ...Fails with errors below
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: Problem parsing dependency 2
E: Error occurred while processing dpkg (NewVersion2)
E: Problem with MergeList /var/lib/dpkg/status
E: The package lists or status file could not be parsed or opened.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated? No
- Can this issue be replicated on a fresh installation of DietPi? Have not tried, would rather not set this up again

- Bug report ID | `echo $G_HW_UUID`

careless-havoc, 2021-01-11:
bug reference code 
7c68cf53-ef7e-4ea3-8823-c788d2b9a2b8

Joulinar, 2021-01-11:
Hi,

many thanks for your message. Looks like the dpkg status file got corruptede
`E: Problem with MergeList /var/lib/dpkg/status`

You could try following 

```
mv /var/lib/dpkg/status /var/lib/dpkg/status.old
cp /var/backups/dpkg.status.0 /var/lib/dpkg/status
apt update
apt upgrade
```

careless-havoc, 2021-01-11:
Ran those and everything went smoothly other than the apt upgrade. Got the same error at the end of apt upgrade and when trying dietpi-update after the apt upgrade failed

Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 25543 files and directories currently installed.)
Preparing to unpack .../libapt-pkg5.0_1.8.2.2_arm64.deb ...
Unpacking libapt-pkg5.0:arm64 (1.8.2.2) over (1.8.2.1) ...
dpkg: error processing archive /var/cache/apt/archives/libapt-pkg5.0_1.8.2.2_arm64.deb (--unpack):
 unable to stat './usr/share/locale/it/LC_MESSAGES/libapt-pkg5.0.mo' (which I was about to install): Structure needs cleaning
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libapt-pkg5.0_1.8.2.2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
/boot/dietpi/func/dietpi-globals: line 541: 15452 Illegal instruction     stty size
/boot/dietpi/func/dietpi-globals: line 543: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/boot/dietpi/func/dietpi-globals: line 549: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")

Joulinar, 2021-01-11:
as long as `apt upgrade` is failing, `dietpi-update` will fail as well because it's doing exactly the same. Pls can you post the output of

`apt update && apt upgrade`

careless-havoc, 2021-01-11:
root@DietPi:~# apt update && apt upgrade
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease            
Hit:4 https://deb.debian.org/debian buster-backports InRelease                   
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
9 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  apt libapt-pkg5.0 libp11-kit0 libssl1.1 openssl raspberrypi-bootloader raspberrypi-kernel rpi-eeprom tzdata
9 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/31.3 MB of archives.
After this operation, 199 MB disk space will be freed.
Do you want to continue? [Y/n] y
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 25543 files and directories currently installed.)
Preparing to unpack .../libapt-pkg5.0_1.8.2.2_arm64.deb ...
Unpacking libapt-pkg5.0:arm64 (1.8.2.2) over (1.8.2.1) ...
dpkg: error processing archive /var/cache/apt/archives/libapt-pkg5.0_1.8.2.2_arm64.deb (--unpack):
 unable to stat './usr/share/locale/it/LC_MESSAGES/libapt-pkg5.0.mo' (which I was about to install): Structure needs cleaning
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libapt-pkg5.0_1.8.2.2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~#

Joulinar, 2021-01-11:
can you check for file system errors `fsck -a /dev/mmcblk0p2`

careless-havoc, 2021-01-11:
root@DietPi:~ fsck -a /dev/mmcblk0p2
fsck from util-linux 2.33.1
/dev/mmcblk0p2 is mounted.
e2fsck: Cannot continue, aborting.

The next option failed with the same error as before

Joulinar, 2021-01-11:
ah right my fault. it's in use and mounted. what about cleaning cache.

careless-havoc, 2021-01-11:
root@DietPi:~ /boot/dietpi/func/dietpi-set_software apt-cache clean
[ SUB1 ] DietPi-Set_software > apt-cache (clean)
[  OK  ] apt-cache clean | Completed

That cache clean ran fine but after trying the update again, it failed with the same error

Joulinar, 2021-01-11:
pls can you post full output

careless-havoc, 2021-01-11:
root@DietPi:~# /boot/dietpi/func/dietpi-set_software apt-cache clean
[ SUB1 ] DietPi-Set_software > apt-cache (clean)
[  OK  ] apt-cache clean | Completed
root@DietPi:~# apt update && apt upgrade
Get:1 https://deb.debian.org/debian buster InRelease [121 kB]
Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]                
Get:3 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB] 
Get:4 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]               
Get:5 https://deb.debian.org/debian buster/non-free armhf Packages [62.0 kB]           
Get:6 https://deb.debian.org/debian buster/contrib arm64 Packages [38.4 kB]
Get:7 https://deb.debian.org/debian buster/main arm64 Packages [7737 kB]         
Get:8 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:9 https://deb.debian.org/debian buster/contrib armhf Packages [40.1 kB]
Get:10 https://deb.debian.org/debian buster/non-free arm64 Packages [53.8 kB]
Get:11 https://deb.debian.org/debian buster/main armhf Packages [7698 kB]
Get:12 https://deb.debian.org/debian buster-updates/non-free arm64 Packages [600 B]
Get:13 https://deb.debian.org/debian buster-updates/non-free armhf Packages [604 B]
Get:14 https://deb.debian.org/debian buster-updates/main arm64 Packages [7848 B]
Get:15 https://deb.debian.org/debian buster-updates/main armhf Packages [7848 B] 
Get:16 https://deb.debian.org/debian-security buster/updates/main armhf Packages [254 kB]
Get:17 https://deb.debian.org/debian-security buster/updates/main arm64 Packages [254 kB]
Get:18 https://deb.debian.org/debian buster-backports/main arm64 Packages [410 kB]
Get:19 https://deb.debian.org/debian buster-backports/main armhf Packages [404 kB]
Get:20 https://deb.debian.org/debian buster-backports/contrib armhf Packages [7120 B]
Get:21 https://deb.debian.org/debian buster-backports/non-free armhf Packages [13.1 kB]
Get:22 https://deb.debian.org/debian buster-backports/non-free arm64 Packages [18.7 kB]
Get:23 https://deb.debian.org/debian buster-backports/contrib arm64 Packages [7144 B]
Get:24 https://archive.raspberrypi.org/debian buster/main armhf Packages [351 kB]
Get:25 https://archive.raspberrypi.org/debian buster/main arm64 Packages [260 kB]
Fetched 17.9 MB in 7s (2485 kB/s)                                                                                                                                                                  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
9 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  apt libapt-pkg5.0 libp11-kit0 libssl1.1 openssl raspberrypi-bootloader raspberrypi-kernel rpi-eeprom tzdata
9 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 31.3 MB of archives.
After this operation, 199 MB disk space will be freed.
Do you want to continue? [Y/n] y
Get:1 https://deb.debian.org/debian-security buster/updates/main arm64 libapt-pkg5.0 arm64 1.8.2.2 [897 kB]
Get:2 https://deb.debian.org/debian-security buster/updates/main arm64 apt arm64 1.8.2.2 [1370 kB]
Get:3 https://deb.debian.org/debian-security buster/updates/main arm64 libp11-kit0 arm64 0.23.15-2+deb10u1 [303 kB]
Get:4 https://deb.debian.org/debian buster-updates/main arm64 tzdata all 2020e-0+deb10u1 [266 kB]
Get:5 https://deb.debian.org/debian-security buster/updates/main arm64 libssl1.1 arm64 1.1.1d-0+deb10u4 [1382 kB]
Get:6 https://deb.debian.org/debian-security buster/updates/main arm64 openssl arm64 1.1.1d-0+deb10u4 [823 kB]
Get:7 https://archive.raspberrypi.org/debian buster/main arm64 raspberrypi-kernel arm64 1.20210111.master-1 [21.4 MB]
Get:8 https://archive.raspberrypi.org/debian buster/main arm64 raspberrypi-bootloader arm64 1.20210111.master-1 [4484 kB]
Get:9 https://archive.raspberrypi.org/debian buster/main arm64 rpi-eeprom arm64 11.1-1 [407 kB]
Fetched 31.3 MB in 3s (9392 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 25543 files and directories currently installed.)
Preparing to unpack .../libapt-pkg5.0_1.8.2.2_arm64.deb ...
Unpacking libapt-pkg5.0:arm64 (1.8.2.2) over (1.8.2.1) ...
dpkg: error processing archive /var/cache/apt/archives/libapt-pkg5.0_1.8.2.2_arm64.deb (--unpack):
 unable to stat './usr/share/locale/it/LC_MESSAGES/libapt-pkg5.0.mo' (which I was about to install): Structure needs cleaning
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libapt-pkg5.0_1.8.2.2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2021-01-11:
@MichaIng 
can you have a look. You are the expert on such apt package issues. 😃

MichaIng, 2021-01-12:
"structure needs cleaning" means file system corruption. So that should be sorted before doing any other steps. Please try to have it repaired at reboot:
```
> /forcefsck
reboot
```
I am a bit worried due to the "Illegal instruction" on the stty call, which could mean there are more files affected.

After reboot, please print the following to check for I/O kernel errors:
```
dmesg -l emerg,alert,crit,err
```

careless-havoc, 2021-01-12:
root@DietPi:~ dmesg -l emerg,alert,crit,err
[    0.423319] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[   17.023242] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   17.023246] [vc_sm_connected_init]: failed to initialize shared memory service
[   17.405415] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
root@DietPi:~#

careless-havoc, 2021-01-12:
I re-ran the update after getting the above output and it appears to be working. It is much further along than it had been. I'll update with the results. 
Many thanks to you both for helping out!

careless-havoc, 2021-01-12:
It worked and updated, it appeared to nuke all my settings and block lists and the dashboard was blank until I rebooted. 

Thanks for the help!

closed 2021-01-12 by careless-havoc