MichaIng/DietPi #4049: Pulse audio

symbios24, 2021-01-13:
Hello
I installed pulseaudio for a program that only has sound with pulse,but i only have audio from headphones 
How do i get audio with pulse from the hdmi/speakers ?
Thanks

MichaIng, 2021-01-13:
Many thanks for your question.

Since we consequently use plain ALSA for our (audio/media) software installs and config scripts, I have zero experience with PulseAudio and how to configure it. Did you try to get sound from HDMI via ALSA, e.g. `aplay` test, to assure that it works hardware-wise?

And on which device our trying this? In case of RPi, it would be possible to completely disable the headphones (3.5mm jack) so that HDMI would be the only available sound card interface, if it's only that PA selects the wrong one.

`aplay -l` is lastly a good command to get an overview of kernel-wise detected sound cards/devices. This is true for PA as well, as it basically an ALSA wrapper.

symbios24, 2021-01-13:
The sound in dietpi works only in the program does not work, managed to get it work in jack3.5 headphones only by installing pulseaudio 
Its an Rpi4 i tried in an Rpi3 and without pulseaudio installed the sound works there,so it must be an Rpi4 issue.

MichaIng, 2021-01-13:
Hmm, the sound card interface on RPi 3 and 4 should work identical, besides the fact that on RPi 4 there are two HDMI cards available, when two HDMI monitors are attached. `modprobe snd-bcm2835` should be necessary when `dtparam=audio=on` has been added to `/boot/config.txt`, or RPi onboard audio enabled in `dietpi-config`. In this case, all kernel-wise modules and interfaces are fully loaded automatically.

Really, check `aplay -l`, if you see the HDMI sound card and to know its PCM name (respectively card index and device index) and if you can play sound through it, via something like `aplay -D hw0,0 /path/to/testfile.wav` (as HDMI should be on card 0 device 0, when available).

It is expected that sound output via ALSA does not work while PulseAudio "blocks" it. ALSA generally (when not configured specifically) can only play one sound source at a time. PulseAudio has mixing included, so you can play multiple sound sources concurrently through PulseAudio, but since it is an own ALSA stream, plain ALSA then does not work concurrently.

symbios24, 2021-01-14:
Strange thing it like looks like something changed in Rpi4

C0D3-M4513R, 2021-01-22:
Have you tried explicitly forcing hdmi output on the pi? Maybe only one audio/alsa card is active at a time, which causes pulse to only see that one. Maybe you forced the audio-card to 3.5mm jack, instead to your desired hdmi. 

For me it is currently the same thing, although I do have 3.5mm jack forced, since I do not use hdmi output. Therefore only the 3.5mm jack on my pi pops up.

MichaIng, 2021-01-22:
Is one of you guys using the RPi KMS driver (`vc4-kms-v3d-pi4` resolution option in dietpi-config or manually added as dtoverlay)? At least on RPi 3 and prior this adds an additional sound device on card 0 (incrementing all existing sound card indices) and what we never checked is if it adds even two additional sound devices on RPi 4 and if so, if this is done regardless if both HDMI adapters are enabled or disabled or monitors attached.

This might at least break our asound.conf on RPi 4 when this driver is enabled and probably it as well causes issues with PulseAudio card detection.

MichaIng, 2021-01-22:
And now I see where I got the idea from, lol: https://github.com/MichaIng/DietPi/issues/4059#issuecomment-764599639
So indeed two additional sound cards are added, at least even if either no HDMI monitor is attached or HDMI audio is disabled.

closed 2021-05-07 by MichaIng