MichaIng/DietPi #4050: vmtouch fails with segmenation fault on raspberry RPi B (armv6l)

berndhose, 2021-01-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-01-14:
Many thanks for your report.

Did you face this issue on a fresh image?

Could you try the following:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/buster/vmtouch_armv6l.deb
apt install --reinstall ./vmtouch_armv6l.deb
```

berndhose, 2021-01-15:
yes, happened on fresh image right after automatic first time upgrade.

reinstalling vmtouch from dietpi repository resolved the issue

many thanks

MichaIng, 2021-01-15:
Strange, probably I created that image before I added forced armv6l for 32-bit RPi images to the creation script. I'll update all our RPi images tomorrow.

MichaIng, 2021-02-07:
All RPi images have been updated meanwhile.

closed 2021-02-14 by MichaIng