MichaIng/DietPi #4051: Nextcloud install failed, please help.

Reen85, 2021-01-15:
### Creating a bug report/issue
This happened, when I tried to install Nextcloud via Dietpi-Software. I tried it a few time with always the same result, please see details below. What went rent or what did I do wrong. Just to let you know, I moved the Dietpi-User-Data to an external drive in advance (via drive-manager) to store NC data on external drive...
Thanks a lot for your support, keep on going with this fantastic software project!

#### Details:
- Date 	  	| Thu Jan 14 22:07:25 GMT 2021

- DietPi version	| v6.34.3 (Michalng/master)

- Image creator 	| DietPi Core Team

- Pre-image		| Raspbian Lite

- Hardware		| RPi 4 Model B (armv71) (ID=4)

- Kernel version 	| Linux DietPi 5.4.83-v71+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 arm v7 GNU/Linux

- Distro 		| buster (1D=5,RASPBIAN=1)

- Command 		| systemcl restart mariadb

- Exit code 		| 1

- Software title	| Dietpi-Software

#### Steps to reproduce:

1...

#### Expected behaviour:

#### Actual behaviour:

#### Extra details:

#### Additional logs: 

Job for mariadb.service failed because the control process exited with error code. See “systemctl status mariadb.service" and "journalctl -xe" for details for details.

Joulinar, 2021-01-15:
Hi,

many thanks for your message. Pls can you share following

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

Reen85, 2021-01-15:
Hi, 
Thanks a lot for your, please see results:

systemctl restart mariadb.service
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status mariadb.service
● mariadb.service - MariaDB 10.3.27 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset:
 enabled)
   Active: failed (Result: exit-code) since Fri 2021-01-15 22:07:46 G
MT; 17s ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 5769 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/ru
n/mysqld (code=exited, status=0/SUCCESS)
  Process: 5770 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START
_POSITION (code=exited, status=0/SUCCESS)
  Process: 5772 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR
= ||   VAR=`cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemc
tl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/S
UCCESS)
  Process: 5829 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WS
REP_START_POSITION (code=exited, status=1/FAILURE)
 Main PID: 5829 (code=exited, status=1/FAILURE)

Jan 15 22:07:43 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 15 22:07:43 DietPi mysqld[5829]: 2021-01-15 22:07:43 0 [Note] /usr/sbin/mysq
ld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 5829 ...
Jan 15 22:07:43 DietPi mysqld[5829]: 2021-01-15 22:07:43 0 [Warning] Could not i
ncrease number of max_open_files to more than 16384 (request: 32186)
Jan 15 22:07:46 DietPi systemd[1]: mariadb.service: Main
 process exited, code=exited, status=1/FAILURE
Jan 15 22:07:46 DietPi systemd[1]: mariadb.service: Fail
ed with result 'exit-code'.
Jan 15 22:07:46 DietPi systemd[1]: Failed to start Maria
DB 10.3.27 database server.

root@DietPi:~# journalctl -u mariadb

-- Logs begin at Fri 2021-01-15 13:38:54 GMT, end at Fri 2021-01-15 22:09:34 GMT. --
Jan 15 13:39:49 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 15 13:39:53 DietPi mysqld[689]: 2021-01-15 13:39:53 0 [Note] /usr/sbin/mysqld (mysqld 10.3
.27-MariaDB-0+deb10u1) starting as process 689 ...
Jan 15 13:39:53 DietPi mysqld[689]: 2021-01-15 13:39:53 0 [Warning] Could not increase number 
of max_open_files to more than 16384 (request: 32186)
Jan 15 13:39:54 DietPi systemd[1]: mariadb.service: Main process exite
d, code=exited, status=1/FAILURE
Jan 15 13:39:54 DietPi systemd[1]: mariadb.service: Failed with result
 'exit-code'.
Jan 15 13:39:54 DietPi systemd[1]: Failed to start MariaDB 10.3.27 dat
abase server.
Jan 15 13:42:43 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 15 13:42:43 DietPi mysqld[4375]: 2021-01-15 13:42:43 0 [Note] /usr/sbin/mysqld (mysqld 10.
3.27-MariaDB-0+deb10u1) starting as process 4375 ...
Jan 15 13:42:43 DietPi mysqld[4375]: 2021-01-15 13:42:43 0 [Warning] Could not increase number
 of max_open_files to more than 16384 (request: 32186)
Jan 15 13:42:47 DietPi systemd[1]: mariadb.service: Main process exite
d, code=exited, status=1/FAILURE
Jan 15 13:42:47 DietPi systemd[1]: mariadb.service: Failed with result
 'exit-code'.
Jan 15 13:42:47 DietPi systemd[1]: Failed to start MariaDB 10.3.27 dat
abase server.
Jan 15 13:43:12 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 15 13:43:12 DietPi mysqld[4508]: 2021-01-15 13:43:12 0 [Note] /usr/sbin/mysqld (mysqld 10.
3.27-MariaDB-0+deb10u1) starting as process 4508 ...
Jan 15 13:43:12 DietPi mysqld[4508]: 2021-01-15 13:43:12 0 [Warning] Could not increase number
 of max_open_files to more than 16384 (request: 32186)
Jan 15 13:43:13 DietPi systemd[1]: mariadb.service: Main process exite
d, code=exited, status=1/FAILURE
[... 263 lines cut ...]
2021-01-15 22:07:46 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-01-15 22:07:46 1 [ERROR] InnoDB: Failed to find tablespace for table `nextcloud`.`oc_appconfig` in the cache. Attempting to load the tablespace with space id 8
2021-01-15 22:07:46 1 [ERROR] InnoDB: Operating system error number 13 in a file operation.
2021-01-15 22:07:46 1 [ERROR] InnoDB: The error means mysqld does not have the access rights to the directory.
2021-01-15 22:07:46 1 [ERROR] InnoDB: Cannot open datafile for read-only: './nextcloud/oc_appconfig.ibd' OS error: 81
2021-01-15 22:07:46 1 [ERROR] InnoDB: Operating system error number 13 in a file operation.
2021-01-15 22:07:46 1 [ERROR] InnoDB: The error means mysqld does not have the access rights to the directory.
2021-01-15 22:07:46 1 [ERROR] InnoDB: Could not find a valid tablespace file for ``nextcloud`.`oc_appconfig``. Please refer to https://mariadb.com/kb/en/innodb-data-dictionary-troubleshooting/ for how to resolve the issue.
2021-01-15 22:07:46 0 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2021-01-15 22:07:46 0 [Note] Recovering after a crash using tc.log
2021-01-15 22:07:46 0 [Note] Starting crash recovery...
2021-01-15 22:07:46 0 [Note] Crash recovery finished.
2021-01-15 22:07:46 0 [Note] InnoDB: Buffer pool(s) load completed at 210115 22:07:46
2021-01-15 22:07:46 0 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2021-01-15 22:07:46 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-15 22:07:46 0 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist

root@DietPi:~# readlink /var/lib/mysql
/mnt/dietpi_userdata/mysql
root@DietPi:~# readlink -f /var/lib/mysql
/mnt/HDD1/dietpi_userdata/mysql

Regards

Reen85

MichaIng, 2021-01-15:
Looks like MariaDB has no write access to the directory. Cloud you show:
```
df -T /mnt/HDD1 # to see the file system
ls -l /mnt/HDD1/dietpi_userdata/mysql # to see file permissions
```

Joulinar, 2021-01-15:
as well you could try to create a `demo.file` to verify the access

`sudo -u mysql -g mysql touch /var/lib/mysql/demo.file`

Reen85, 2021-01-16:
Hi guys, sorry for the late answer and thanks a lot for your support!
Seems like I was wrong on how I wanted to set up NC on the raspi with a working external drive to store the data. I would really like to understand what I actually did wrong. I set up a new Dietpi install, mounted the HDD via drive manager, shifted user data to the HDD and than tried to install NC. What was the mistake? I thought drive manager would set correct permission to make everything work smooth?!?

Please find the logs:
root@DietPi:~# df -T /mnt/HDD1
Filesystem     Type 1K-blocks      Used Available Use% Mounted on
/dev/sda1      ext4 960379920 199610012 711915484  22% /mnt/HDD1
root@DietPi:~# ls -l /mnt/HDD1/dietpi_userdata/mysql
total 188488
-rw-rw---- 1 sshd  mysql    16384 Jan 14 21:18 aria_log.00000001
-rw-rw---- 1 sshd  mysql       52 Jan 14 21:18 aria_log_control
-rw-r--r-- 1 root  root         0 Jan 14 21:18 debian-10.3.flag
-rw-rw---- 1 mysql mysql     1686 Jan 14 21:18 ib_buffer_pool
-rw-rw---- 1 sshd  mysql 50331648 Jan 15 22:07 ib_logfile0
-rw-rw---- 1 sshd  mysql 50331648 May 14  2020 ib_logfile1
-rw-rw---- 1 sshd  mysql 79691776 Jan 14 21:18 ibdata1
-rw-rw---- 1 mysql mysql 12582912 Jan 15 22:07 ibtmp1
-rw-rw---- 1 sshd  mysql        0 May 14  2020 multi-master.info
drwx------ 2 sshd  mysql     4096 May 14  2020 mysql
-rw-rw---- 1 root  mysql       16 May 14  2020 mysql_upgrade_info
drwx------ 2 sshd  mysql    12288 May 14  2020 nextcloud
drwx------ 2 sshd  mysql     4096 May 14  2020 performance_schema
-rw-rw---- 1 mysql mysql    24576 Jan 15 22:07 tc.log

Reen85, 2021-01-16:
and yes I could create the demo.file to the folder, but this folder is not located on the external HDD?!?

Joulinar, 2021-01-16:
not sure why but all your files are owned by user `sshd`. This should be the SSH Server user?? strange. This would need to be fixed

```
chown -R mysql /mnt/HDD1/dietpi_userdata/mysql
systemctl restart mariadb.service
```

MichaIng, 2021-01-16:
Probably that database was migrated from a different system where the mysql user ID matched the one of the sshd user on the new system?

Reen85, 2021-01-17:
Hi guys, you were absolutely right, changing the ownership solved the issue, install worked. Yes I did take the HDD from a former NC install, but I thought there were just some photos and videos on it. OK seem like I underestimated some other fragments on it, that caused trouble. I'll keep that in mind for next time!
Thank you so much for your support, appreciate your work!

Reen85, 2021-01-30:
Hi guys, 
unfortunately I am facing the same issue again, but this time on a different system: a fresh dietpi install, boot from usb (everything OK), mounted external HDD (fresh formatted, no data), shifted dietpi user data to external HDD. And now tried to install nextcloud and facing the same issues with mariadb again... 

ob for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

root@DietPi:~# ls -l /mnt/HDD1/dietpi_userdata/mysql
total 110596
-rwxr-xr-x 1 mysql mysql        0 Jan 30 10:38 debian-10.3.flag
-rwxr-xr-x 1 mysql mysql 50331648 Jan 30 10:38 ib_logfile0
-rwxr-xr-x 1 mysql mysql 50331648 Jan 30 10:38 ib_logfile1
-rwxr-xr-x 1 mysql mysql 12582912 Jan 30 10:38 ibdata1
drwxr-xr-x 2 mysql mysql     4096 Jan 30 10:38 mysql 

Are the permissions wrong again? What did I do wrong, I used system tools to prepare NC install, but mariadb service causes problems. Please help me guys...

Thanks a lot for your work,
Regards

Reno

MichaIng, 2021-01-30:
The permissions look fine. Could you show the full log:
```
journalctl -u mariadb
```
And let's assure that symlinks are still pointing correctly to the actual database dir:
```
ls -l /var/lib/mysql /mnt/dietpi_userdata/mysql
```

Reen85, 2021-01-30:
root@DietPi:~# journalctl -u mariadb
-- Logs begin at Thu 2019-02-14 10:11:58 GMT, end at Sat 2021-01-30 22:50:01 GMT. --
Jan 30 10:38:38 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 30 10:38:38 DietPi mysqld[2429]: 2021-01-30 10:38:38 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 2429
 ...
Jan 30 10:38:38 DietPi mysqld[2429]: 2021-01-30 10:38:38 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 
32186)
Jan 30 10:38:38 DietPi mysqld[2429]: 2021-01-30 10:38:38 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 10:38:38 DietPi mysqld[2429]: [90B blob data]
Jan 30 10:38:38 DietPi mysqld[2429]: 2021-01-30 10:38:38 0 [ERROR] Aborting
Jan 30 10:38:38 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 10:38:38 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 30 10:38:38 DietPi systemd[1]: Failed to start MariaDB 10.3.27 database server.
Jan 30 10:40:38 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 30 10:40:38 DietPi mysqld[7318]: 2021-01-30 10:40:38 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 7318
 ...
Jan 30 10:40:38 DietPi mysqld[7318]: 2021-01-30 10:40:38 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 
32186)
Jan 30 10:40:38 DietPi mysqld[7318]: 2021-01-30 10:40:38 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 10:40:38 DietPi mysqld[7318]: [90B blob data]
Jan 30 10:40:38 DietPi mysqld[7318]: 2021-01-30 10:40:38 0 [ERROR] Aborting
Jan 30 10:40:38 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 10:40:38 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 30 10:40:38 DietPi systemd[1]: Failed to start MariaDB 10.3.27 database server.
Jan 30 10:41:00 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 30 10:41:00 DietPi mysqld[7403]: 2021-01-30 10:41:00 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 7403
 ...
Jan 30 10:41:00 DietPi mysqld[7403]: 2021-01-30 10:41:00 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 
32186)
Jan 30 10:41:00 DietPi mysqld[7403]: 2021-01-30 10:41:00 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 10:41:00 DietPi mysqld[7403]: [90B blob data]
Jan 30 10:41:00 DietPi mysqld[7403]: 2021-01-30 10:41:00 0 [ERROR] Aborting
Jan 30 10:41:00 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 10:41:00 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 30 10:41:00 DietPi systemd[1]: Failed to start MariaDB 10.3.27 database server.
Jan 30 10:41:40 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 30 10:41:41 DietPi mysqld[7484]: 2021-01-30 10:41:41 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 7484
 ...
Jan 30 10:41:41 DietPi mysqld[7484]: 2021-01-30 10:41:41 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 
32186)
Jan 30 10:41:41 DietPi mysqld[7484]: 2021-01-30 10:41:41 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 10:41:41 DietPi mysqld[7484]: [90B blob data]
Jan 30 10:41:41 DietPi mysqld[7484]: 2021-01-30 10:41:41 0 [ERROR] Aborting
Jan 30 10:41:41 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 10:41:41 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 30 10:41:41 DietPi systemd[1]: Failed to start MariaDB 10.3.27 database server.
Jan 30 10:41:42 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 30 10:41:42 DietPi mysqld[7565]: 2021-01-30 10:41:42 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 7565
 ...
Jan 30 10:41:42 DietPi mysqld[7565]: 2021-01-30 10:41:42 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 
32186)
Jan 30 10:41:42 DietPi mysqld[7565]: 2021-01-30 10:41:42 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 10:41:42 DietPi mysqld[7565]: [90B blob data]
Jan 30 10:41:42 DietPi mysqld[7565]: 2021-01-30 10:41:42 0 [ERROR] Aborting
Jan 30 10:41:42 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 10:41:42 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 30 10:41:42 DietPi systemd[1]: Failed to start MariaDB 10.3.27 database server.
Jan 30 10:46:53 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Jan 30 10:46:53 DietPi mysqld[7826]: 2021-01-30 10:46:53 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 7826
 ...
Jan 30 10:46:53 DietPi mysqld[7826]: 2021-01-30 10:46:53 0 [Warning] Could not increase number of max_open_files to more than 16384 (request: 
32186)
Jan 30 10:46:53 DietPi mysqld[7826]: 2021-01-30 10:46:53 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 10:46:53 DietPi mysqld[7826]: [90B blob data]
Jan 30 10:46:53 DietPi mysqld[7826]: 2021-01-30 10:46:53 0 [ERROR] Aborting
Jan 30 10:46:53 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 10:46:53 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
[... cut ...]
Jan 30 22:50:01 DietPi mysqld[22148]: 2021-01-30 22:50:01 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 30 22:50:01 DietPi mysqld[22148]: [90B blob data]
Jan 30 22:50:01 DietPi mysqld[22148]: 2021-01-30 22:50:01 0 [ERROR] Aborting
Jan 30 22:50:01 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 22:50:01 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 30 22:50:01 DietPi systemd[1]: Failed to start MariaDB 10.3.27 database server.

Reen85, 2021-01-30:
oot@DietPi:~# ls -l /var/lib/mysql /mnt/dietpi_userdata/mysql
lrwxrwxrwx 1 root  root    26 Jan 30 22:48 /var/lib/mysql -> /mnt/dietpi_userdata/mysql

/mnt/dietpi_userdata/mysql:
total 110596
-rwxr-xr-x 1 mysql mysql        0 Jan 30 10:38 debian-10.3.flag
-rwxr-xr-x 1 mysql mysql 50331648 Jan 30 10:38 ib_logfile0
-rwxr-xr-x 1 mysql mysql 50331648 Jan 30 10:38 ib_logfile1
-rwxr-xr-x 1 mysql mysql 12582912 Jan 30 10:38 ibdata1
drwxr-xr-x 2 mysql mysql     4096 Jan 30 10:38 mysql

Joulinar, 2021-01-30:
Pls post `cat /var/log/mysql/error.log`

Reen85, 2021-01-30:
oot@DietPi:~# cat /var/log/mysql/error.log
2021-01-30 10:38:31 0 [ERROR] mysqld: Can't create/write to file '/var/lib/mysql/aria_log_control' (Errcode: 13 "Permission denied")
2021-01-30 10:38:31 0 [ERROR] mysqld: Got error 'Can't create file' when trying to use aria control file '/var/lib/mysql/aria_log_control'
2021-01-30 10:38:31 0 [ERROR] Plugin 'Aria' init function returned error.
2021-01-30 10:38:31 0 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
ERROR: 1  Can't create/write to file '/var/lib/mysql/mysql/db.MYI' (Errcode: 13 "Permission denied")
2021-01-30 10:38:33 0 [ERROR] Aborting

2021-01-30 10:38:33 0 [ERROR] InnoDB: Cannot open '/mnt/HDD1/dietpi_userdata/mysql/ib_buffer_pool.incomplete' for writing: Permission denied
Warning: Memory not freed: 392

Joulinar, 2021-01-30:
`ls -la /mnt/HDD1/dietpi_userdata/mysql/` pls

Reen85, 2021-01-30:
oot@DietPi:~# ls -la /mnt/HDD1/dietpi_userdata/mysql/
total 110604
drwxr-xr-x 3 mysql  mysql      4096 Jan 30 10:38 .
drwxrwxr-x 9 dietpi dietpi     4096 Jan 30 10:37 ..
-rwxr-xr-x 1 mysql  mysql         0 Jan 30 10:38 debian-10.3.flag
-rwxr-xr-x 1 mysql  mysql  50331648 Jan 30 10:38 ib_logfile0
-rwxr-xr-x 1 mysql  mysql  50331648 Jan 30 10:38 ib_logfile1
-rwxr-xr-x 1 mysql  mysql  12582912 Jan 30 10:38 ibdata1
drwxr-xr-x 2 mysql  mysql      4096 Jan 30 10:38 mysql

Joulinar, 2021-01-30:
This is working?

`sudo -u mysql -g mysql touch /var/lib/mysql/demo.file`

Reen85, 2021-01-30:
No seems not to work...
root@DietPi:~# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
touch: cannot touch '/var/lib/mysql/demo.file': Permission denied

Joulinar, 2021-01-30:
What file system format your disk has?

And pls following 

```
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

Reen85, 2021-01-30:
format ext4

root@DietPi:~# readlink /var/lib/mysql
/mnt/dietpi_userdata/mysql

root@DietPi:~# readlink -f /var/lib/mysql
/mnt/HDD1/dietpi_userdata/mysql

Joulinar, 2021-01-30:
Try following

```
chown -R mysql:mysql /mnt/HDD1/dietpi_userdata/mysql
sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
```

Reen85, 2021-01-30:
root@DietPi:~# chown -R mysql:mysql /mnt/HDD1/dietpi_userdata/mysql
root@DietPi:~# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
touch: cannot touch '/var/lib/mysql/demo.file': Permission denied

MichaIng, 2021-01-30:
Good test to write a file manually, so any service-implied hardenings are ruled out, like that the MariaDB service cannot write to a /root and /home, e.g. when being used as mount point.

Permissions and parent dir permissions are correct, user IDs match username, hmm. AFAIK at least execute permissions need to be granted to all parent directories to permit access. For symlink targets now the question is whether this applies for the symlink's parent dirs or the target parent dirs, or both, or if it depends on with which path the target is accessed 😄. Let's check both:
```
ls -dl /mnt{,/HDD1} /var{,/lib}
```
I just wanted to find a way to set modes of a symlink (not the target) but this is not possible as those are ignored, so no need to check.

Reen85, 2021-01-30:
root@DietPi:~# ls -dl /mnt{,/HDD1} /var{,/lib}
drwxr-xr-x  6 root   root   4096 Jan 30 10:35 /mnt
drwx------  4 dietpi dietpi 4096 Jan 30 10:35 /mnt/HDD1
drwxr-xr-x 12 root   root   4096 Jan 30 10:08 /var
drwxr-xr-x 20 root   root   4096 Jan 30 22:48 /var/lib

MichaIng, 2021-01-30:
Bingo:
```
chmod +x /mnt/HDD1
systemctl restart mariadb
```

Reen85, 2021-01-30:
oot@DietPi:~# chmod +x /mnt/HDD1
root@DietPi:~# systemctl restart maria
Failed to restart maria.service: Unit maria.service not found.
root@DietPi:~# chmod +x /mnt/HDD1
root@DietPi:~# systemctl restart mariadb
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

Reen85, 2021-01-30:
hmmm... it did not work yet?!?

MichaIng, 2021-01-30:
Probably read permissions as well?
```
chmod 755 /mnt/HDD1
systemctl restart mariadb
```

Reen85, 2021-01-30:
root@DietPi:~# chmod 755 /mnt/HDD1
root@DietPi:~# systemctl restart mariadb
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

but I think creating the demo file worked.

Joulinar, 2021-01-30:
Pls can you post `error.log` again

Reen85, 2021-01-30:
root@DietPi:~# cat /var/log/mysql/error.log
2021-01-30 10:38:31 0 [ERROR] mysqld: Can't create/write to file '/var/lib/mysql/aria_log_control' (Errcode: 13 "Permission denied")
2021-01-30 10:38:31 0 [ERROR] mysqld: Got error 'Can't create file' when trying to use aria control file '/var/lib/mysql/aria_log_control'
2021-01-30 10:38:31 0 [ERROR] Plugin 'Aria' init function returned error.
2021-01-30 10:38:31 0 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
ERROR: 1  Can't create/write to file '/var/lib/mysql/mysql/db.MYI' (Errcode: 13 "Permission denied")
2021-01-30 10:38:33 0 [ERROR] Aborting

2021-01-30 10:38:33 0 [ERROR] InnoDB: Cannot open '/mnt/HDD1/dietpi_userdata/mysql/ib_buffer_pool.incomplete' for writing: Permission denied
Warning: Memory not freed: 392
2021-01-30 23:45:20 0 [Note] InnoDB: Using Linux native AIO
2021-01-30 23:45:20 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2021-01-30 23:45:20 0 [Note] InnoDB: Uses event mutexes
2021-01-30 23:45:20 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-01-30 23:45:20 0 [Note] InnoDB: Number of pools: 1
2021-01-30 23:45:21 0 [Note] InnoDB: Using generic crc32 instructions
2021-01-30 23:45:21 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2021-01-30 23:45:21 0 [Note] InnoDB: Completed initialization of buffer pool
2021-01-30 23:45:21 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-01-30 23:45:21 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-01-30 23:45:21 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-01-30 23:45:21 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-01-30 23:45:21 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-01-30 23:45:21 0 [Note] InnoDB: 10.3.27 started; log sequence number 1602119; transaction id 8
2021-01-30 23:45:21 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/HDD1/dietpi_userdata/mysql/ib_buffer_pool
2021-01-30 23:45:21 0 [Note] InnoDB: Cannot open '/mnt/HDD1/dietpi_userdata/mysql/ib_buffer_pool' for reading: No such file or directory
2021-01-30 23:45:21 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-01-30 23:45:21 0 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2021-01-30 23:45:21 0 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2021-01-30 23:45:21 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-30 23:45:21 0 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist
2021-01-30 23:49:23 0 [Note] InnoDB: Using Linux native AIO
2021-01-30 23:49:23 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2021-01-30 23:49:23 0 [Note] InnoDB: Uses event mutexes
2021-01-30 23:49:23 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-01-30 23:49:23 0 [Note] InnoDB: Number of pools: 1
2021-01-30 23:49:23 0 [Note] InnoDB: Using generic crc32 instructions
2021-01-30 23:49:23 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2021-01-30 23:49:23 0 [Note] InnoDB: Completed initialization of buffer pool
2021-01-30 23:49:23 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-01-30 23:49:24 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-01-30 23:49:24 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-01-30 23:49:24 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-01-30 23:49:24 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-01-30 23:49:24 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-01-30 23:49:24 0 [Note] InnoDB: 10.3.27 started; log sequence number 1602119; transaction id 8
2021-01-30 23:49:24 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-01-30 23:49:24 0 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2021-01-30 23:49:24 0 [Note] Recovering after a crash using tc.log
2021-01-30 23:49:24 0 [Note] Starting crash recovery...
2021-01-30 23:49:24 0 [Note] Crash recovery finished.
2021-01-30 23:49:24 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/HDD1/dietpi_userdata/mysql/ib_buffer_pool
2021-01-30 23:49:24 0 [Note] InnoDB: Cannot open '/mnt/HDD1/dietpi_userdata/mysql/ib_buffer_pool' for reading: No such file or directory
2021-01-30 23:49:24 0 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2021-01-30 23:49:24 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-30 23:49:24 0 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist
2021-01-30 23:51:20 0 [Note] InnoDB: Using Linux native AIO
2021-01-30 23:51:20 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2021-01-30 23:51:20 0 [Note] InnoDB: Uses event mutexes
2021-01-30 23:51:20 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-01-30 23:51:20 0 [Note] InnoDB: Number of pools: 1
2021-01-30 23:51:20 0 [Note] InnoDB: Using generic crc32 instructions
2021-01-30 23:51:21 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2021-01-30 23:51:21 0 [Note] InnoDB: Completed initialization of buffer pool
[... cut ...]
2021-01-30 23:51:31 0 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2021-01-30 23:51:31 0 [Note] Recovering after a crash using tc.log
2021-01-30 23:51:31 0 [Note] Starting crash recovery...
2021-01-30 23:51:31 0 [Note] Crash recovery finished.
2021-01-30 23:51:31 0 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2021-01-30 23:51:31 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-30 23:51:31 0 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist

Joulinar, 2021-01-30:
@MichaIng 
I guess MariaDB would need to be reinstalled. It doesn't looks healthy. It's missing system tables. 🤔  What do you think.

MichaIng, 2021-01-31:
Indeed, the buffer ppol file is missing. What about the mysql internal database tables?
```
ls -l /mnt/dietpi_userdata/mysql/mysql
```

MichaIng, 2021-03-15:
Is this still an issue or can it be closed?

MichaIng, 2021-03-24:
Marking as closed. Feel free to reopen if required.

closed 2021-03-24 by MichaIng