MichaIng/DietPi #4054: Orange PI Image Request

retro24, 2021-01-16:
Hello is it possible to add an image for the Orange PI one ??

if you cannot do it, are there any instructions to do it my self ???

thanks

Joulinar, 2021-01-16:
Hi, 

many thanks for your message. You could try to install Armbian. Not sure what your device is. There a couple of them support. https://www.armbian.com/download/?tx_maker=xunlong

Once done, run DietPi PREP script to install DietPi on top https://dietpi.com/docs/dietpi_sbc/#make-your-own-distribution

MichaIng, 2021-05-07:
I'm gonna mark Orange Pi related image requests as closed for now. Referring to our preparation script guide: https://dietpi.com/docs/hardware/#make-your-own-distribution

Orange Pi is on my first priority list when image creation has been automated enough, for for now I personally cannot effort adding a new manufacturer to my creation schedule.

closed 2021-05-07 by MichaIng