MichaIng/DietPi #4055: [FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

nanyOynan, 2021-01-18:
#### Details:
- Date           | Mon Jan 18 09:43:06 -03 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Trying to install software from dietpi-software
2. Installing SONARR
#### Expected behaviour:

- Install Sonnar
#### Actual behaviour:

- Failed when APT-upgrade
- #### Extra details:

- ...
#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```

Joulinar, 2021-01-18:
Hi,

many hanks for your message. Did you tried running `dpkg --configure -a` ??

nanyOynan, 2021-01-18:
Thanks for the quick answer. Yes I did. 
It was taking too long so I submited the issue. 
It's working now.

closed 2021-01-18 by nanyOynan