MichaIng/DietPi #4056: Software | Yacht | Container Management UI

SelfhostedPro, 2021-01-19:
Creating a software request
Formal software information

* Software title | Yacht
* Short description | A web interface for managing docker containers with an emphasis on templating to provide 1 click deployments. Think of it like a decentralized app store for servers that anyone can make packages for. 
* Official URL (if available) | https://yacht.sh
* GitHub URL (if available) | https://github.com/selfhostedpro/yacht
    (Official) install documentation | https://yacht.sh/Installation/yacht/ (Additional ARM info here: https://yacht.sh/Installation/arm/ )

#### Are there similar/alternative software titles available with DietPi-Software?
Portainer

#### What makes your requested software better than the above solutions, if available?
Better docker-compose support (using docker-compose instead of libcompose), more templating options/multiple templates, theming, and a dashboard so you can see the resource usage of all containers at a glance.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

Install with:

```
docker volume create yacht_data
docker run -d -p 8000:8000 --name yacht --restart always -v /var/run/docker.sock:/var/run/docker.sock -v yacht_data:/config selfhostedpro/yacht

```
Additional theming/whitelabel support with `-E LOGO=<url for logo image>` and `-E THEME=<theme name> ` List of themes [here](https://github.com/SelfhostedPro/Yacht/blob/master/frontend/src/plugins/vuetify.js). If you would like to provide colors for a custom diet-pi theme I would happily add it in for you, just open an issue and I'll get it setup ASAP.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
Install docker if it's not installed already.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? Yes, I could provide a PR if you provide me with the software id you'd like me to use. I'd just be copying this for the most part: 

https://github.com/MichaIng/DietPi/pull/3933 

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/
https://feathub.com/MichaIng/DietPi/+206

Joulinar, 2021-01-19:
Hi,

many thanks for your message. That similar to Portainer 😃

SelfhostedPro, 2021-01-19:
Yeah, it's similar but more focused on small businesses/selfhosting than on the enterprise side of things. It should be very similar as far as adding it to DietPi. If there's anything I can do on my end please feel free to let me know. I'm also happy to see if there are any integrations the community would like added as I'm the developer.

jodfie, 2021-04-05:
i really would like this added!

uniphonic, 2022-06-09:
I'd love to see Yacht added to DietPi too! 

I also wonder though, how Yacht would run as a Docker container through Portainer onto DietPi? Even if that's a viable option though, I still would love to see Yacht available directly in DietPi.

Thanks!

Joulinar, 2022-06-09:
> I also wonder though, how Yacht would run as a Docker container through Portainer onto DietPi?

Basically Portainer as well as Yacht are Docker container itself. Both are somehow equal as they allow managing docker container via a web interface. Portainer is not running any container by it's own. It's just a tool to manage the docker engine.

madmedix, 2023-03-24:
the beginning of the end for Portainer. The inevitable grab has started with castrating the ce version and obnoxious nagware plowed into the code. Yacht please.