MichaIng/DietPi #4057: Network is unreachable using automated installation on Rockpro64

cmoulliard, 2021-01-19:
### issue

The network cannot be configured using either the Ethernet or Wifi adapter. During the process which configures automatically `DietPi` using the `dietpi.txt` and `dietpi-wifi.txt` files, the following bluescreen is generated

[image]

#### Required Information
- DietPi version : `6.34.3` released today
- SBC model : `RockPro64`
- Linux Kernel: `5.9.16-rockship64`

#### Steps to reproduce
1. Download the latest image released `DietPi_ROCKPro64-ARMv8-Buster.img` and flash it to a SD card
2. Modify the `dietpi.txt` using these parameters
```
AUTO_SETUP_ACCEPT_LICENSE=1
AUTO_SETUP_LOCALE=en_US.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=gb
AUTO_SETUP_TIMEZONE=Europe/Brussels

AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=1

AUTO_SETUP_NET_ETH_FORCE_SPEED=0
AUTO_SETUP_NET_WIFI_COUNTRY_CODE=BE

AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.1.200
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.1.1
AUTO_SETUP_NET_STATIC_DNS=8.8.8.8

# Hostname
AUTO_SETUP_NET_HOSTNAME=DietPi
AUTO_SETUP_SWAPFILE_SIZE=0
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
AUTO_SETUP_HEADLESS=0
AUTO_UNMASK_LOGIND=0
AUTO_SETUP_CUSTOM_SCRIPT_EXEC=0
AUTO_SETUP_SSH_SERVER_INDEX=-1
AUTO_SETUP_FILE_SERVER_INDEX=0
AUTO_SETUP_LOGGING_INDEX=-1
AUTO_SETUP_RAMLOG_MAXSIZE=50
AUTO_SETUP_WEB_SERVER_INDEX=-2
AUTO_SETUP_AUTOSTART_TARGET_INDEX=0
AUTO_SETUP_AUTOSTART_LOGIN_USER=root
AUTO_SETUP_AUTOMATED=1
...
CONFIG_BOOT_WAIT_FOR_NETWORK=1
```
File used is available here: https://github.com/ch007m/home-devices/blob/master/config/dietpi.txt

4. Create a `dietpi-wifi.txt` file containing the config of the local Wifi SSID
```
aWIFI_SSID[0]='WiFi-2.4-0290_2GEXT'
aWIFI_KEY[0]='wifi_pwd'
aWIFI_KEYMGR[0]='WPA-PSK'
```
5. Copy the files to the SD card --> `/boot/`
6. Plug the SDCard, Wifi external USB module, RJ45 cable and HDMI cable and power the SBC

Joulinar, 2021-01-19:
Hi,

many thanks for your message. Some comments from my side.

> AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=1

I would recommend to use Ethernet or WiFi to complete the setup but not both at the same time

> AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.1.200
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.1.1
AUTO_SETUP_NET_STATIC_DNS=8.8.8.8

If you like to use static IP, you would need to set it to `1`. Otherwise your system will try to get an IP address using DHCP

> Wifi external USB module, RJ45 cable
As above, try to use either Ethernet or WiFi

cmoulliard, 2021-01-19:
> If you like to use static IP, you would need to set it to `1`.

Will the config of the static IP work too for the Wifi connection ?

cmoulliard, 2021-01-19:
> I would recommend to use Ethernet or WiFi to complete the setup but not both at the same time

I suggest then to review the comment of the existing file `# - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled.` as it it not mentioned that we cannot use both adapters

Joulinar, 2021-01-19:
you can use both. But I would not recommend it to have it connected to same network at the same moment.

cmoulliard, 2021-01-19:
I did the changes as you suggested. Network has been configured using Ethernet adapter - RJ45 and dietpi is now installing the list of the defined software :-)

cmoulliard, 2021-01-20:
This is strange as the password `dietpi` cannot be used anymore to logon

```
sshpass -p "dietpi" ssh root@192.168.1.200
Permission denied, please try again.
```

Joulinar, 2021-01-20:
This setting for `AUTO_SETUP_GLOBAL_PASSWORD` is correct as the value is used during initial first boot only.

MichaIng, 2021-01-22:
> Permission denied, please try again.

This does not (necessarily) mean that the password is wrong, but probably the authentication method or peer. Did you change something on the SSH server settings and did you try it from a different client, like plain interactive `ssh` or PuTTY from a Windows system?

Of course the easiest way to verify the login password is using the `login` command or local console login.

cmoulliard, 2021-01-22:
> Did you change something on the SSH server settings

No

cmoulliard, 2021-01-22:
> did you try it from a different client

No. Only ssh in a terminal

MichaIng, 2021-01-22:
This means you are locked out from this system currently? `dietpi-config` can be used to change the password, but you need to have console access of course. Else this method would work (credits to @Joulinar for finding this related to another report): https://howtoraspberrypi.com/recover-password-raspberry-pi/
_Replace "pi" user with "root"._

However, I still don't think that the password is the issue, but the SSH client, since really nothing is touching the login passwords aside of the `AUTO_SETUP_GLOBAL_PASSWORD` setting on very first boot only.

cmoulliard, 2021-01-26:
I re-installed everything from a new microSD card where DietPi was mounted and I cannot reproduce this problem anymore

closed 2021-01-26 by cmoulliard

MichaIng, 2021-01-26:
One thing I found a potential problem for causing wrong passwords is:
- When the password has been applied on first run setup, in `dietpi.txt` setting is replaced with an informational sentence that the password has been stored encrypted to disk, for use by `dietpi-software` as default for software installs. So far so good.
- But if one now takes the exact `dietpi.txt` from one system and copies it to a new system, that informational sentence is detected and taken as login and software password.
- So we need to comment that setting and let first run setup fallback to the default "dietpi", if no non-commented password setting has been found in `dietpi.txt`.