MichaIng/DietPi #4058: Cannot configure wifi

cmoulliard, 2021-01-20:
### Creating a bug report/issue

I cannot select the Wifi slot0 (as defined within the dietpi-wifi.txt file) using dietpi-config. When I select the slot0, then I see the message `remove ....` and no wifi SSI is then selected

[image]

**Remark**: After click and many buttons, the slot0 hs been configured but the Wifi adapter has not been enabled as the IP address assigned by the DHCP server to the `wlan0` adapter concerns the Eth adapter

```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.200  netmask 255.255.255.0  broadcast 192.168.1.255
        ether 12:eb:64:62:74:e2  txqueuelen 1000  (Ethernet)
        RX packets 1398  bytes 132367 (129.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1021  bytes 385269 (376.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 27

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 243  bytes 22024 (21.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 243  bytes 22024 (21.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 2312
        inet 192.168.1.53  netmask 255.255.255.0  broadcast 192.168.1.255
        ether d8:50:e6:f3:96:c7  txqueuelen 1000  (Ethernet)
        RX packets 3802  bytes 582739 (569.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 92  bytes 15250 (14.8 KiB)
        TX errors 0  dropped 6 overruns 0  carrier 0  collisions 0
```

#### Required Information
- DietPi version
  ```
  G_DIETPI_VERSION_CORE=6
  G_DIETPI_VERSION_SUB=34
  G_DIETPI_VERSION_RC=3
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  ```
- Distro version | `10.7`
- Kernel version | `Linux DietPi 5.9.16-rockchip64 #trunk.33 SMP PREEMPT Tue Dec 22 15:51:09 CET 2020 aarch64 GNU/Linux`
- SBC model | `ROCKPro64 (aarch64)`

Joulinar, 2021-01-20:
Hi,

best to my knowledge, this means these WiFi details are already selected on slot 0. But @MichaIng could explain it better than I 😃

cmoulliard, 2021-01-20:
**Remark**: We need a doc page explaining how to setup the Wifi network and/or Ethernet or both ;-)

MichaIng, 2021-01-22:
I'm reworking the whole network setup anyway, and it's possible to chance SSID and password and also WPA2/EAP switch and settings individually. I also found it a bit counter-intuitive that the only possibility to alter a slot currently is to remove and re-add it.

But please be partitioned, it's a larger change with a rats tail behind it that needs to be taken care of as well.

closed 2021-05-07 by MichaIng