MichaIng/DietPi #4059: Audio problems on RPI4B aarch64

k1r0s, 2021-01-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `DietPi v6.34.3`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux`
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | transformer 5V 3A
- SDcard used | x, booted from USB3.0 15GB stick

#### Additional Information (if applicable)
- fresh install, only LXDE, alsamixer, and X server.
- Bug report ID | 1781f433-2dfb-4442-b2f9-5e903b853638

#### Steps to reproduce
- burn the image
- boot
- install previously mentioned software
- open any mp3 file with FF esr or any youtube tab
outcome: no audio

... I've been investigating a bit however my linux knowledge is kinda limited

apparently alsamixer can't select the proper audio channel. or soundcard

dietpi-justboom ALSA output info does not show any output at all: 

```
 Mode: ALSA Output Info

closed
closed
closed

[  OK  ] DietPi-JustBoom | Completed, press any key to exit...
```
I keep receiving this error on dmesg: 

`[ 2803.309903] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/mailbox@7e00b840/bcm2835_audio/status`

:\ does anyone can help me through this debug process?

C0D3-M4513R, 2021-01-20:
Is the audio card getting detected?
Run the following commands, and give the output please.
`cat /proc/asound/cards`
`amixer`
`amixer contents`

k1r0s, 2021-01-21:
Thanks for the response @C0D3-M4513R 

```sh
dietpi@DietPi:~$ cat /proc/asound/cards
 0 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 1 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1

```

```sh
dietpi@DietPi:~$ amixer
(empty)
```
```sh
dietpi@DietPi:~$ amixer contents
numid=2,iface=MIXER,name='IEC958 Playback Default'
  ; type=IEC958,access=rw------,values=1
  : values=[AES0=0x04 AES1=0x82 AES2=0x00 AES3=0x02]
numid=3,iface=MIXER,name='IEC958 Playback Mask'
  ; type=IEC958,access=rw------,values=1
  : values=[AES0=0xff AES1=0xff AES2=0xff AES3=0xff]
numid=1,iface=PCM,name='ELD'
  ; type=BYTES,access=r--v----,values=128
  : values=0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00
numid=4,iface=PCM,name='Playback Channel Map'
  ; type=INTEGER,access=r----R--,values=0,min=0,max=36,step=0
amixer: Control default element read error: Invalid argument

```

C0D3-M4513R, 2021-01-22:
Do you have any audio plugged in to your hdmi monitor? 

If you want to use onboard audio, you should configure the sound card in dietpi-config respectively.

k1r0s, 2021-01-22:
I've tried to setup through dietpi-config many times with no success. My monitor has a 3,5mm jack. I have sound, for instance, with my odroidc2 (yes, with dietpi)

MichaIng, 2021-01-22:
There are two issues:
1. As C0D3-M4513R mentioned you seem to not have an available sound card selected, or you did not yet reboot, else at least one additional sound card would be listed on `/proc/asound/cards`.
2. You have the RPi 4 KMS GL driver enabled, which adds additional sound cards, although I have no idea if/how those work. This is known from other RPi models as well and when selecting sound cards while this driver is enabled, we increment the index by 1 to match the correct card. On RPi 4, since there are two HDMI ports, two additional sound cards are added, so the index needs to be incremented by 2.

Also I see we should show an info that a reboot is required when RPi onboard audio was newly selected.

So do the following:
1. `dietpi-config` > `Audio Options` > `Sound card` > select the one you want to use, if not yet done, then reboot.
2. Edit `/etc/asound.conf` and increment all card indices by 1, e.g. `card 1` => `card 2` or `card 2` => `card 3`

MichaIng, 2021-01-22:
This should fix the kernel message you faced and fixes sound card selection on RPi 4 with the full KMS driver enabled: https://github.com/MichaIng/DietPi/commit/317c35d9fe69f8105b3cab18a53b392797c8a9cd
And this suggests a reboot after doing sound card selections with RPi: https://github.com/MichaIng/DietPi/commit/45ac1d93ae2fa56d510abcbee54814e558df9d89

k1r0s, 2021-01-22:
okay so i've restored /etc/asound.conf

and applied `317c35d`

rebooted the RPI

the headphones where connected to the monitor and I had no sound, but i realized that the browser was in fact outputting audio so I've switched to the rpi 3,5mm jack and voila!

I guess my monitor does not tell the rasp that an audio plug is connected so the driver thinks that no sound is available through hdmi? :(

`$ cat /proc/asound/cards`
```
 0 [Headphones     ]: bcm2835_headphonbcm2835 Headphones - bcm2835 Headphones
                      bcm2835 Headphones
 1 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 2 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1
```

thanks for helping me!!

closed 2021-01-22 by k1r0s

MichaIng, 2021-01-22:
Ah, and very interesting now is your `/proc/asound/cards` output since the VC4 driver injected sound cards are not before but behind the 3.5mm (headphones) sound card! Looks like this has been changed with newest 5.10 kernel/firmware.

Looks like we need to make this sound card index increment based on kernel version until 5.10 has been released for 32-bit Raspbian... And the question is now if attached USB DACs are located before or behind the VC4 cards then...

About your HDMI audio. The [above](https://github.com/MichaIng/DietPi/issues/4059#issuecomment-764599639) `/proc/asound/cards` output that contains only the two vc4hdmi cards was after you selected `rpi-bcm2835-auto` or `rpi-bcm2835-hdmi` in dietpi-config and did a reboot? Probably it was and is still only the wrong sound card index. The real HDMI sound card(s) in `/proc/asound/cards` are named "HDMI 0" respectively "HDMI 1" without "vc4".