MichaIng/DietPi #4060: Bluetooth doesn't enable

C0D3-M4513R, 2021-01-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v6.34.3 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 5`
- Distro version | `ebuster`
- Kernel version | `Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | Official 5v3a Pi psu
- SDcard used | SanDisk ulta, I think

#### Additional Information (if applicable)
- Software title | N/A
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `75437122-e1e9-436c-9a03-230366fd58c2`

#### Steps to reproduce

1. go to `dietpi-config`-> Advanced-> Bluetooth
2. Try to toggle to on

#### Expected behaviour

Bluetooth should work

#### Actual behaviour

Some error is popping up

#### Extra details

```
Job for hciuart.service failed because the control process exited with error code.
See "systemctl status hciuart.service" and "journalctl -xe" for details.
```
- Date           | Wed Jan 20 19:36:22 CET 2021
- Bug report     | 75437122-e1e9-436c-9a03-230366fd58c2
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | systemctl enable --now hciuart
- Exit code      | 1
- Software title | DietPi-Set_hardware
```
root@DietPi:~# systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2021-01-20 19:36:22 CET; 9min ago
  Process: 2435 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Jan 20 19:35:52 DietPi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jan 20 19:36:22 DietPi btuart[2435]: Initialization timed out.
Jan 20 19:36:22 DietPi btuart[2435]: bcm43xx_init
Jan 20 19:36:22 DietPi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Jan 20 19:36:22 DietPi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Jan 20 19:36:22 DietPi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
root@DietPi:~# journalctl -u hciuart
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Wed 2021-01-20 19:46:15 CET. --
Jan 20 19:29:15 DietPi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jan 20 19:29:22 DietPi btuart[290]: Can't set line discipline: Invalid argument
Jan 20 19:29:22 DietPi btuart[290]: Can't initialize device: Invalid argument
Jan 20 19:29:22 DietPi btuart[290]: bcm43xx_init
Jan 20 19:29:22 DietPi btuart[290]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Jan 20 19:29:22 DietPi btuart[290]: Set Controller UART speed to 3000000 bit/s
Jan 20 19:29:22 DietPi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Jan 20 19:29:22 DietPi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Jan 20 19:29:22 DietPi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Jan 20 19:30:30 DietPi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jan 20 19:31:00 DietPi btuart[1880]: Initialization timed out.
Jan 20 19:31:00 DietPi btuart[1880]: bcm43xx_init
Jan 20 19:31:00 DietPi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Jan 20 19:31:00 DietPi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Jan 20 19:31:00 DietPi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Jan 20 19:35:52 DietPi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jan 20 19:36:22 DietPi btuart[2435]: Initialization timed out.
Jan 20 19:36:22 DietPi btuart[2435]: bcm43xx_init
Jan 20 19:36:22 DietPi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Jan 20 19:36:22 DietPi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Jan 20 19:36:22 DietPi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART. 
```
#3018

C0D3-M4513R, 2021-01-21:
Ok,  dunno what I did wrong. After a while (and a reboot) it fixed itself? I don't know, what fixed it. Will keep it open, in case it breaks again at the next reboot. If there is no update in the future, close this.

closed 2021-01-22 by C0D3-M4513R

MichaIng, 2021-01-22:
We need to remove the forced start of hciuart service. If Bluetooth was disabled before, this is doomed to fail as the Bluetooth-UART device is not available in the device tree until a reboot.

MichaIng, 2021-01-22:
Forced service start removed: https://github.com/MichaIng/DietPi/commit/ad559c37c83b6f8b6d2e72f731985ed61de81c1f
Reboot prompt appears now in dietpi-config when Bluetooth is enabled on RPi with onboard Bluetooth adapter: https://github.com/MichaIng/DietPi/commit/52c348eb73f5781f06a96918f5f134f2d1ed9876