MichaIng/DietPi #4061: Unable to install Pi-Hole

theChef613, 2021-01-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Samsung 2A Power supply
- SDcard used | 8 Gig Sandisk

#### Additional Information (if applicable)
- Software title | pi-hole and papermc
- Was the software title installed freshly or updated/migrated? Trying to install for the first time
- Can this issue be replicated on a fresh installation of DietPi? unsure

- Bug report ID |  ce9ed080-8f5e-40a9-a39b-906b12ea4cc1

#### Steps to reproduce

1. use dietpi-software to install pi-hole

#### Expected behaviour

pi-hole (or papermc) should install

#### Actual behaviour

- encounters error with exit code 28 on command `curl -ILfvm 10 https://install.pi-hole.net`

#### Extra details
I was able to install MineOS just fine. I made sure it's connected to the internet with the "Test Connection" and it is successful.

Part of the error log below:

 Error log:
│ * Expire in 0 ms for 6 (transfer 0x5585517c10)
│ * Expire in 10000 ms for 8 (transfer 0x5585517c10)
│ * Expire in 1 ms for 1 (transfer 0x5585517c10)
│   % Total    % Received % Xferd  Average Speed   Time    Time     Time
│ Current
│                                  Dload  Upload   Total   Spent    Left
│ Speed
│ ^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--
│ 0* Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 1 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 2 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)
│ * Expire in 0 ms for 1 (transfer 0x5585517c10)

Joulinar, 2021-01-22:
Hi,

Many thanks for your message. Maybe a temporary issue to download Pihole install script. Can you run manually `curl -ILfvm 10 https://install.pi-hole.net`

theChef613, 2021-01-22:
> Hi,
> 
> Many thanks for your message. Maybe a temporary issue to download Pihole install script. Can you run manually `curl -ILfvm 10 https://install.pi-hole.net`

Thanks for your help!
I tried that and this is the response I get..
Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jan 22 07:17:30 2021 from 192.168.10.102

 ─────────────────────────────────────────────────────
 DietPi v6.34.3 : 08:05 - Fri 01/22/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (aarch64)
 - CPU temp : 55'C : 131'F (Running warm, but safe)
 - LAN IP : 192.168.10.110 (wlan0)
curl: (28) Resolving timed out after 2000 milliseconds
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Raspberry Pi OS 64-bit)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# curl -ILfvm 10 https://install.pi-hole.net
* Expire in 0 ms for 6 (transfer 0x55ae2fcc10)
* Expire in 10000 ms for 8 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 1 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 2 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 0 ms for 1 (transfer 0x55ae2fcc10)
* Expire in 2 ms for 1 (transfer 0x55ae2fcc10)
[... 641 lines cut ...]
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=www.github.com
*  start date: May  6 00:00:00 2020 GMT
*  expire date: Apr 14 12:00:00 2022 GMT
*  subjectAltName: host "raw.githubusercontent.com" matched cert's "*.githubusercontent.com"
*  issuer: C=US; O=DigiCert Inc; OU=www.digicert.com; CN=DigiCert SHA2 High Assurance Server CA
*  SSL certificate verify ok.
* Operation timed out after 5745 milliseconds with 0 out of 0 bytes received
* Closing connection 1
curl: (28) Operation timed out after 5745 milliseconds with 0 out of 0 bytes received
root@DietPi:~#

Joulinar, 2021-01-22:
ok looks like there was a temp issue at PiHole side. I had similar behaviour during my test but it is gone now. At least I could complete a test installation. can you try again be running `dietpi-software install 93`

theChef613, 2021-01-22:
> ok looks like there was a temp issue at PiHole side. I had similar behaviour during my test but it is gone now. At least I could complete a test installation. can you try again be running `dietpi-software install 93`

Hello,
I tried again with both the command you suggested and going through dietpi-software and I get the same error.. :(

Joulinar, 2021-01-22:
hmm I did another test and for me this is working without issues. Are you able to open https://install.pi-hole.net on a browser on your computer? It should display the PiHole install script code

theChef613, 2021-01-22:
> 
> 
> hmm I did another test and for me this is working without issues. Are you able to open https://install.pi-hole.net on a browser on your computer? It should display the PiHole install script code

thank you again for your help. I am able to get to that website from my computer and just to double-check I used "wget" on my dietpi to download it too which worked fine. not sure if this will help but I also noticed recently that the following message shows up right below the MOTD when I log into dietpi: `curl: (28) Resolving timed out after 2000 milliseconds` 
this is weirdly similar to the error that I've been getting. maybe there is something going on with the curl command? I really don't know just trying to help and give you as much info as I can.

Joulinar, 2021-01-22:
well there is a way to change from `curl` to `wget` but this would require to change 1 line of code inside `dietpi-software` script

https://github.com/MichaIng/DietPi/blob/b354689141bf3b238b87d9ff83f33df5cf52173f/dietpi/dietpi-software#L4210

`nano /boot/dietpi/dietpi-software`

now change
`curl -sSfL "$INSTALL_URL_ADDRESS" -o install.sh` into `wget -O install.sh https://install.pi-hole.net`

save the file and run installation again

MichaIng, 2021-01-22:
> curl: (28) Resolving timed out after 2000 milliseconds

This is btw not uncommon since we limit connections to two seconds for the banner, so that login does not hang too long. Still an outstanding task to optionally receive those info via background jobs instead so that the banner script itself does not need to perform any network connection.
For the download of the Pi-hole installer the timeout is longer, 5 seconds is default on GNU/Linux systems, AFAIK.

Joulinar, 2021-01-22:
@MichaIng 
On PiHole is using as well `wget` to download install script manually. https://github.com/pi-hole/pi-hole/#method-2-manually-download-the-installer-and-run

Should we probably adjust?

MichaIng, 2021-01-22:
I just migrated all `wget` calls to `curl` calls, since curl is slightly more reliable (from my experience) and a little bit faster (not the download, just the naked command execution) and it can upload and verbose connection debug as well. IMO it does not make sense to replace single downloads with wget, either curl works better, then we use it everywhere, or curl does, then we leave it as it is 😉. At least there doesn't seem to be a correlation between the reliability of the download tool and the remote host.

@theChef613 did you try to increase the connection test timeout via `dietpi-config` > `Network Options: Misc` > `G_CHECK_URL Timeout` to e.g. 20 seconds?

Btw, the connection check failed, not the script download itself, so when you want to instead test the install with wget, replace:
```
curl -ILfvm 10 https://install.pi-hole.net
```
with
```
wget --spider https://install.pi-hole.net
```
and in case the script download itself fails as well (although there is no limited timeout set for this), the command as mentioned by Joulinar: https://github.com/MichaIng/DietPi/issues/4061#issuecomment-765499856

theChef613, 2021-01-24:
Hello,
thanks for all the help so far. I ended up reinstalling a fresh image of dietpi and that didn't even help. I also tried the network options mentioned above, didn't help.. 
what ended up working was subituing the 'wget' command for the  'curl' command as described above.

MichaIng, 2021-01-24:
Okay, good that it worked that way. To get some more info around your issue, I saw that the manual curl call above as well timed out on DNS resolution after default 5 seconds. Could you check how long wget actually needed to resolve the hostname?
```
time wget --spider https://install.pi-hole.net
```
In case it's also a few seconds, which DNS server are you using?
```
cat /etc/resolv.conf
```

theChef613, 2021-01-24:
> Okay, good that it worked that way. To get some more info around your issue, I saw that the manual curl call above as well timed out on DNS resolution after default 5 seconds. Could you check how [...]

hi. thank you again for helping resolve this issue. really great to see you digging deeper to help others :)
using that command i get the folloing results:
real    0m6.858s
user    0m0.033s
sys     0m0.025s

The DNS server I'm using is 1.1.1.1 (cloudflare i think?)

Joulinar, 2021-01-24:
nearly 7 seconds, that's quite long. for me it's 1 seconds

```
real    0m1.039s
user    0m0.055s
sys     0m0.018s
```

can you try changing DNS to `9.9.9.9` (Quad9) or `8.8.8.8` (Google) and repeat the test

MichaIng, 2021-01-24:
Okay, nearly 7 seconds, that is veeery long or DNS resolution 😉. Yes 1.1.1.1 is Cloudflare, which is usually the fastest average server.

Yes as Joulinar suggested, probably a different DNS server will work better in your case, but it would bet interesting as well if other hostnames resolve similarly slow or if this is install.pi-hole.net only, e.g.:
```
time wget --spider https://dietpi.com
```

Joulinar, 2021-01-24:
@MichaIng 
my personal feeling regarding Cloudflare change over the last weeks/month. At least my Pihole installation prefers Quade9 and Google. Cloudflare is used on rar cases only 🤔 

[image]

theChef613, 2021-01-24:
i just realized another factor could be that my raspberry pi is actually connected to a wifi extender, not the router itself? I'll try changing the dns server tonight when there arent so many people in the family using the internet.

Joulinar, 2021-01-24:
Usually the DNS server of the DietPi device should not interfere with the local network. At least it depends where you are going to change the DNS server 🤔

MichaIng, 2021-01-24:
Yes, what you change via `/etc/resolv.conf` or via `dietpi-config` > `Network Options: Adapters` > `WiFi` > `Static DNS` affects the DietPi system only, nothing else. I'd btw not recommend to use Pi-hole as DNS resolver for the hosting server itself, so that it is independent from it and Pi-hole can be stopped and updated without breaking the servers own connection abilities. And usually you don't use the server for browsing where blocking ads would be important.

Also WiFi connection should never lead to a connecting time in the seconds area. Furthermore, if this server should be DNS and in case DHCP for your home network, then + >5 seconds delay for every connection that through that Pi-hole won't be of joy, at least if clients are indeed affected as well.

theChef613, 2021-01-25:
I changed the static dns on my dietpi server to something besides my router and its now much quicker!
`real    0m0.649s
user    0m0.036s
sys     0m0.009s
`
i think my problem was that i was routing my dns though my wifi extender, which went to the router, then to the dns server.

theChef613, 2021-01-25:
so... I just tried to install `papermc `as a test as it was one of the ones that i couldn't install and I'm actually getting the same error again! DX ...
it tried 5 times and i have it set to 60 seconds. but all 5 attempts timed out much before that.

Joulinar, 2021-01-25:
are you able to share the full error message regarding papermc? We are working on couple of enhancement for next release https://github.com/MichaIng/DietPi/pull/4039

MichaIng, 2021-01-25:
No need to check, is it? Since PaperMC install is indeed currently broken: https://github.com/MichaIng/DietPi/pull/4039
They changed download URLs a few days after we released our implementation 😅.

Joulinar, 2021-02-08:
Hi guys,

can this issue be closed? I guess initial issue was fixed. Correct?

MichaIng, 2021-02-08:
I think so. The initial connection issue with Pi-hole remains unsolved, but it looks like one of these temporary connection issues. The PaperMC case is solved with v6.35. I'll close those. But feel free to reopen if required.

closed 2021-02-08 by MichaIng