MichaIng/DietPi #4062: Odroid HC4 | Dietpi-drive_manager / automount at boot stalls trying to mount btrfs partition on hdd

eyduh, 2021-01-22:
#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version |
Linux adacloud 4.9.241-arm64

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Ocroid HC4 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
official odroid hc4
- SDcard used | (EG: SanDisk ultra)
WD Green SSD 120gb
```
/dev/sda1 /boot fat
/dev/sda2 /rootfs ext4
/dev/sda3 /nextcloud_database & dietpi_userdata btrfs
```

- WD Green HDD 3tb
`/dev/sdb1 /nextcloud_data btrfs` or 
`/dev/sdb` depending on how I created the btrfs volume

#### Additional Information
I flashed dietpi & used the nextcloudpi install script, once nextcloud was up and running, I mkfs.btrfs'd sda3 to btrfs then migrated the nextcloud database to this partition. so far so good. reboot worked. While doing this I did a burn in of my hdd to check for errors on another system, using `badblocks -v 4096 -wsv` which took quite some time (note to self: never do this with a sata/usb dock, always plug it straight into your machine with a sata cable). Anyway, it finished with no errors so I went ahead to get the nextcloud data onto a btrfs drive to be able to use ncp brtfs snapshots and this is where it started going south.

It was fine at first but whenever I would reboot the automount/dietpi-drive_manager script would stall on mounting the btrfs hdd.
If I tried to go into the drive_manager, it would stall on "detecting drive /dev/sdb1" and eventually my ssh session would be terminated with "Broken pipe: client_loop". I would try the same on a local session and it would just sit there forever.

I've tried making the btrfs hdd with `mkfs.btrfs`, dietpi-drive_manager, fdisk, parted (on local machine) and kde partition manager on my desktop and I always end up with the same situation.

The reason I want to use btrfs is for the snapshot features in nextcloudpi and eventually to boot the system off a usb->sata for the ssd and add another 3tb hdd for data parity.

At some point I got a "ioctl" error which from some searching seems to point to a mismatch between btrfs-progs and the btrfs supported version in the kernel.

- Can this issue be replicated on a fresh installation of DietPi?
yes, this was on a fresh install.
I've managed to replicate this behaviour by formatting the btrfs drives in four ways:
On the Odroid HC4 with dietpi-drive_manager
On the Odroid HC4 with mkfs.btrfs
On the Odroid HC4 with fdisk
On Kubuntu 20.10 desktop with kde partition manager

- Bug report ID | 
a762da56-beb5-4d98-be27-1003b523c583
#### Steps to reproduce

1. install dietpi & format partition & drive to btrfs using any tool
2. reboot

#### Expected behaviour

- btrfs partitions mount cleanly and allows for nextcloud to read the database & data folders instead these steps in the boot sequence stall:
[image]

#### Actual behaviour

- dietpi-drive_manager stalls trying to mount one/both of btrfs partitions

#### Extra details

- I'm using the nextcloudpi install script and not the dietpi nextcloud install, which should be fine as this board is dedicated to nextcloudpi. the ncp odroid hc4 image wouldn't boot from ssd and since I've had nice experience with dietpi on other systems I went for the dietpi+ncp route knowing they're not fully integrated but should work together as long as the ncp setup doesn't clash with any webstacks on dietpi.

- The only other things I have on the system is influxdb & glances installed while sick of trying to make this work and as a way to monitor the system remotely, as well as zerotier. 

- Ideally I would like to have this setup with ncp for the snapshots on a dietpi base for easy of updates and influxdb and backup. At the moment I'm leaning towards scrapping the idea of ncp-snapshots/btrfs as btrfs seems to be a bit buggy and only use the dietpi nextcloud+talk install with the data-drive on LVMraid for a future data-parity setup or zfs.

I'm not sure which logs to attach, I had a look in /var/logs but couldn't find anything relevant as well as in /var/tmp so please let me know.

MichaIng, 2021-01-22:
Many thanks for your report.

Why is `dietpi-drive_manager` actually running on boot? It should not. Or do you mean when you call it manually to format and mount the Btrfs drive, it hangs?

Mounting on boot is done by systemd via native GNU/Linux commands, `dietpi-drive_manager` is not involved in this, but creates the `/etc/fstab` entries only which are used by systemd then. External drives are added with the `x-systemd-automount` options, so that those are not mounted directly at boot, but instead once the first access to a file within the mount point is done. The mounting itself is nearly immediately, so neither you, not any process should recognise the delay. This is often required for external drives since they might not be fully initialised fast enough at boot to be mounted directly.

eyduh, 2021-01-22:
Hmm, I rebooted without the spinny disk and there was no drive manger present in htop. I must've called it manually last time, during which it always get stuck like this:
[image]
This is booting without the spinny disk and it's getting stuck on the btrfs partition.

When I boot with the spinny disk, the HDMI output of the odroid sits on the blinking cursor forever after progressing from the bootloader. During this time I can ping it and see it the network but not ssh into it. Eventually (a veeery long eventually) I hear the spinny disk react and it progresses to login (set to automatic login terminal). In fstab I can see the spinny disk entry commented out with the same mount options as the SSD btrfs partition that does work sometimes.

If I mount the drives manually with mount -t btrfs /dev/sdX /mnt/mountpoint they kinda of work, but, drive manager still see them as unmounted at first and either hangs when it realises they're actually mounted or throws an error trying to mount them.

If I boot without the spinny disk it boots up and into dietpi fine.
At this stage it gets stuck on postboot/diet-pi-services as the mariadb mysql is on a btrfs that doesnt get mounted properly.

MichaIng, 2021-01-22:
Okay, so one of the drive detection commands hangs then, let's see which one it is:
```
blkid
lsblk
```
When I try it here, `blkid` wakes up the drive to gather information, while `lsblk` does not but is able to give even more information. Not sure why we use `blkid`, the only notable difference I see is that `blkid` only lists drives with an actual file system on them, ready to be mounted, while `lsblk` shoes all block devices, including parents (/dev/sda for /dev/sda1) and unformatted ones, which we currently detect in a separate loop.

eyduh, 2021-01-23:
I'm sorry for being unclear.
The Odroid HC4 has two SATA slots, one with an SSD and one with a HDD:
[image]
```
# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 111.8G  0 disk 
├─sda1   8:1    0   256M  0 part /boot
├─sda2   8:2    0  11.9G  0 part /
└─sda3   8:3    0  99.6G  0 part /mnt/26d02176-cc95-45e8-af6b-333f96b4d328
sdb      8:16   0   2.7T  0 disk 
└─sdb1   8:17   0   2.7T  0 part
```
```
# blkid
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="BOOT" LABEL="BOOT" UUID="3086-D853" TYPE="vfat" PARTUUID="03823826-01"
/dev/sda2: LABEL="rootfs" UUID="e139ce78-9841-40fe-8823-96a304a09859" TYPE="ext4" PARTUUID="03823826-02"
/dev/sda3: LABEL="dbs" UUID="26d02176-cc95-45e8-af6b-333f96b4d328" UUID_SUB="de1722db-ea9c-4101-bf2c-9805772bf2f8" TYPE="btrfs" PARTUUID="03823826-03"
/dev/sdb1: UUID="b68d8d79-3880-4574-b3dc-24dfc9a049fc" UUID_SUB="972e312b-f513-4800-a668-df3858aab589" TYPE="btrfs" PARTLABEL="primary" PARTUUID="dd9a7247-9425-4080-acf1-905f6f562546"
```
It generally won't boot with the HDD (sdb) slotted in, but if I boot with just the SSD it will work. 

I just rebooted without the HDD and hot-plugged it after boot and now fstab does not have the line for the HDD:
```
# Please use "dietpi-drive_manager" to setup mounts
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1856M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAPFILE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / ext4 noatime,lazytime,rw 0 1
UUID=3086-D853 /boot vfat noatime,lazytime,rw 0 2
UUID=26d02176-cc95-45e8-af6b-333f96b4d328 /mnt/26d02176-cc95-45e8-af6b-333f96b4d328 btrfs noatime,lazytime,rw,nofail,noauto,x-systemd.automount
```
I think last time I had rebooted the system after it freezing trying to mount /dev/sdb, then getting stuck again at the blinky cursor. So I shut it down, plugged root into my desktop and commented out the HDD line in fstab and tried again. That time it booted after a long time.

Anyway, so according to lsblk or fdisk or fstab, /dev/sda3 is mounted on /mnt/[UUID], but then:
[image]
dietpi-drive_manager says it's not, and stalls. _Sometimes_ it does detect /dev/sda3 as mounted but throws the same behaviour at /dev/sdb

eyduh, 2021-01-23:
In these gifs I might be zipping around quite fast but I've left it at any of the bits where it stalls overnight to come back to the same screen.
The interesting part I find was that it stalled, broken pipe disconnection, ssh back in and the drive is "unmounted". but when I run 
```
# btrfs check /dev/sdb1
Opening filesystem to check...
ERROR: /dev/sdb1 is currently mounted, use --force if you really intend to check the filesystem
```

Some btrfs info:
```
# btrfs filesystem show
Label: 'dbs'  uuid: 26d02176-cc95-45e8-af6b-333f96b4d328
        Total devices 1 FS bytes used 1.70GiB
        devid    1 size 99.63GiB used 3.01GiB path /dev/sda3

Label: none  uuid: b68d8d79-3880-4574-b3dc-24dfc9a049fc
        Total devices 1 FS bytes used 1.45GiB
        devid    1 size 2.73TiB used 4.02GiB path /dev/sdb1
```
Trying to label /dev/sdb I get:
```
# btrfs filesystem label /mnt/adacloud_data

# btrfs filesystem label /mnt/adacloud_data adacloud_data
Segmentation fault
```

eyduh, 2021-01-23:
oh services after boot looks like this:
```
# dietpi-services 
[ INFO ] DietPi-Services | Concurrent execution of DietPi-Services detected, retrying... (1/5)
[ INFO ] DietPi-Services | Concurrent execution of DietPi-Services detected, retrying... (2/5)
[ INFO ] DietPi-Services | Concurrent execution of DietPi-Services detected, retrying... (3/5)
[ INFO ] DietPi-Services | Concurrent execution of DietPi-Services detected, retrying... (4/5)
[ INFO ] DietPi-Services | Concurrent execution of DietPi-Services detected, retrying... (5/5)
[FAILED] DietPi-Services | Cancelled DietPi-Services due to concurrent execution

Odroid C4 (aarch64) | IP: 192.168.178.57
┌───────────────────────────────────────────────┤ DietPi-Services ├───────────────────────────────────────────────┐
│                                                                                                                 │
│ WARNING: Concurrent execution of DietPi-Services detected                                                       │
│                                                                                                                 │
│ Please check if one of the following applies:                                                                   │
│  - This script already runs on another terminal/SSH session.                                                    │
│  - Currently a cron or systemd background job executes the script.                                              │
│  - You started this script from within another DietPi program, causing a loop.                                  │
│                                                                                                                 │
│ Please assure that the concurrent execution has finished, before retrying, otherwise cancel this instance.      │
│                                                                                                                 │
│ The following info might help:                                                                                  │
│  2166 root     ?        /usr/bin/vmtouch -l /boot/dietpi /boot/dietpi.txt                                       │
│  2851 root     ?        /bin/bash /boot/dietpi/postboot                                                         │
│  2902 root     ?         \_ /bin/bash /boot/dietpi/dietpi-services start                                        │
│  3761 root     pts/0     |       \_ /bin/bash /boot/dietpi/dietpi-drive_manager                                 │
│  7789 root     pts/2             \_ /bin/bash /boot/dietpi/dietpi-services                                      │
│  7797 root     pts/2                 \_ /bin/bash /boot/dietpi/dietpi-services                                  │
│                                                                                                                 │
│                                 <Retry>                                  <Cancel>                               │
│                                                                                                                 │
└─────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

MichaIng, 2021-01-23:
Okay let's go through it:
- In your first GIF, you compare the unmounted HDD partition with the actually successfully mounted SDD partition 3. The latter is one line above correctly detected as mounted partition by drive manager.
- What fails is the detection of the HDD partition. `lsblk` shows it, but `blkid` does not show it, as if it was not formatted with an actual file system. Strangely DietPi-Drive_Manager detects it using `blkid` as well.

Let's go from the point on where the script hangs: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-drive_manager#L295
```
blkid -o device # A list of all formatted drives/partitions
blkid -s PARTUUID -o value /dev/sdb1 # I should hang here, or below
blkid -s TYPE -o value /dev/sdb1
lsblk -nro NAME # A list of all drives/partitions, even unformatted ones, but we had that above already
lsblk -nro PARTUUID,UUID /dev/sdb1 # Let's compare those with the ones from the SSD Btrfs partition 3
```

Ah and I should have read your edits first, okay so UUID/PARTUUID is not the issue, still above the exact hanging command can be identified. Whether the drive is mounted or not is btw done by parsing:
```
df -a --output=source --exclude-type=tmpfs --exclude-type=ecryptfs --exclude-type=vboxsf --exclude-type=glusterfs | mawk '/\// && !x[$0]++'
```
We exclude special mount types which are preserved in fstab anyway and which we would add with wrong mount options, where a regular physical block device is expected. So let's see how the full output looks like:
```
df -a
```

I see `UUID_SUB` which makes me believe that we do not handle Btrfs volumes correctly, where e.g. the partition itself might not contain a file system but the Btrfs volume on that partition.

Can you try to label the Btrfs volume via it's device path instead of mount point?
```
btrfs filesystem label /dev/sdb1
btrfs filesystem label /dev/sdb1 adacloud_data
```
_____________
I'll also run some tests with Btrfs. Simply formatting and mounting that file system type worked fine on all my tests, but I never made use of volumes e.g. to add two drives to a single Btrfs mount, one of the core advantages of Btrfs of course 😉.

eyduh, 2021-01-23:
Thanks for helping to structure this troubleshoot.

### Without HDD
The first section here is without the HDD (sdb1) right after a reboot:
```
~# dietpi-drive_manager 
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/sda2 > /
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/sda1 > /boot
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/sda3 > /mnt/26d02176-cc95-45e8-af6b-333f96b4d328

```
where it has been stuck since. =/ 

So, I open up another tab and ssh into the machine and run the other commands:
```
root@adacloud:~# blkid -o device
/dev/sda1
/dev/sda2
/dev/sda3

root@adacloud:~# blkid -s PARTUUID
/dev/sda1: PARTUUID="03823826-01"
/dev/sda2: PARTUUID="03823826-02"
/dev/sda3: PARTUUID="03823826-03"

root@adacloud:~# blkid -s TYPE
/dev/sda1: TYPE="vfat"
/dev/sda2: TYPE="ext4"
/dev/sda3: TYPE="btrfs"

root@adacloud:~# lsblk -nro NAME
sda
sda1
sda2
sda3

root@adacloud:~# lsblk -nro PARTUUID,UUID /dev/sdb1
lsblk: /dev/sdb1: not a block device
root@adacloud:~# lsblk -nro PARTUUID,UUID /dev/sda3
03823826-03 26d02176-cc95-45e8-af6b-333f96b4d328
root@adacloud:~# df -a --output=source --exclude-type=tmpfs --exclude-type=ecryptfs --exclude-type=vboxsf --exclude-type=glusterfs | mawk '/\// && !x[$0]++'
/dev/sda2
/dev/sda1
/dev/sda3

root@adacloud:~# df -a
Filesystem     1K-blocks    Used Available Use% Mounted on
sysfs                  0       0         0    - /sys
proc                   0       0         0    - /proc
udev             1893788       0   1893788   0% /dev
devpts                 0       0         0    - /dev/pts
tmpfs             380120    8092    372028   3% /run
/dev/sda2       12168368 7646688   4504948  63% /
securityfs             0       0         0    - /sys/kernel/security
tmpfs            1900588       0   1900588   0% /dev/shm
tmpfs               5120       0      5120   0% /run/lock
tmpfs            1900588       0   1900588   0% /sys/fs/cgroup
cgroup2                0       0         0    - /sys/fs/cgroup/unified
cgroup                 0       0         0    - /sys/fs/cgroup/systemd
pstore                 0       0         0    - /sys/fs/pstore
bpf                    0       0         0    - /sys/fs/bpf
cgroup                 0       0         0    - /sys/fs/cgroup/debug
cgroup                 0       0         0    - /sys/fs/cgroup/cpu,cpuacct
cgroup                 0       0         0    - /sys/fs/cgroup/devices
cgroup                 0       0         0    - /sys/fs/cgroup/memory
cgroup                 0       0         0    - /sys/fs/cgroup/freezer
cgroup                 0       0         0    - /sys/fs/cgroup/cpuset
cgroup                 0       0         0    - /sys/fs/cgroup/net_cls
cgroup                 0       0         0    - /sys/fs/cgroup/pids
[... 302 lines cut ...]
                  229376 -   442368:       65 ###
                  475136 -   966656:       60 ###
                  999424 -  2080768:       61 ###
                 2113536 - 20103168:       30 #
        Total clusters: 69
                Avg cluster size: 45.68KiB
                Min cluster size: 32.00KiB
                Max cluster size: 160.00KiB
        Total disk spread: 20.11MiB
        Total read time: 0 s 825861 us
        Levels: 3
```  
 
</details>

### next steps?
I've `rsync -avz` all the data on sdb1 to sda3 so I can reformat the drive in whichever way is better. The last time I did it was through `dietpi-drive_manager` in two batches. The first time it threw an error and left it with a gpt partition table but no partitions or volumes. Then, I tried again and it ended up the way it is now.

When I format through kde partition manager, dietpi-drive_manager, fdisk or parted I have ended up with /dev/sdb1 through:
(delete previous partioning)
- make GPT
- make 1 primary partition
- format primary partition with btrfs

When I format through mkfs.btrfs I end up with just /dev/sdb which works fine for all the btrfs tools but makes it not detectable by the aforementioned tools, only `btrfs-progs`

MichaIng, 2021-01-23:
Okay, so the mounted/unmounted state of both Btrfs partitions is not detected correctly and labelling sdb1 via device path worked now while of course it cannot work via mount point as long as it's not mounted.

So from the last outputs, only the hanging `dietpi-drive_manager`, which seems to be not within the detection loop, as all those commands succeed.

Actually to verify this, you can use the debug mode which only does the loop and then prints all found values:
```
G_DEBUG=1 dietpi-drive_manager
```

If that hangs, here are a few other commands done, to test:
```
modprobe autofs4 --dry-run # Checks if systemd automount can be done
systemctl daemon-reload # Reload systemd generator to re-create the mount unit based on fstab entries
sync # Force sync of file system cache to disk
mount -a # Mounts all drives present in fstab which do not have `noauto` flag
```
I checked what is done after the detected until the menu prompt, but it's only variable checks and creation, nothing that is able to hang.

Next I reviewed your output further above: https://github.com/MichaIng/DietPi/issues/4062#issuecomment-765814925
It looks like `dietpi-drive_manager` is called as a child of `dietpi-services`, but there is no service that we install which would do that. Can you show which services are controlled, and if/where a restart hangs?
```
dietpi-services status
dietpi-services restart
```

A bug report upload could also help so that I can review kernel and service logs: `dietpi-bugreport 1`

And one last question for now: Did I get it right that booting without the HDD attached works as expected, without any hanging and the sda3 successfully mounted?

eyduh, 2021-01-23:
<p>
<summary> G_DEBUG=1 dietpi-drive_manager</summary>
<details> 

```
# G_DEBUG=1 dietpi-drive_manager
[ INFO ] DietPi-Globals loaded
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/sda2 > /
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/sda1 > /boot
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/sda3 > /mnt/26d02176-cc95-45e8-af6b-333f96b4d328
[ INFO ] DietPi-Drive_Manager |  - Detected unmounted drive: /dev/sdb1
[  OK  ] DietPi-Drive_Manager | DEBUG INFO:
[  OK  ] DietPi-Drive_Manager | Index=0
aDRIVE_UUID e139ce78-9841-40fe-8823-96a304a09859
aDRIVE_PART_UUID 03823826-02
aDRIVE_MOUNT_SOURCE /dev/sda2
aDRIVE_MOUNT_TARGET /
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE ext4
aDRIVE_SIZE_TOTAL 11.6G
aDRIVE_SIZE_USED 7.3G
aDRIVE_SIZE_PERCENTUSED 63%
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=1
aDRIVE_UUID 3086-D853
aDRIVE_PART_UUID 03823826-01
aDRIVE_MOUNT_SOURCE /dev/sda1
aDRIVE_MOUNT_TARGET /boot
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE vfat
aDRIVE_SIZE_TOTAL 255.7M
aDRIVE_SIZE_USED 86.4M
aDRIVE_SIZE_PERCENTUSED 34%
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=2
aDRIVE_UUID 26d02176-cc95-45e8-af6b-333f96b4d328
aDRIVE_PART_UUID 03823826-03
aDRIVE_MOUNT_SOURCE /dev/sda3
aDRIVE_MOUNT_TARGET /mnt/26d02176-cc95-45e8-af6b-333f96b4d328
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE btrfs
aDRIVE_SIZE_TOTAL 99.6G
99.6G
aDRIVE_SIZE_USED 1.7G
1.7G
aDRIVE_SIZE_PERCENTUSED 2%
2%
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=3
aDRIVE_UUID b68d8d79-3880-4574-b3dc-24dfc9a049fc
aDRIVE_PART_UUID dd9a7247-9425-4080-acf1-905f6f562546
[... 145 lines cut ...]
RestartSec=30s
TimeoutSec=30s

[Install]
WantedBy=multi-user.target
```
</details>
</p>

The other things (apache, mariadb, redis & php) were installed with the nextcloudpi install script.

> And one last question for now: Did I get it right that booting without the HDD attached works as expected, without any hanging and the sda3 successfully mounted?
Ok, so did some tests now and it seems like when I had thought it was stuck on boot before, it was actually stuck on shutdown.
I had called `reboot now` and blinky was from the shutdown sequence.
When booting with the HDD plugged in it just takes a fair bit longer before reaching the bootloader as the Odroid HC4 looks through the plugged in devices for boot partitions and it seems to scan the HDD first.

After boot I'm in the state as described before with `bash /boot/dietpi/postboot` getting stuck on `dietpi-services` failing to start one of the nextcloud services mariadb or redis.

I don't think that sda3 is properly mounted seeing as I just `cd` into a nextcloud data folder and tried to open a .txt file with vim which seems to have stalled. I also `cd` into the folder where the mysql folder is for mariadb and did an `ls` to have it stuck forever.

So, no, sorry for bad info on my part:
- booting works whether or not HDD is plugged in
- it seems sda3 is not mounted properly

( How do I get a verbose output instead of Ms. Blinky during boot/shutdown?)

MichaIng, 2021-01-23:
Okay, probably `dietpi-drive_manager` being child of `dietpi-services` is only shown somehow wrong and the actual issue is accessing data on to the Btrfs partition hangs. That would explain the hangs of MariaDB (with databases in dietpi_userdata which is on the Btrfs partition, right?), hangs of file access or directory listing and `sync`.

Let me check the bug report.

Meanwhile, you did already try to mount it with no (default only) mount options, right? E.g.
```
umount /dev/sda3
sed -i '/btrfs/d' /etc/fstab
systemctl daemon-reload
mount /dev/sda /mnt/26d02176-cc95-45e8-af6b-333f96b4d328
```

MichaIng, 2021-01-24:
<details>
<summary>There is a kernel error:</summary>

```
[   14.262290] BTRFS info (device sda3): disk space caching is enabled
[   14.263101] BTRFS info (device sda3): has skinny extents
[   14.297156] BTRFS info (device sda3): detected SSD devices, enabling SSD mode
[   14.793546] Unable to handle kernel NULL pointer dereference at virtual address 00000044
[   14.796171] pgd = ffffffc0b8c80000
[   14.799777] [0000000000000044] *pgd=0000000000000000, *pud=0000000000000000
[   14.806827] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[   14.812547] Modules linked in: tun amvdec_ports i2c_meson_master stream_input decoder_common firmware media_clock sunrpc ip_tables x_tables ipv6 btrfs xor raid6_pq spidev spi_meson_spicc ahci libahci libata
[   14.831007] CPU: 1 PID: 2605 Comm: mysqld Not tainted 4.9.241-arm64 #1
[   14.837676] Hardware name: Hardkernel ODROID-HC4 (DT)
[   14.842877] task: ffffffc0b8bc3800 task.stack: ffffffc0b8c14000
[   14.848962] PC is at __do_page_cache_readahead+0x3c/0x290
[   14.854492] LR is at ondemand_readahead+0x13c/0x278
[   14.859536] R0  : ffffff80091cc0cc, PFN: 11cc
[   14.864023] pc : [<ffffff80091cbd3c>] lr : [<ffffff80091cc0cc>] pstate: 20400145
[   14.871562] sp : ffffffc0b8c17390
[   14.875029] x29: ffffffc0b8c17390 x28: 0000000000003200
[   14.880489] x27: 0000000000000027 x26: 0000000000000020
[   14.885948] x25: 0000000000000000 x24: 0000000000000000
[   14.891408] x23: 0000000000000080 x22: 0000000000000000
[   14.896868] x21: 0000000000000000 x20: 0000000000000041
[   14.902328] x19: ffffffc07aa88798 x18: 00000000bfb7d911
[   14.907788] x17: 0000000000000010 x16: 0000000000001800
[   14.913248] x15: 0000000000001400 x14: 0000000000001c00
[   14.918708] x13: 0000000000001000 x12: 0000000000000040
[   14.924168] x11: 0000000000000400 x10: 0000000000000001
[   14.929629] x9 : 0000000000000000 x8 : ffffffc0b890f000
[   14.935092] x7 : 0000000000000000 x6 : 000000000000003f
[   14.940548] x5 : 000000000000003f x4 : 0000000000000041
[   14.946008] x3 : 0000000000000080 x2 : 0000000000000000
[   14.951470] x1 : 0000000000000000 x0 : ffffff80091cc0cc
[   14.956931]
               SP: 0xffffffc0b8c17310:
[   14.962215] 7310  00000000 00000000 00000080 00000000 00000000 00000000 00000000 00000000
[   14.970535] 7330  00000020 00000000 00000027 00000000 00003200 00000000 b8c17390 ffffffc0
[   14.978855] 7350  091cc0cc ffffff80 b8c17390 ffffffc0 091cbd3c ffffff80 20400145 00000000
[   14.987175] 7370  b8c17410 ffffffc0 01d15138 ffffff80 ffffffff 0000007f b8c17588 ffffffc0
[   14.995495] 7390  b8c17460 ffffffc0 091cc0cc ffffff80 b890efc0 ffffffc0 00000000 00000000
[   15.003815] 73b0  7aa88798 ffffffc0 00000000 00000000 0000003f 00000000 00000400 00000000
[   15.012135] 73d0  00000000 00000000 00000020 00000000 00000027 00000000 00003200 00000000
[   15.020456] 73f0  b8c17420 ffffffc0 00000000 00000000 7aa88798 ffffffc0 00000189 00000000
[   15.028779]
               X8: 0xffffffc0b890ef80:
[   15.034062] ef80  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.042382] efa0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.050701] efc0  00000000 00000000 00000080 00000041 00000400 00000000 ffffffff ffffffff
[   15.059026] efe0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.067343] f000  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.075661] f020  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.083981] f040  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.092301] f060  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.100626]
               X19: 0xffffffc07aa88718:
[   15.105995] 8718  7aa88718 ffffffc0 7aa88718 ffffffc0 7aa88728 ffffffc0 7aa88728 ffffffc0
[   15.114315] 8738  7aa88300 ffffffc0 c9b88608 ffffffc0 7aa88748 ffffffc0 7aa88748 ffffffc0
[   15.122634] 8758  00000000 00000000 00000000 00000000 0000001a 00000000 00000000 00000000
[   15.130955] 8778  00000002 00000000 00000000 00000000 01dd0fa0 ffffff80 00000000 00000000
[   15.139277] 8798  7aa88628 ffffffc0 02180020 00000000 00000000 00000000 00000000 00000000
[   15.147595] 87b8  00000000 00000000 00000000 00000000 7aa887c8 ffffffc0 7aa887c8 ffffffc0
[   15.155914] 87d8  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.164234] 87f8  00000000 00000000 01dd07b0 ffffff80 00000000 00000000 00000000 02420048
[   15.172560]
               X29: 0xffffffc0b8c17310:
[   15.177938] 7310  00000000 00000000 00000080 00000000 00000000 00000000 00000000 00000000
[   15.186248] 7330  00000020 00000000 00000027 00000000 00003200 00000000 b8c17390 ffffffc0
[   15.194568] 7350  091cc0cc ffffff80 b8c17390 ffffffc0 091cbd3c ffffff80 20400145 00000000
[... 135 lines cut ...]
[   15.396096] [<ffffff8009247ce8>] path_openat+0xcc8/0x1158
[   15.396099] [<ffffff800924945c>] do_filp_open+0x6c/0xe8
[   15.396104] [<ffffff8009234a6c>] do_sys_open+0x13c/0x260
[   15.396107] [<ffffff8009234c14>] SyS_openat+0x3c/0x50
[   15.396111] [<ffffff8009083740>] el0_svc_naked+0x34/0x38
[   15.396117] Code: a906cfa1 aa0303f7 aa0403f4 d503201f (b94046a2)
[   15.396121] ---[ end trace 6ee7e33eeec4e2b5 ]---
```
</details>

- `[   14.793546] Unable to handle kernel NULL pointer dereference at virtual address 00000044`
- `[   15.246138] Process mysqld`
[... cut ...]
If the issues persist with no/default mount options, would be interesting if it fails as well with disabled file system cache. This is at least the only idea I currently have:
```
mount -o sync /dev/sda3 /mnt/26d02176-cc95-45e8-af6b-333f96b4d328
```
But I'd not recommend this as production option, if it worked, but instead better use a different file system, if we do not find a solution.

@meveric
Is there a known issue with Btrfs on C4 with Hardkernel kernel or explicitly via SATA on HC4? I could nearly find no mentions of it in the C4 sub forum, but probably Btrfs is simply rarely used.

eyduh, 2021-01-24:
> Meanwhile, you did already try to mount it with no (default only) mount options, right? E.g.
> 
> ```
> umount /dev/sda3
> sed -i '/btrfs/d' /etc/fstab
> systemctl daemon-reload
> mount /dev/sda /mnt/26d02176-cc95-45e8-af6b-333f96b4d328
> ```

I tried to `umount` but was told 
```
umount: /mnt/26d02176-cc95-45e8-af6b-333f96b4d328: target is busy.
```
so I waited and did other stuff for a few hours, tried again now and same result.
I tried `dietpi-services stop` but
```
Odroid C4 (aarch64) | IP: 192.168.178.57

          ┌─────────────────────────────────────────────────┤ DietPi-Services ├──────────────────────────────────────────────────┐
          │                                                                                                                      │ 
          │ WARNING: Concurrent execution of DietPi-Services detected                                                            │ 
          │                                                                                                                      │ 
          │ Please check if one of the following applies:                                                                        │ 
          │  - This script already runs on another terminal/SSH session.                                                         │ 
          │  - Currently a cron or systemd background job executes the script.                                                   │ 
          │  - You started this script from within another DietPi program, causing a loop.                                       │ 
          │                                                                                                                      │ 
          │ Please assure that the concurrent execution has finished, before retrying, otherwise cancel this instance.           │ 
          │                                                                                                                      │ 
          │ The following info might help:                                                                                       │ 
          │  2175 root     ?        /usr/bin/vmtouch -l /boot/dietpi /boot/dietpi.txt                                            │ 
          │  2869 root     ?        /bin/bash /boot/dietpi/postboot                                                              │ 
          │  2920 root     ?         \_ /bin/bash /boot/dietpi/dietpi-services start                                             │ 
          │ 12367 root     pts/0             \_ /bin/bash /boot/dietpi/dietpi-services stop                                      │ 
          │ 12394 root     pts/0                 \_ /bin/bash /boot/dietpi/dietpi-services stop                                  │ 
          │                                                                                                                      │ 
          │                                   <Retry>                                    <Cancel>                                │ 
          │                                                                                                                      │ 
          └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘ 
                                                                                                                                   ```
guess I could kill that start job and try to make the partition not busy to unmount it and try to mount it again..

meveric, 2021-01-24:
Btrfs is not well supported in Kernel 4.9.
The crashes are actually well known: https://forum.odroid.com/viewtopic.php?p=316609#p316609 and https://forum.odroid.com/viewtopic.php?f=202&t=40917
apparently they go away with newer Kernel versions.
For now with Kernel 4.9 I would stay away from btrfs.

eyduh, 2021-01-24:
> Btrfs is not well supported in Kernel 4.9.
> The crashes are actually well known: https://forum.odroid.com/viewtopic.php?p=316609#p316609 and https://forum.odroid.com/viewtopic.php?f=202&t=40917
> apparently they go away with newer Kernel versions.
> For now with Kernel 4.9 I would stay away from btrfs.

Awesome! I should've investigated this a week ago so I could've abandoned trying to make it work. 

I always do this, assume it's something I've done wrong instead of looking up if it's a known issue.

Wipe time!

closed 2021-01-24 by eyduh

eyduh, 2021-01-24:
Quick question:
what about zfs?
Or should I stick to a simple ext or an lvm+ext4?

meveric, 2021-01-24:
ZFS seems to work, but you have to install the zfs-dkms from buster-backports.
```bash
apt install -t buster-backports zfs-dkms
```
I personally only use lvm+ext4, but that's personal preference.

MichaIng, 2021-01-24:
Many thanks for the info, I should have widened the search from "Other OS" sub forum to the whole C4 forum, whoops 😄.