MichaIng/DietPi #4063: Unable to update from v6.30.0 to v6.34.3

DevPoole, 2021-01-23:
#### Details:
- Date           | Sat Jan 23 01:03:20 GMT 2021
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux PlexServer 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | chmod a+x /mnt/dietpi_userdata
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:

1. Run "sudo dietpi-upgrade"

#### Expected behaviour:

- OS upgrades from current release to latest release
#### Actual behaviour:

- chmod fails due to no /mnt/dietpi_userdata directory
#### Extra details:

Backed up prior to upgrade attempt
Raspberry Pi 4 Model B (armv7l) - 4GB RAM

DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v6.30.0
[ INFO ] DietPi-Update | Latest version  : v6.34.3
[ INFO ] DietPi-Patch | Patching 6.30 to 6.31
[FAILED] DietPi-Patch | chmod a+x /mnt/dietpi_userdata

#### Additional logs:
```
chmod: cannot access '/mnt/dietpi_userdata': No such file or directory

Joulinar, 2021-01-23:
Hi,

Many thanks for your message. Did you transferred DietPi users data directory to an external drive?

DevPoole, 2021-01-23:
I haven't transferred users data at all.
I installed a while back, then installed Plex server and Deluge on top, and haven't made many (if any changes) since. 

From dietpi-software:
 User Data Location   : [SD/eMMC | /mnt/dietpi_userdata]           

(~) PlexServer →  ls /mnt
dietpi-backup  ftp_client  nfs_client  samba  usb

If there's any other information I can provide, please let me know

Joulinar, 2021-01-23:
ok not sure why but you are missing `dietpi_userdata`. as a workaround you would need to create it or copy the folder back from `dietpi_userdata`. At least you have the backup folder. So maybe you can have a look into if there is a `dietpi_userdata` that can be copied back (not to restore entire system)

DevPoole, 2021-01-23:
`dietpi_userdata` wasn't found in `dietpi-backup`

Resolved by creating the directory `/mnt/dietpi_userdata`

MichaIng, 2021-01-23:
Plex and Deluge do by default use /mnt/dietpi_userdata/downloads /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/Music, however, this can be changed of course, so probably you use your own downloads/media directories and removed the obsolete dietpi_userdata then.

While for our software installs, it is currently required as default userdata location, during DietPi updates we should probably not expect it to be present. I'll see if I can make all calls regarding those being skipped if no directory or symlink exists at that path. During dietpi-software installs, where it's required, it's anyway recreated, so no need to force it's existence during updates.

The shown userdata location in the `dietpi-software` menu is btw estimated via `readlink -f /mnt/dietpi_userdata`. This shows the correct actual location if it does exist or is a symlink. But if it does not exist, that command simply prints the input path, at least if the parent directory exists 😄:
```
2021-01-23 15:11:14 root@micha:~# readlink -f /mnt/doesnotexist
/mnt/doesnotexist
2021-01-23 15:11:21 root@micha:~# readlink -f /mnt/doesnotexist/subdir
2021-01-23 15:11:25 root@micha:~#
```
So while it looks a bit misleading, I would leave it as the default location is then the correct one "after" an install has been done. That matches somehow the other menu selectors which do not necessarily represent what is installed already, but what will be installed once you hit "Install" 😉.

MichaIng, 2021-02-07:
Done: https://github.com/MichaIng/DietPi/commit/aad108546c28940b7ec87bddbb2b02f7b5fe2b2e

closed 2021-02-14 by MichaIng