MichaIng/DietPi #4064: Powershell Core on Dietpi

Gill-Bates, 2021-01-23:
What's about adding PowershellCore to the Dietpi-Software Catalog?

```bash
wget https://github.com/PowerShell/PowerShell/releases/download/v7.0.4/powershell-7.0.4-linux-arm32.tar.gz
mkdir ~/powershell; tar -xvf ./powershell-7.0.4-linux-arm32.tar.gz -C ~/powershell
export DOTNET_SYSTEM_GLOBALIZATION_INVARIANT=1
~/powershell/pwsh
```

**Voilá:**

[image]

MichaIng, 2021-01-23:
Many thanks for your request.

There are even amd64 Debian packages available, but to keep it simple I'd use the binary archives for all architectures.

May I ask what the use case is to use the Windows shell on Linux? I mean all instructions to do anything on a Linux system assume that the current shell is bash or bourne shell, so I cannot currently think of another reason than that the syntax or features of PowerShell are superior for certain console tasks, at least based on personal preferences 🙂. Or do users get more used to it when using PowerShell within WSL?

Gill-Bates, 2021-01-23:
Microsoft is working to make PowerShell platform agnostic. Even though Powershell is at home in the Windows world, it can make sense to run Powershell on Linux.

Use Case:
- Running my backup scripts from Dietpi without redesigning into Bash.
- TerraforM

MichaIng, 2021-01-23:
But how or why did you run DietPi backup scripts in PowerShell in the first place? Of course it's possible when it runs on Linux but if someone checks docs or forum or anything else for backup scripts, all will be bash/bourne shell scripts.

Do you mean https://www.terraform.io/ (which works on regular Linux shells as well) or "terraforming" the usage of PowerShell on UNIX/Linux distros? 😃

But do not understand me wrong, it's great to have alternatives and help making cross-platforms work easier via cross-platform tools. I'm just interested what the initial intention in this case is where it would not come to my mind to write scripts for UNIX or GNU/Linux systems in PowerShell or batch, or scripts for Windows systems in UNIX shells.

Josverl, 2022-04-25:
I use powershell commonly for cross-platform CI/CD scripts where the same script can run across linux/mac/windows 
bash/sh/zsh do not offer that portability , or even if the shell does, it still relies on system utilities that are not even the same across linux distros.
So that leaves either Python or pwsh , and Python requires a lot of additional prtocessing (subprocess.run) compared to any of the shells.