MichaIng/DietPi #4066: DietPi-Backup | Restore on First Boot

ravenclaw900, 2021-01-24:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- It can be annoying when you have to go through the full setup process when all you want to do is restore a backup.

#### Describe the solution you'd like:

- Add option to restore a DietPi-Backup on first boot
#### Describe alternatives you've considered:

- Do what is done currently, wait until first boot setup finishes, and restore then.
#### Additional context

- https://github.com/MichaIng/DietPi/issues/3570#issuecomment-760521828

MichaIng, 2021-01-25:
Many thanks for your request.

Reasonable indeed. I think it should be done even before first run setup as everything that is done with it is anyway overwritten by the backup, including install states and all other meta and dietpi.txt info. It could be part of DietPi-FirstBoot which runs right after file system expansion and when all drives (including RAMlog) have been mounted.

But of course this means that it works for new images only. For a while we could add it additionally as first step of first run setup in `dietpi-software`, which runs after `dietpi-update` so that it is effective on older images as well.

ravenclaw900, 2021-02-08:
How exactly would you suggest this be implemented?  I was thinking scan for (external) drives (with the `dietpi-backup` folder on them), and if they are found show a message asking if you want to restore from a backup and a `dietpi-drive_manager` selection menu.

MichaIng, 2021-02-08:
First of all, since DietPi-FirstBoot would require a local screen + keyboard attached, which cannot be done in every situation/with every SBC, let's keep it simple for now and only add it to `dietpi-software`, so that it works on SSH connections as well, but as a first step so that the main menu is not even shown, password and serial console prompts are skipped.

We could either do the drive scan (copy&paste from dietpi-backup code) automatically, and if backup(s) were found, offer to restore them via `dietpi-backup -1 /mnt/sdXn/dietpi-backup`, or we add a new command to `dietpi-backup` which opens the drive scan + selection menu in it directly + restores directly, once selection has done. Ah the second idea sounds too complex...

Instead of doing the scan automatically, it could be triggered by a dietpi.txt option instead. It reduces the potential long scan overhead, but adds another task which can be a little difficult on single-partition images, when using a Windows host which cannot access the ext4 partition.

Additionally or alternatively the target drive/directory could be defined via dietpi.txt as well so that no drive scan and selection would be required, hence an automated restore.

But one issue is that external drives are not mounted by default. so it would require to mount all attached drives with valid file system are mounted automatically to at least temporary mount points, to then run the scan on. This breaks the drive/directory setting in dietpi.txt since in case of multiple attached drives the device path is not predictable and the UUID is probably too technical.
____________
Okay I would do it like that:
- `dietpi.txt` text option to enable interactive `=1` or non-interactive `=2` backup restore.
- As first step of dietpi-software firstrun setup, as early as possible, check for that option.
- Attach all unmounted drives with valid file system to temporary mount points and scan them, for backups, basically like it is done in `dietpi-backup` search feature.
- If `=2` was set, restore the first found backup automatically, so this option is aimed for the usual case where a single backup has been created only.
- If `=1` was set, show the list and let the user select interactively, so that this option can be used as well if multiple backups have been created.

Joulinar, 2021-02-08:
personally I prefer the `dietpi.txt` option and have it set to `0` by default. It might be rar case and probably most users did not even know about backups. I have seen quite some cases where users do upgrades without any backup at hand. 😃

MichaIng, 2021-02-08:
> personally I prefer the `dietpi.txt` option and have it set to `0` by default.

I agree. It add quite some chance to trigger errors as well, so that should not be done automatically.

closed 2021-02-28 by MichaIng