MichaIng/DietPi #4067: rtorrent randomly crashing

shanew1694, 2021-01-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.34.3
- Distro version | buster
- Kernel version | Linux DietPi 5.4.79-v7l+
- SBC model | rpi4 4GB
- Power supply used | official rpi4 power supply
- SDcard used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | rtorrent/rutorrent
- Was the software title installed freshly or updated/migrated? | fresh install
- Can this issue be replicated on a fresh installation of DietPi? | untested

#### Steps to reproduce
I don't think you can reproduce. It's a completely random crash, sometimes it 24hours, sometimes it's multiple days.

#### Expected behaviour
rtorrent should run indefinitely until stopped by me.

#### Extra details
rtorrent status: https://i.imgur.com/rOWDZLy.png this means nothing to me. And can find any other information online.

Currently have 800+ torrents in rtorrent of various sizes. The smallest torrent is 4KB and the largest torrent is 120GB+.  I know rtorrent can handle many more torrents than this and it's not like the RPI is under a lot of load most 0.5 - 1.5 load average with 1.5 - 1.8GB RAM use.

I rather keep the current setup I have now but I might need to change clients, to be honest. qBittorrent looks nice.

MichaIng, 2021-01-25:
Many thanks for your report.

About the abort signal: https://stackoverflow.com/questions/3413166
So yeah, quite unspecific.

We'd get more info when enable file logging:
```
> /mnt/dietpi_userdata/rtorrent/rtorrent.log
chown rtorrent:dietpi /mnt/dietpi_userdata/rtorrent/rtorrent.log
cat << '_EOF_' >> /mnt/dietpi_userdata/rtorrent/.rtorrent.rc

### Logging
log.open_file = "rtorrent.log", "/mnt/dietpi_userdata/rtorrent/rtorrent.log"
log.add_output = "info", "rtorrent.log"
_EOF_
```

I have no idea who verbose the "info" level is, so how fast this file grows, so have a look at it and in case reduce the log level: https://github.com/rakshasa/rtorrent/wiki/LOG-Logging
It can be moved to `/var/log/...` as well, but if it's RAMlog, it's cleared every hour, which might currently break the debugging effort.

We should add such a commented logging section by default to our config file, probably we can even get it logging to STDOUT (`journalctl -u rtorrent`) so that it's combined with the wrapping service log.

shanew1694, 2021-01-25:
> Many thanks for your report.
>
> About the abort signal: https://stackoverflow.com/questions/3413166
> So yeah, quite unspecific.
>
> We'd get more info when enable file logging:
>
> ```
> chown rtorrent:dietpi [...]

Ok! I set-up logging using this information. The logs don't seem very verbose even at "info". Logging has been running for 15minutes now and they're only 2 entries: 

- 1611583519 N rtorrent main: Starting thread.

- 1611583519 N rtorrent scgi: Starting thread.

I would of expect more entries. Maybe later I'll change it to "debug" either way this is gonna take a few days I need to wait for it crash again.

shanew1694, 2021-01-30:
Well, it crashed again about 3 days ago. I checked the log file we set-up and there were no further entries.

So, I started looking in the Github repository and noticed that at some point in time, rtorrent got a little fresh up as the .rtorrent.rc had changed ([here](https://github.com/MichaIng/DietPi/blob/ffdbf684a45498710be7a927c40bcc698d028b70/dietpi/dietpi-software#L11170)) massively from the version I had. I went ahead and did a back-up of my .session folder and purged rtorrent from my machine and did a full reinstall.

Now ~4days later rtorrent is still running fine. I don't want to jinx anything but I think it's fixed. 

I did however need to make a small change to the new .rtorrent.rc file. I had to comment out "network.port_open.set = no" with this enabled I was getting "Tracker: [Failure reason \"Missing port\"]" from the trackers I use.

MichaIng, 2021-01-31:
Great, now we have a match 😀: https://github.com/rakshasa/rtorrent/issues/939
This seems to have made things better, but the underlying bug seems to not have been identified yet: https://github.com/rakshasa/libtorrent/issues/198#issuecomment-610230434
Since a scheduler rework was planned, you could test the current version, but you need to compile from sources: https://github.com/rakshasa/rtorrent/wiki/Installing
```
apt update
apt install build-essential
cd /tmp
curl -sSfLO https://github.com/rakshasa/rtorrent/archive/master.tar.gz
tar xf master.tar.gz
rm master.tar.gz
cd rtorrent-master
./configure
make
make install
```
Not sure how familiar you are with compiling software from sources. It might fail due to missing development headers, or a missing library. Such dependencies are not listed explicitly in the wiki, but can be usually easily derived from the error message.

shanew1694, 2021-01-31:
I'll give this a go when I get the chance.

dietpi-backup does an entire system back-up, right? so If try this and mess it up I could just restore from the back-up.

MichaIng, 2021-01-31:
> dietpi-backup does an entire system back-up, right?

If dietpi_userdata has been moved to an external drive, it is excluded (as all /mnt/*), so e.g. `/mnt/dietpi_userdata/rtorrent` would need to be backed up manually. AFAIK, it contains the config file only, which is not touched by the daemon or install, so should be fine to leave it or even play a bit with the values I linked above.

shanew1694, 2021-02-02:
Spent a few hours trying to install from the source. Lots and lots of tinkering and a couple restores later and I got this far: https://i.imgur.com/sXY0ZB1.png

These are permissions issues I don't know how to fix. Also, I don't think libtorrent0.13.8 is loading either.

MichaIng, 2021-02-02:
Ah I didn't think about `libtorrent`. Probably the newer rTorrent also requires a newer libtorrent 🤔.
Indeed it requires the newest one as well: `PKG_CHECK_MODULES([DEPENDENCIES], [libtorrent >= 0.13.8])`
Was it compiled automatically together with rTorrent (somehow pulled in) or did you need to compile it separately?

And does the systemd unit actually loads the new binary? `which -a rtorrent`
The one from the package should be `/usr/bin/rtorrent` and the compiled one is usually installed to `/usr/local/bin/rtorrent`. So `/etc/systemd/system/rtorrent.service` needs to call the correct executable. Permissions should be fine as user and it's home and config dir were already prepared by the `dietpi-software` install.

shanew1694, 2021-02-02:
Yeah, the newest rTorrent (0.9.8) requires libtorrent (0.13.8) which I compiled both from source. 

And, yeah you're right, it installs to `/usr/local/bin/rtorrent` by default. I tried this first and then also changed `rtorrent.service` to reflect but still had issues with it assumably not loading libtorrent and the same permissions issues. 

I then restored from backup and removed `/usr/bin/rtorrent` and tried `./configure --prefix=/usr --with-xmlrpc-c` which did install new rtorrent to `/usr/bin/rtorrent` but i still had issues with libtorrent not loading and permissions issues. 

I was originally having issues compiling rTorrent at all but then I found out you need to compile using the `--with-xmlrpc-c` tag otherwise you'd get some weird XMLRPC issues when launching rTorrent.

MichaIng, 2021-02-02:
I'll try it out as well, but probably not tomorrow yet as finishing one PR has priority 😉.

shanew1694, 2021-02-03:
I've done it! 

I just pulled the latest version of ruTorrent and put it into `/var/www/rutorrent` all permissions issues now solved. And libtorrent 0.13.8 now loads

Now I just need to do it for real on my Raspberry pi and not in a dietpi VM.

So the upgrade process was:

- Stop all running services
- Compile xmlrpc-c from source 
- Compile Libtorrent 0.13.8 from source
- Compile rTorrent 0.9.8 from source using `--with-xmlrpc-c` tag
- Pull in ruTorrent 3.10 and cp to `/var/www/rutorrent
- Change `rtorrent.service` from `usr/bin/rtorrent` to `usr/local/bin/rtorrent`
- `systemctl daemon-reload`
- `sudo ldconfig`
- Start all services

MichaIng, 2021-02-03:
Let's help that resolves the crashes, otherwise we can revive the issue on the rTorrent and/or libtorrent GitHub.

shanew1694, 2021-02-04:
Nope! 

I can get it to run on an amd64 virtual machine. But doesn't seem to work well with arm

`1612395913 C Caught signal: 'Segmentation fault.`
`---DUMP---`
`Caught Segmentation fault, dumping stack:`
`/usr/local/bin/rtorrent() [0x1f480]`
`/lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer+0) [0xb694e120]`
`---END---`

Seen people having these issues before and it was a curl issue. So I update to the latest (7.75.0) but I'm still having this issue.

Maybe tomorrow ill try downgrading to rtorrent(0.9.6), after all, it does say it's the only 'stable' release. If that doesn't work then I'll have to tinker with qBittorrent.

MichaIng, 2021-02-04:
But you did build it on the ARM machine, didn't you?

shanew1694, 2021-02-06:
Not for testing no. I wanted to see if I could actually do it before I tried it on my Raspberry Pi so I did it in a VM. Which it did work in the VM but when I tried it on the Raspberry Pi, it compiles and installs fine, but then crashes with the error message in the previous post. Think it has something to do with ruTorrent and how that all works, as rtorrent would continue to run until I launched the ruTorrent WebUI then it would crash.

Also tried rtorrent0.9.6 yesterday with no luck. I forgot what the error messages were but something like AD__CPPUNIT=(1.9.6) during the ./configure stage, even though the installed version of what it was asking for was already much higher. 

I've already spent too much time on this. So now I've installed qbittorent-nox and updated it a little to v4.2.5. Still can't fully uninstall rtorrent yet as I need some plugins from ruTorrent. But, as rtorrent and ruTorrent run kind of independently from one-another if rtorrent crashes for any reason it won't affect my "workflow" as I can still access the plugins needed from the WebUI.

MichaIng, 2021-02-06:
Okay, sounds like it has to do with the RPC connection from ruTorrent to rTorrent. I'll try it myself a little later.

MichaIng, 2021-04-14:
As part of #4166, I switched to a socket based RPC connection, which should reduce overhead a bit. It can be tested by migrating to the development branch:
```
dietpi-backup 1 # Always good to have an updated backup in place
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /boot/dietpi.txt
dietpi-update 1
dietpi-software reinstall 107
```

MichaIng, 2021-04-24:
I mark this as closed. Please feel free to reopen if issue persists.

closed 2021-04-24 by MichaIng