MichaIng/DietPi #4068: [RPI 4] Detect usb drives through usb-c dock

k1r0s, 2021-01-25:
Using this device I was able to mount different drives with other opperating systems, like ubuntu mate. 
https://www.amazon.co.uk/VEMONT-Multiport-Ethernet-Compatible-Chromebook-gray/dp/B088FKC7ZH

### Creating a bug report/issue

#### Required Information
    DietPi version | DietPi v6.34.3
    Distro version | buster
    Kernel version | Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
    SBC model | RPi 4 Model B (aarch64) REV1.2
    Power supply used | transformer 5V 3A
    SDcard used | x, booted from USB3.0 15GB stick

Doing `sudo dietpi-drive_manager` does not detect any drive connected to it. Even refreshing, any tips? is there any software requirement?

thanks in advance!

MichaIng, 2021-01-25:
Many thanks for your report.

I hope you power it by external USB-C plug and not via RPi 4 USB, right? Else "different" drives (if you mean multiple ones concurrently) will very likely not work due to insufficient 5W max overall USB power that RPi slots can serve.

Nice dock actually, does it look that high-quality-like in real as well?

Back to topic: Is it detected as USB device and the sub-devices as well?
```
lsusb
```
And/or is the second Ethernet interface detected?
```
ip -l
```

k1r0s, 2021-01-26:
> Nice dock actually, does it look that high-quality-like in real as well?

Oh yeah, its pretty cool but kinda useless with my current build :(

k1r0s, 2021-01-26:
`ip -l` does nothing so
```
dietpi@DietPi:~$ ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:ef:b1:1c brd ff:ff:ff:ff:ff:ff
```

I don't know if its detected but here it is.

lsusb does not show any additional device connected through the dock.

likely to need more power. Its like all the power is going through the dock

k1r0s, 2021-01-26:
[image]

For the sake of clearness

k1r0s, 2021-01-26:
Usb devices do not receive any power but with other OS they do :/

Joulinar, 2021-01-26:
How is it behaviour if you use plain Raspberry OS. As well did you test the 32bit image?

k1r0s, 2021-01-29:
Hello, im back, sorry for the delay. 

> Plug the USB-C data connector of the dock via USB-C to USB-A adapter to one of that RPi USB data jacks.

It will be kinda weird to power them separately.

I removed a drive from the RPI and usbs ports from the dock did manage to "react" to usb sticks when plugged I mean because these sticks that have a let manage to light on for a second, then fade down again. However the RPI was unable to detect any.

Next I'll try to do this: 

> Our 32-bit image is based on the regular 32-bit Raspberry Pi OS: https://www.raspberrypi.org/software/operating-systems/#raspberry-pi-os-32-bit

Cheers

k1r0s, 2021-01-29:
> [image]
> 
> For the sake of clearness

yes indeed the usb-c cable that goes into the dock is from the PSU even though it looks like it comes from the RPI.

It would be nice to use a more powerful PSU but I don't have any other :|

MichaIng, 2021-01-31:
> It will be kinda weird to power them separately.

But I think it's not stable to do it differently. As said, from all I know, plugging the dock to the RPi 4 USB-C slot can not work, as it can not receive any data stream through that port. If you really want to avoid two PSUs, as a prove of concept, try the following:
- Plug the docks data cable via USB-C to USB-A adapter to one of the RPi 4 USB-A slots.
- Remove the separat USB-C power input cable from the dock.
- Plug a single USB stick (to rule out any voltage issues) into the dock the check whether it's detected then.

k1r0s, 2021-02-02:
will check this in detail but I have to insist that I managed to power the RPI and connect multiple devices with other OS. 

Here is some quick googling results:

https://magpi.raspberrypi.org/articles/connect-raspberry-pi-4-to-ipad-pro-with-a-usb-c-cable
https://www.interelectronix.com/raspberry-pi-4-usb-c-host-mode.html

and more specifically:

> the Raspberry can also be operated via the USB-C port as a peripheral device – e.g., as an Ethernet adapter or as a mass storage device. To do this, add the following line at the end of the Raspberry configuration file (/boot/config.txt):

`dtoverlay=dwc2,dr_mode=peripheral`

> Depending on the intended use, this will require various additional software configuration adjustments. You will find plenty of instructions for this online.

MichaIng, 2021-02-02:
Good find. I know the gadget mode but though that it can only be used to make the RPi 4 a peripheral, so that you can access from another PC or such to the RPi as an attached drive or Ethernet device. But `dtoverlay=dwc2,dr_mode=host` turns RPi 4 into the host, so that attached devices are accessible from the RPi 4 via the USB-C port. That is awesome! The link suggests to power the RPi 4 then still separately via GPIO pins, but that does not seem to be mandatory. In peripheral mode the RPi can definitely powered + accessed through USB, like any other peripheral device, so I hope in host mode it's the same.

Did you add `dtoverlay=dwc2,dr_mode=host` to `/boot/config.txt`?

k1r0s, 2021-02-02:
here is my /boot/config.txt

```
# Docs: https://www.raspberrypi.org/documentation/configuration/config-txt/

#-------Display---------
# If you get no picture, set the following to "1" to apply most compatible HDMI settings.
#hdmi_safe=1

# Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
# - Values from "0" to "11" are allowed, use values above "7" only if required, e.g. with very long HDMI cable.
# - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
#config_hdmi_boost=5

# Uncomment if HDMI display is not detected and composite is being outputted.
#hdmi_force_hotplug=1

# Uncomment to disable HDMI even if plugged, e.g. to force composite output.
#hdmi_ignore_hotplug=1

# Uncomment to force a console size. By default it will be display's size minus overscan.
#framebuffer_width=16
#framebuffer_height=16

# Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
#enable_tvout=0
# SDTV mode
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This can make audio work in DMT (computer monitor) modes.
#hdmi_drive=2
#hdmi_drive:1=2

# Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
# With default value "0", the display shows a blank screen instead, but will not go into standby.
# NB: With "1" some applications (e.g. Kodi, OMXPlayer) cannot prevent display standby due to missing DPMS signal.
#hdmi_blanking=1

# Set to "1" if your display has a black border of unused pixels visible.
disable_overscan=1

# Uncomment the following to adjust overscan. Use positive numbers if console goes off screen, and negative if there is too much border.
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# Rotation
display_hdmi_rotate=0
lcd_rotate=0

#-------RPi camera module-------
#start_x=1
#disable_camera_led=1

#-------GPU memory splits-------
gpu_mem_256=128
gpu_mem_512=128
gpu_mem_1024=128

#-------Max USB current---------

#-------Boot splash screen------
disable_splash=1

#-------Onboard sound----------
dtparam=audio=on

#-------I2C-------------
dtparam=i2c_arm=off
dtparam=i2c1=off
dtparam=i2c_arm_baudrate=100000

#-------SPI-------------
dtparam=spi=off

#-------Serial/UART-----
# NB: Enabled for 1st run only. If you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_ENABLE=1 in dietpi.txt before 1st boot.
core_freq=500
enable_uart=0

#-------Overclock-------
temp_limit=75
initial_turbo=20

over_voltage=4
arm_freq=1900
#core_freq=500

#arm_freq_min=300
#core_freq_min=250
#sdram_freq_min=400
arm_64bit=1
dtoverlay=disable-wifi
dtoverlay=vc4-kms-v3d-pi4
dtoverlay=disable-bt
dtoverlay=dwc2,dr_mode=host
```

nothing has changed. It does not recognize any drive and it seems that is a matter of electrical power. Maybe the raspberry is taking too much power?

MichaIng, 2021-02-02:
Maybe. I see you have also applied quite some overclocking and overvoltage. Do you run a desktop or other graphical software or is it basically a headless setup? ... Ah sorry I see you use one of the mini HDMI ports.

Ah I see you use the 64-bit image. It could be the reason at least since it uses a very fresh Linux 5.10 version that may not yet contain all features. So what could be tried:
- Removing overclocking, to rule out low power issues. Also unplugging other USB devices temporarily could help to further be sure.
- Use our 32-bit image to have the stable and fully featured kernel that also matches the one that is used on most guides around.
- Test the official Raspberry Pi OS, which uses the same kernel and firmware, to rule out any issues or missing packages on DietPi side: https://www.raspberrypi.org/software/operating-systems/#raspberry-pi-os-32-bit

Ah another one: `rpi-eeprom-update -a`
To update the EEPROM which contains also USB drivers. It should actually be done automatically on every boot, but to be sure.
To check the service: `systemctl status rpi-eeprom-update`

k1r0s, 2021-02-05:
> Do you run a desktop or other graphical software or is it basically a headless setup? ... Ah sorry I see you use one of the mini HDMI ports.

Yup, its my work station xd. Im sorry but I will be unable to test everything u've suggested but im pretty sure that will fix the issue.

Will try it soon and come back with updates :) thanks for the help!

MichaIng, 2021-04-10:
I'm turning this issue into a feature request, since we collected information and links here about the RPi gadget mode and it would be nice to have this as a selectable feature for `dietpi-config`.

bam80, 2021-11-01:
> In peripheral mode the RPi can be definitely powered + accessed through USB, like any other peripheral device, so I hope in host mode it's the same.

I'm struggling to find prove of that.
Could someone try to power RPi via USB-C while it's in a host mode? Any info disclosing it?

MichaIng, 2021-11-01:
Indeed it is actually uncommon that the USB "host" is powered by the attached USB device/peripheral. Needs to be tested definitely, else the only way are two PSUs, one for the RPi and one for the USB dock to assure all attached USB devices have sufficient power, especially when 2.5" drives are among them.

bam80, 2021-11-01:
>  Could someone try to power RPi via USB-C while it's in a host mode? Any info disclosing it?

Opened question upstream:
https://forums.raspberrypi.com/viewtopic.php?t=322783

MichaIng, 2021-11-02:
Looks like it is generally possible to power an RPi in USB-C host mode through an attached USB device like a dock, while data transfer (using the dock as dock) remains possible.