MichaIng/DietPi #4069: PINE A64

javilonso, 2021-01-26:
### Creating a bug report/issue

#### Required Information
- DietPi version |  **DietPi_PINEA64-ARMv8-Buster.7z | 2020-05-17 17:30 | 98M**

#### Steps to reproduce

1. Flash image on sd card
2. Power the Pine A64

#### Expected behaviour

- Normal boot

#### Actual behaviour

- Initialization of the system starts but it freezes on **Starting Kernel ...**. There are no lineas displayed after that line.

#### Extra details

- Pine A64 normal version, not the LTS version.

MichaIng, 2021-01-26:
Many thanks for your report.

Actually we could provide an image for LTS version / SOPINE as well, the one we have however is for non-LTS version, so from that point of view it should work. I'll update the image first, which is already quite old, before we go on debugging kernel/bootloader issues that may be already solved 😉.

javilonso, 2021-01-27:
@MichaIng Thank you for your reply. I have a PINE A64 lying around for a long time and I wanted to install a light terminal based distro for running some bots and staff. 

PINE A64 support is "dead" on the official site.
As DietPi is lighter, I want to have a try. For now I was able to use the 2017 version of Dietpi for Pine A64. Luckily it was stored on web cache.

I made a mistake, neither the 2018 nor the 2020 releases version work. 
The 2018 release is stack on  **BOOTP Broadcast 1, BOOTP Broadcast 2,** and so on infinitely.
The 2020 release freezes on **Starting Kernel ...**

If you have much work in other recent boards, go for it. Dont waste time for this old board. I will find other alternartives.

javilonso, 2021-01-27:
I will update soon again, it might be a problem with the sd card I am using being too fast.

MichaIng, 2021-01-28:
The new image is available: https://dietpi.com/downloads/images/DietPi_PINEA64-ARMv8-Buster.7z

javilonso, 2021-01-28:
Thank you for your reply @MichaIng, really appreciated.

The main problem I found was the micro SD card I was using. I change it with a samsung evo plus and the last image of DietPi was working. The new image you attached is working too.

However, there is a problem with the internet connection. In both cases I had to set a static IP in the ethernet interface (no wifi).

This is the process I did, so you can understand what might be the problem.

1. Flash the image
2. Boot for the first time and set the static IP
3. Reboot
4. In the second boot, the system has internet connection and starts to download DietPi stuff (I believe are some updates or dependences).
5. After that update, the system reboots and there is no internet connection although the router detects the static ip of the board. I can´t ping any device or web. I tried to change again the ip, but still fails to do connects. 

I suffer this problem in armbian with kernel 5.10 (same as the last DietPi image). I found on armbian forum that other users with specific boards such as Orange Pi boards were having similar problems with no internet connections and they recommend switching back to kernel 5.8. I tried an Armbian Buster Image with kernel 5.8 and the internet works great, even after upgrading packages.

Currently this Armbian Image works great for me, hope you can use it as reference:
https://minio.k-space.ee/armbian/archive/pine64/archive/Armbian_20.11_Pine64_buster_current_5.8.16.img.xz

Other References:
https://forum.armbian.com/topic/16430-orange-pi-3-h6-ethernet-not-working-on-59-and-above/
https://github.com/armbian/build/issues/2315

MichaIng, 2021-01-28:
Many thanks for the info. Hmm, yeah probably we'll simply create an image based on Debian kernel, this is already the second case short after another where a DietPi kernel caused major issues, among some other things that are difficult when using Armbians kernel and firmware.

It's not that easy to use an old kernel, since it is auto upgraded on first boot. So we would need to manipulate the image, block or remove the kernel package, and then users would stay with that kernel until manually unlocking or reinstalling it. That does all not sound like a great solution 🤔. I'm currently experimenting much with the Debian ARM kernel and how to create images best from scratch. Kinda learning curve to master, but in terms of stability and reliability I hope it will be a worth enhancement.

I'll have a look into the issue at the Armbian forum, maybe we find a solution or workaround.

javilonso, 2021-01-29:
In summary in case others PINE A64 users have problems:

- if I use armbian based on kernel 5.10 the internet does not work. 
- By using any image with kernel 5.8, internet works great by setting static ip.

Keep in mind that in the Armbian Image with kernel 5.8.16 I had to freeze kernel updates in armbian-config. 
If not, kernel will update to 5.10 when I upgrade packages and the internet stops working.

@MichaIng Dont waste to much time on this topic if you have other tasks buddy. This board is "quite old" but I appreciate your interest ;)

javilonso, 2021-01-29:
After more research, my problem could come from the ethernet.
I dont have a wifi module, so I have to use ethernet to connect to the internet.
DHCP didn't work for me in any image I have tried so fire with kernel 5.x

I have been reading that first units of pine A64 might have a different ethernet chip. Other user says that by using kernel 3.10 longsleep it worked.
So I decided to tried an image based on that kernel and it worked!! DHCP now works great and sets the ip.

I tried the following image
[https://www.stdin.xyz/downloads/people/longsleep/tmp/pine64-images/ubuntu/](https://www.stdin.xyz/downloads/people/longsleep/tmp/pine64-images/ubuntu/)

MichaIng, 2021-01-30:
Nasty, that is the original hopelessly outdated manufacturer kernel.

Strange that with Linux 5.10 on second boot internet connection worked. Aside of an update check and APT package upgrades (if already available), nothing is installed anymore and network settings are not touched. To investigate the issue, `dmesg` can be used to check for kernel errors regarding the Ethernet device or network in general, `ifdown eth0` and `ifup eth0` to deconfigure and reconfigure the interface, probably this fixes the internet connection temporarily and/or can be used to trigger fresh related kernel logs.

javilonso, 2021-02-10:
I apprecitate your advices but I already tried those commands to play with the ethernet connections (and many more). I decided not to lose more time and keep with an the kernel 3.10 longsleep that just works fine for running some scripts 24/7

I close this issue. AlthoughI did't find solution I was looking for, I got the board working properly :)

SUMMARY: Old image of Ubuntu with kernel 3.10 longsleep for PINE A64
https://github.com/ayufan-pine64/linux-build/releases/download/0.7.30/stretch-minimal-pine64-bspkernel-0.7.30-140.img.xz

closed 2021-02-10 by javilonso

MichaIng, 2021-02-10:
Nasty, but good to keep in mind, as an alternative when others report Ethernet issues with that image.