MichaIng/DietPi #4070: PaperMC fails to install

Camry2731, 2021-01-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v6.34.3`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.4.83-v7l+`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | `CanaKit 3.5A USB-C Power Supply with Noise Filter`
- SDcard used | `Samsung EVO+ 64 GB Class 10`

#### Additional Information (if applicable)
- Software title | `PaperMC`
- Was the software title installed freshly or updated/migrated? | `Fresh`
- Can this issue be replicated on a fresh installation of DietPi? | `Yes`

- Bug report ID | `7b82aa2a-daa5-413a-a920-46c468657132`

#### Steps to reproduce
1. Go to dietpi-software
2. Select and install PaperMC

#### Expected behavior
- DietPi installs prerequisites, installs PaperMC, asks to accept EULA, configures it, etc.

#### Actual behaviour
- Installs prerequisites, asks to accept EULA, then errors out with a long error, which after skimming through, says towards the end: "The requested URL returned error: 403"

#### Extra details
Consult the bug report for the full error log, unless I should post it here?

Joulinar, 2021-01-28:
Hi,

many thanks for your message. Unfortunately download URL changes short after we released PaperMC as port of DietPi software catalogue. There will be a fix provided on next release. https://github.com/MichaIng/DietPi/pull/4039

A possible workaround could be this https://dietpi.com/phpbb/viewtopic.php?p=30193&sid=e7e02e0636bf3360a7184ec642ff34de#p30193

Camry2731, 2021-01-28:
Ah, sorry about that. I checked to see if this was a known issue and I must of overlooked it, appreciate the info.

MichaIng, 2021-01-28:
Hi Cameron, many thanks for you hint. Yes indeed an unlucky timing with the changed download API. This time we checked back with the developers which one the future prove is.

closed 2021-02-28 by MichaIng