MichaIng/DietPi #4071: Cannot install ProFTPD

Olfried2, 2021-01-29:
#### Details:
- Date           | Fri Jan 29 16:05:18 CET 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.4.83+ #1379 Mon Dec 14 13:06:05 GMT 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Being logged in via ssh (fixed IP)
2. Opened dietpi-software and chose to install ProFTPD
#### Expected behaviour:

- ProFTPD should be installed
#### Actual behaviour:

-  ssh.service - OpenBSD Secure Shell server
 ```  Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-01-29 16:05:17 CET; 123ms ago
     Docs: man:sshd(8)
           man:sshd_config(5)
  Process: 2311 ExecStartPre=/usr/sbin/sshd -t (code=exited, status=0/SUCCESS)
  Process: 2312 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255/EXCEPTION)
 Main PID: 2312 (code=exited, status=255/EXCEPTION)

Jan 29 16:05:17 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
Jan 29 16:05:17 DietPi sshd[2312]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Jan 29 16:05:17 DietPi sshd[2312]: error: Bind to port 22 on :: failed: Address already in use.
Jan 29 16:05:17 DietPi sshd[2312]: fatal: Cannot bind any address.
Jan 29 16:05:17 DietPi systemd[1]: ssh.service: Main process exited, code=exited, status=255/EXCEPTION
Jan 29 16:05:17 DietPi systemd[1]: ssh.service: Failed with result 'exit-code'.
Jan 29 16:05:17 DietPi systemd[1]: Failed to start OpenBSD Secure Shell server.
dpkg: error processing package openssh-server (--configure):
 installed openssh-server package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 openssh-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Software | APT upgrade
 - Command: apt-get -qq upgrade
```

#### Extra details:

- ...
#### Additional logs:
```
Setting up openssh-server (1:7.9p1-10+deb10u2) ...
rescue-ssh.target is a disabled or a static unit, not starting it.
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "start" failed.

Joulinar, 2021-01-29:
Hi,

many thanks for your message. Just for clarification. Are you talking about FTP Server or SSH server. because the error message you posted is on SSH server error while you are trying to install ProFTPD Server. 

The SSH server installation is failing because you are running another SSH server already and the port 22 us already in use.

```
Jan 29 16:05:17 DietPi sshd[2312]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Jan 29 16:05:17 DietPi sshd[2312]: error: Bind to port 22 on :: failed: Address already in use.
Jan 29 16:05:17 DietPi sshd[2312]: fatal: Cannot bind any address.
```

MichaIng, 2021-01-29:
Probably Dropbear SSH server is installed already?
```
dpkg -l dropbear-run
```

Joulinar, 2021-01-29:
@MichaIng 
If I'm not mistaken, OpenSSH server is failing during `apt-get -qq upgrade`. Therefore I guess it was already installed. But good to check for Dropbear anyway.

Olfried2, 2021-02-02:
Oh, sorry - forgot to answer. 
OpenSSH was running of course (_zero W_) - it's a headless system. I did a reboot and started the installation again, but didn't got ProFTP to work. (sorry can't remember what happened exactly) . Changed to scp instead of ftp. Maybe this installation routine needs a checkup if OpenSSH is already available and ignore the installation if available.

MichaIng, 2021-02-02:
The packages OpenSSH service is not running, if the error was the same, so I'd verify that. Dropbear SSH server is pre-installed on all DietPi systems, so would could instead allow you SSH access and even invoke the SCP binary of OpenSSH. To get a list of all ports that is listened on and the related process names, use: `ss -tulp`

Olfried2, 2021-02-03:
My fault, Dropbear was running. On my other machines OpenSSH is running (sorry for the confusion)

MichaIng, 2021-02-03:
No problem. If you want to switch, I suggest the selector in `dietpi-software` which stops Dropbear (only the listener process, not the active connection) before installing OpenSSH and uninstall Dropbear automatically.

closed 2021-02-09 by Olfried2