MichaIng/DietPi #4072: error while updating from 6.28.0 to 6.34.3

seniorgod, 2021-01-29:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Jan 29 17:54:04 CET 2021
- Bug report     | 00a3ec72-cae0-4382-9d0d-ff197c6bff6c
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | tar -xf 3.10.38.bz2 -C /lib/modules/
- Exit code      | 2
- Software title | DietPi-Patch
#### Steps to reproduce:

i try to update a very old also sac with dietpi from 6.28.0 6.34.3 
during install it try to download data from 

curl -sSfL https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/dragon_fly_check/uImage -o /boot/uImage

[  OK  ] DietPi-Patch | curl -sSfL https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/dragon_fly_check/3.10.38.bz2

[FAILED] DietPi-Patch | tar -xf 3.10.38.bz2 -C /lib/modules/

#### Expected behaviour:

it should work :-)
- 
#### Actual behaviour:

if i run the curl command manually there is now file downloaded 
because 

#### Extra details:

- ...

#### Additional logs:
```
tar: 3.10.38.bz2: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
```
---------------------------------------------------------------------

MichaIng, 2021-01-29:
Many thanks for your report.

Please try again, there was a bug in our patch file so that the download was not stored in a file. I just fixed it in the master branch.

seniorgod, 2021-01-29:
now it download and untarnished... 

next error: 
[image]

MichaIng, 2021-01-29:
Looks like this block has never really been tested. The update should go on. To apply the intended setting, do:
```
G_CONFIG_INJECT 'options wlan_8812au rtw_power_mgnt=0' 'options wlan_8812au rtw_power_mgnt=0' /etc/modprobe.d/dietpi-disable_wifi_powersaving.conf
```
I just fixed it in the patch file as well.

seniorgod, 2021-01-29:
ok, thanks

closed 2021-01-29 by seniorgod