MichaIng/DietPi #4075: Recovery module needed!

cocoflan, 2021-01-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-01-31:
Hi cocoflan, thanks for your request.

Hui that is a difficult one since different type of failure require a different level of recovery. In some cases a simple reinstall of system packages might help, in other cases exactly that will fail or not be sufficient to repair the system. Sometimes configs themselves might need to be restored/adjusted, while usually you want to preserve user customised configs. A full restore image build would need to exist for every SBC separately, as all of them boot with a different bootloader (settings), kernel and firmware, so we would need to basically maintain two images for each SBC. It's easier then to boot with a fresh DietPi image (or any other Debian-based image), mount the broken image and repair it from there.

OpenMediaVault ships only x86_64/amd64 images, no SBC images anymore, so it's easy to maintain some sort of recovery partition for them, although the question remains until which point things are recovered, in which case custom user data or at least configs are preserved and from which point on that recovery won't be enough. It's much easier on Windows or macOS, a single system with limited customisation, non-modular, for a single architecture, and even they have multiple levels of recovery to cover all cases.
____________
I plan to create a installer image along with the actual DietPi image for SBCs, i.e. a very minimal image to install DietPi from to an internal drive: no network, very limited user-space tools, hardcoded init into a script that does nothing more than allows to select a target drive to flash the actual packaged DietPi image to. I never realised that this can be done quite easily based on a regular image. So it's a bit similar to the Clonezilla-based installer image for x86_64 systems. Taking this as basis, also a minimal recovery image would be thinkable, but only when the SBC allows booting from USB or other separate drive. A few common drive/file system and chroot tools could be pre-installed, booting into a simple single-user shell to start investigating and fixing the issue. We could add common non-intrusive recovery scripts by times, but for me it's difficult to define what a full recovery would be without risking to break user configs, software services and such. But that is clearly some steps in the future, if not someone else is able to pick it up 😉.

cocoflan, 2021-01-31:
[ https://wiki.gnome.org/Apps/DejaDup | https://wiki.gnome.org/Apps/DejaDup ] 

Works like a charm on my linux mint system and is configurable to exclude folders and recover per date of backup. 

Greetings

MichaIng, 2021-01-31:
Ah, to recover a backup of course would be less of an issue. I guess I misunderstood "there is a need to have a standard restore image build in" to be something like a reference system to recovery/repair the main system from (like file-by-file), like Windows SFS or similar. But you mean a minimal system to boot when the main system is broken, to then recover a backup from or do other repair steps. E.g. `dietpi-backup` could be included to restore a `dietpi-backup` or any other kind of backup or manual repair steps like fsck and such.

Would be possible, but most SBCs cannot be easily instructed to boot from a different partition of the same drive. It would require something like boot menu and on RPi with default bootloader this doesn't exist, with U-Boot I have at least currently no idea if/how it is possible. Pressing a certain button to boot from alternate partition would be practical. However, most SBCs can boot from USB, either automatically if attached with bootable image, or when manually unplugging the SD card. So a recovery USB stick is something I find very doable. If the SD card needed to be removed, it can be plugged back in and mounted to start repair or recovery. What do you think?

cocoflan, 2021-01-31:
Good idea!! 
When installing the system ask for a backup usb stick for recovering nice thinking!! 

Just looking for a posible solutions after my recovery problems yesterday. 

Greetings, 

will post an request maby some people have ideas about this.

Joulinar, 2021-01-31:
similar request #4066

MichaIng, 2021-01-31:
At least it deals with a similar issue: An unbootable system where a backup exists. However a recovery image could be also used for repairing, investigating or saving/extracting data, if no backup exists, or it can be repaired easily, e.g. re-downloading or fixing some broken kernel/bootloader files or configs. Theoretically any booting Linux image on a stick can do it, but one that has a set of useful related tools pre-installed, so that one won't usually require to even setup network, could come quite handy.

StephanStS, 2021-02-08:
See also https://github.com/MichaIng/DietPi-Docs/issues/235.

cocoflan, 2021-10-09:
is it not possible to create a micro image of the basic system? if there is possibility on systems with external disks to image on? more and more I have a problem to get everything right again from system and where restore simple is not enough.