MichaIng/DietPi #4076: Error with apt upgrade or dietpi-update

omavoss, 2021-01-31:
Hello, my  DietPi v6.34.3 (beta) : 22:07 - So 31.01.2021 say error:

[code]
#### Details:
- Date           | Sun Jan 31 22:10:30 CET 2021
- DietPi version | v6.34.3 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: lzma_read: Read error (9)
W: You may want to run apt-get update to correct these problems
E: The package cache file is corrupted
```
[/code]

Please help him, thank you.

Joulinar, 2021-01-31:
Hi,

many thanks for your message. Pls can you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

omavoss, 2021-01-31:
Hello,
that solved the problem, thank you and kind regards.

closed 2021-01-31 by omavoss