MichaIng/DietPi #4077: Unbound: Root Hints automatic update?

APraxx, 2021-02-02:
Hi, 

I'm using dietpi and it is great. Thx

I just noticed that in the unbound package there is no automatic update of root hints file.
They don't change often but they do.

Can we add a systemd timer like 
[https://wiki.archlinux.org/index.php/Unbound#Roothints_systemd_timer](https://wiki.archlinux.org/index.php/Unbound#Roothints_systemd_timer)
or is it on purpose?

Joulinar, 2021-02-02:
Hi,

many thanks for your question. The feature was already on the agenda but it is not yet done. For the time being, you could use following to set it up

```
echo -e '#!/bin/dash\ncurl -sSfL https://www.internic.net/domain/named.root -o /var/lib/unbound/root.hints' > /etc/cron.monthly/dietpi-unbound
chmod +x /etc/cron.monthly/dietpi-unbound
```

APraxx, 2021-02-02:
I could make a PR.

Which way would you like it implemented?
Like your example with cron or with a systemd timer?

Joulinar, 2021-02-02:
let's wait on the developer @MichaIng

MichaIng, 2021-02-02:
Yes definitely, thanks for the reminder. I'd go with the cron job, which is slightly simpler to implement and understand for most users.

Cron is still required for at least two important core system tasks, so that we cannot yet think about migrating to systemd timers completely, which I would consider when DietPi scripts are the only ones on a fresh system that require the cron daemon to remain 😉.

MichaIng, 2021-02-07:
Done: https://github.com/MichaIng/DietPi/commit/690b01be8f34bc8a09f3524e945a8ff23bf75d3b
Changelog: https://github.com/MichaIng/DietPi/commit/98b5be8b7b1746b1e95963daa3911fd572c44b39

closed 2021-02-14 by MichaIng