MichaIng/DietPi #4081: fstrim still disabled

TheOriginalMrWolf, 2021-02-03:
Hi,

Almost 4½ years later and it seems that fstrim is still disabled (DietPi v6.34.3, on Odroid N2) - any plans to re-enable on, say, a weekly basis?  Not having this causes significant slowdowns & additional wear due to write amplification.

I would hope that the underlying issue is resolved by now?

Thanks!

_Originally posted by @TheOriginalMrWolf in https://github.com/MichaIng/DietPi/issues/440#issuecomment-772215959_

MichaIng, 2021-02-03:
Many thanks for your report.

Since Debian Buster there is actually a weekly systemd timer to run TRIM. Check:
```
systemctl status fstrim.service
systemctl status fstrim.timer
```

> on boot sdcard

It's not supported by all SD cards, btw. You can try it manually, but in my case, e.g.:
```
2021-02-03 12:59:31 root@micha:~# fstrim /boot
fstrim: /boot: the discard operation is not supported
2021-02-03 12:59:53 root@micha:~# fstrim /
fstrim: /: the discard operation is not supported
```

TheOriginalMrWolf, 2021-02-03:
Thanks for that.  

I'm on Stretch (as are probably a lot of other users) - does that mean our only option is to upgrade to Buster?

Was the original underlying issue which prompted Fourdee to disable weekly fstrim in DietPi all those years ago ever resolved?

Running `fstrim -a` manually worked for my boot SD card - is it safe to put that in a weekly cron (or add fstrim.service & timer back)?

Cheers!

MichaIng, 2021-02-03:
To me it looks like it was never even proven that fstrim was responsible for the increased file system corruptions. I cannot say, if so, if it was related to the kernel or user space tools (fstrim executable itself) or drive/file system drivers. But since `util-linux` upstream developers created that weekly systemd timer and Debian didn't patch it out, it can be seen as pretty stable, at least on Debian Buster. The package version on Stretch is older of course (upstream version didn't contain the timer back then), the kernel version as well in most cases (RPi and x86_64 at least), so taking into account the various possible sources or errors, I will not change the Debian defaults for DietPi on a distro version which is "oldoldstable" in a few months.

Our daily cron was however disabled during Jessie/Wheezy cycle, so I am pretty sure that it is safe for you to enable weekly fstrim on Stretch. Just do a backup first and check `dmesg -l emerg,alert,crit,err` after the first fstrim executions for file system or I/O errors.

Upgrading to Debian Buster may however be a good reason anyway 😉.

TheOriginalMrWolf, 2021-02-04:
So.... I set up a fresh DietPi Buster install.

Turns out that fstrim.timer is still disabled by default 😞 

Needed to run (as root):
```
systemctl enable fstrim.timer
systemctl start fstrim.timer
```
Before `systemctl list-timers` would show it as active.

MichaIng, 2021-02-04:
Oh, this was the N2 image? We don't touch that timer at all, but it should be enabled by default.

I'll add a command to our script to actively enable it.

TheOriginalMrWolf, 2021-02-04:
> Oh, this was the N2 image? 
Oops, sorry, yes indeed 😄 

> I'll add a command to our script to actively enable it
👍

closed 2021-02-14 by MichaIng