MichaIng/DietPi #4082: dietpi-drive_manager stores network usernames & passwords in clear text in world-readable fstab file

TheOriginalMrWolf, 2021-02-03:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch

- Kernel version | `uname -a`
Linux homehub 4.9.241-arm64 #1 SMP PREEMPT Tue Jan 5 00:03:12 CET 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Odroid N2

- Power supply used | (EG: 5V 1A RAVpower)
Odroid N2 supplied

- SDcard used | (EG: SanDisk ultra)
SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
dietpi-drive_manager

- Was the software title installed freshly or updated/migrated?
fresh install

- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
start dietpi-drive_manager and map an SMB/CIFS network drive which requires a password.
Username & password are stored in cleartext in fstab.
fstab is world readable.
This is bad.

#### Expected behaviour

Usernames/passwords should be protected and, ideally, encrypted.  They should _at least_ be stored in a separate, restricted access, credentials file & that file referenced in fstab; though this is still problematic as the credentials remain in cleartext.

Maybe use something like Cifscloak, libpam_mount, or similar?

#### Actual behaviour

- CIFS usernames and passwords are readable by anyone who can log in or physically access the drive storing /etc/fstab

MichaIng, 2021-02-03:
Many thanks for your report.

Indeed, for curlftpfs mounts its actually the same. Good idea to store it in a separate file, which is natively supported. Encryption would be awesome, but it's not natively supported, sadly.

My suggestions is to store it in `/var/lib/dietpi/`, with `root:root` `600` for now. I'll further check the options to encrypt it, but depending on the amount of additional dependencies we'd leave that to the user, probably giving some hint about the options.

TheOriginalMrWolf, 2021-02-03:
Thanks for looking at this so quickly.

My main concern is that using dietpi-drive_manager is practically mandated (ie the _Please use "dietpi-drive_manager" to setup mounts_ message in fstab, plus interesting behaviours if you manually edit fstab & then subsequently use dm) - so any solution should be implemented/supported by that.

Failing that, at least documentation (& fstab message) should be updated to clearly articulate the issue and workaround.

Cheers!

MichaIng, 2021-02-03:
> My main concern is that using dietpi-drive_manager is practically mandated

Yes, that is actually something I want to get rid of, similarly for network setup. The goal is to have these scripts as optional, that one might want to use, especially when inexperienced, but can skip or mix when doing more advanced/customised configs. They should be much more compatible with customisations in general.

In case of dm (nice abbreviation 😄) at least CIFS (and a few other) fstab entries are preserved as they are, so you can replace the credentials manually with the external file reference and still use dm. But if one customises a physical drive entry, that is completely overwritten. Unknown mount types are even removed. It is by design like that, but I personally don't like that behaviour and was thinking about creating dm mount entries directly as systemd mount units. All fstab entries are converted by systemd into mount units dynamically at boot anyway, so that would even reduce overhead a tiny little bit and we would be able to leave `/etc/fstab` completely untouched, free for custom user mounts. dm would still list all mounts and drives, regardless if/how mounted, but when mounting them via dm, they could be shown with a flag as permanently mounted via `/etc/systemd/system/mnt-sda1.mount` or such. `fstab` would need to get an info comment about where and how to check those mounts, to not cause confusion.
But that would be a larger project and I'm concerned about compatibility with other system tools that often still refer and check fstab for active mounts.

Another simpler idea would be to just leave fstab completely untouched and only alter those entries where a drive is actively mounted or removed via dm GUI. It was however a decision by DietPi founder to re-create it on access for a reason, and probably we should at least check back when and why it was done like that before rolling back completely 😉.

TheOriginalMrWolf, 2021-02-03:
Thank you so much for the rapid, comprehensive, informative, well considered and articulate response!!! 😄 👍 

The solution of using dm to manage systemd mount units & letting users manage fstab sounds spot on.

I'll leave it in your capable hands confident that we'll eventually get a great outcome.  🙇‍♂️

closed 2021-04-24 by MichaIng