MichaIng/DietPi #4085: dietpi on buster - no network access at first boot

seniorgod, 2021-02-04:
Dear all,

at weekend i set up a old sparky with an new image from DietPi based on buster for sparky.
After flashing it to the sd card, there was no chance to do a headless install. I have to connect keyboard and monitor in order to do the first install. First i thought it was an error. But today i installed a DietPi based on buster to a raspberry pi - with the same effect. No chance for a headless install

ravenclaw900, 2021-02-04:
Hello,
How were you trying to set up network?  For WiFi you need to put the credentials in `dietpi.txt`.

Joulinar, 2021-02-04:
Hi,

many thanks for your message. Are you able to complete the installation once your connect screen and monitor? Did you tried to connect via SSH? Did you used correct IP address? A couple of more information on the issue would be helpful.

seniorgod, 2021-02-04:
Dear all,
i tested again

it seems the the installation of dietPi on buster on raspberry Pi works! 
I try it again. I found out that i did a mistake while using the „old“ ip, while my DHCP server was defining a new ip address. In fact DietPi on buster on raspberry pi work. 

What not worked at weekend was the installation of DietPi on buster on all sparky. I need to connect with screen and keyboard. The final installation was „waiting“ at the console. After finishing the final installation steps, the reboot works and the system was healthy and reachable vie ssh.

seniorgod, 2021-02-04:
Always fix connection with lan cable and dhcp

Joulinar, 2021-02-09:
I guess you are back to Stretch for the Sparky. Can this be closed than?

seniorgod, 2021-02-09:
I‘m back. You can close the issue

closed 2021-02-09 by Joulinar