MichaIng/DietPi #4087: Exit code | 7

MichaelLikesJazz, 2021-02-05:
#### Details:
- Date           | Fri Feb  5 14:42:46 GMT 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | curl -sSfL https://github.com/syncthing/syncthing/releases/download/v1.13.1/syncthing-linux-arm-v1.13.1.tar.gz -o syncthing-linux-arm-v1.13.1.tar.gz
- Exit code      | 7
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (7) Failed to connect to github.com port 443: Connection refused
```

MichaIng, 2021-02-05:
Many thanks for your report.

Could you please try to manually download the file?
```
curl -sSfL https://github.com/syncthing/syncthing/releases/download/v1.13.1/syncthing-linux-arm-v1.13.1.tar.gz -O /tmp/syncthing-linux-arm-v1.13.1.tar.g
rm syncthing-linux-arm-v1.13.1.tar.g
```

Do you use Pi-hole or any similar ad block tool on your server or network, that might block connections to the GitHub download server?

MichaelLikesJazz, 2021-02-05:
Thanks. No Pi-hole is not running at the moment. I reinstalled dietpi, now it's running.

MichaIng, 2021-02-05:
Okay great. Probably it was only a temporary server issue.

Joulinar, 2021-02-08:
Hi Guys,

I'm going to close the issue as it seems to be fixed. Fell free to reopen if needed.

closed 2021-02-08 by Joulinar