MichaIng/DietPi #4088: apt and wget segmentation fault

chippey5, 2021-02-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6 G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 5.10.12-rockchip64 #&#x2060;21.02.1 SMP PREEMPT Wed Feb 3 20:55:02 CET 2021 aarch64 GNU/Linux
- SBC model | ROCKPro64 (aarch64)
- Power supply used | RockPro64 5V 2A
- SDcard used | eMMC

#### Steps to reproduce

1. sudo apt-get update

#### Expected behaviour

`Reading package lists... Done`

#### Actual behaviour

```
Reading package lists... Done
E: Method https has died unexpectedly!
E: Sub-process https received a segmentation fault.
```

#### Extra details

Multiple packages were marked as upgradebale after a previous kernel update and reboot, after which these packages were updated and board restarted the issue presented itself. I unfortunately don't recall which packages were updated and `/var/log/dpkg.log` has been flushed since. The same issue affects `wget`:

```
wget https://raw.githubusercontent.com/MichaIng/DietPi/master/README.md
--2021-02-06 14:00:48--  https://raw.githubusercontent.com/MichaIng/DietPi/master/README.md
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.84.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.84.133|:443... connected.
Segmentation fault
```

I have tried the following:

- From [the debian repo](https://mirrors.dotsrc.org/debian-security/pool/updates/main/) downloaded an older version of apt (1.4.11) as well as updating `wget` to [1.21-1](https://mirrors.dotsrc.org/debian/pool/main/w/wget/wget_1.21-1_arm64.deb) to no avail. 

Some forums suggested it could be my sources list(s) that do or don't contain http**s**:

```
cat /etc/apt/sources.list
deb https://deb.debian.org/debian/ buster main contrib non-free
deb https://deb.debian.org/debian/ buster-updates main contrib non-free
deb https://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb https://deb.debian.org/debian/ buster-backports main contrib non-free
```

```
cat /etc/apt/sources.list.d/*.list
deb http://apt.armbian.com buster main buster-utils buster-desktop
```

Changing the latter file to http**s** had no effect. I'm not sure what to do from here on. Any help would be much appreciated.

MichaIng, 2021-02-06:
Many thanks for your report.

I remember this one, similar error, sadly no solution: #2257

To sort it our, are there any kernel errors?
```
dmesg -l emerg,alert,crit,err
```

The APT repos/mirrors are generally fine. Of course you could use HTTP instead of HTTPS on all cases, so `/usr/lib/apt/methods/https` is not used, which probably fixes APT. But it does not solve the underlying issue of HTTPS requests in general.

Does `curl` work?
```
curl -L https://raw.githubusercontent.com/MichaIng/DietPi/master/README.md
```

chippey5, 2021-02-06:
```
$ dmesg -l emerg,alert,crit,err
[    4.258025] rk_gmac-dwmac fe300000.ethernet: cannot get clock clk_mac_speed
[    7.243434] OF: graph: no port node found in /i2c@ff3d0000/typec-portc@22
```

On a second RockPro64 that hasn't been updated the past week yet gives no output on `dmesg -l emerg,alert,crit,err`, but I'm not sure if these errors are linked to the issue at hand.

Curl works fine, I had to resort to curl to download the apt packages as wget didn't work.

As always, thanks for your help @MichaIng.

MichaIng, 2021-02-06:
I don't think the two error during boot are related either, however one at least is related to the Ethernet adapter. But still, maybe it's an issue with the latest kernel upgrade, especially when it works on the other ROCKPro64.

To test the last Linux 5.9 kernel:
```
cd /tmp
curl -sSfLO https://apt.armbian.com/pool/main/l/linux-5.9.14-rockchip64/linux-image-current-rockchip64_20.11.4_arm64.deb
curl -sSfLO https://apt.armbian.com/pool/main/l/linux-5.9.14-rockchip64/linux-dtb-current-rockchip64_20.11.4_arm64.deb
dpkg -i *.deb
rm *.deb
```

Otherwise actually with `curl` I remember a few strange HTTPS/TLS errors, which were "resolved" by switching the network adapter, in one case only by changing the IP. Those were very care cases, even reported to curl developers upstream, but never fully understood. So just to give something else to narrow it down, try to change the IP and switch from Ethernet to WiFi vice versa, if possible. But let's see first if the older kernel solves it, so we can report to Armbian.

chippey5, 2021-02-06:
> I don't think the two error during boot are related either, however one at least is related to the Ethernet adapter. But still, maybe it's an issue with the latest kernel upgrade, especially when it works on the other ROCKPro64.

And right you were, downgrading gives us: 

```
$ uname -a
Linux DietPi 5.9.14-rockchip64 #20.11.4 SMP PREEMPT Tue Dec 15 08:52:20 CET 2020 aarch64 GNU/Linux
chippey5@DietPi:~$ sudo apt-get update
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Hit:5 https://mirrors.dotsrc.org/armbian-apt buster InRelease
Reading package lists... Done
```

In the name of science I upgraded the kernel again to see if the issue could be reproducable:

```
$ uname -a
Linux DietPi 5.10.12-rockchip64 #21.02.1 SMP PREEMPT Wed Feb 3 20:55:02 CET 2021 aarch64 GNU/Linux
$ sudo apt-get update
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Hit:5 https://mirrors.netix.net/armbian/apt buster InRelease
Reading package lists... Done
```

Which it couldn't(?). Seems like a reinstall of the kernel would've/should've done the trick.

Issue solved even though it is a bit disconcerting that things like this happen when upgrading the kernel, which is a good reminder to have a backup close by. Thanks as bunch as always @MichaIng!

closed 2021-02-06 by chippey5

MichaIng, 2021-02-06:
Great to hear, although always a little unsatisfying when the question about the initial issue stays open 😄.