MichaIng/DietPi #4089: Output of "dietpi-software list" prints out links to outdated dietpi.com/phpbb instead of dietpi.com/docs

StephanStS, 2021-02-06:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

If `dietpi-software list` is executed, a list of software packages is generated.

Example:
```
root@dietpi:~# dietpi-software list | grep ' =2'
id 5 | =2 | alsa: linux sound system |
id 16 | =2 | build-essentials: common packages for compile |
id 103 | =2 | dietpi-ramlog: minimal, optimised logging | | https://dietpi.com/phpbb/viewtopic.php?p=68#p68
id 104 | =2 | dropbear: lightweight ssh server | | https://dietpi.com/phpbb/viewtopic.php?p=62#p62
```

This list output contains links to dietpi.com/phpbb. The phpbb software area is outdated and is replaced by the dietpi.com/docs area.

#### Describe the solution you'd like:

The links shall direct to dietpi.com/docs. The example above e.g. should link to

- https://dietpi.com/docs/software/log_system/#dietpi-ramlog
- https://dietpi.com/docs/software/ssh/#dropbear

The solution is somehow a rework of the `aSOFTWARE_...` structure in `dietpi-software`: 

```
software_id=103
aSOFTWARE_NAME[$software_id]='DietPi-RAMlog'
aSOFTWARE_DESC[$software_id]='minimal, optimised logging'
aSOFTWARE_TYPE[$software_id]=-1
aSOFTWARE_CATEGORY_INDEX[$software_id]=0
aSOFTWARE_ONLINEDOC_URL[$software_id]='p=68#p68'
```

which is evaluated in 

```
# - Online docs
[[ ${aSOFTWARE_ONLINEDOC_URL[$i]} ]] && string+=" | \e[90m$FP_ONLINEDOC_URL${aSOFTWARE_ONLINEDOC_URL[$i]}\e[0m"
```

I could support in changing all the `aSOFTWARE_...` entries if the general scripting is changed.

Joulinar, 2021-02-06:
already on the way https://github.com/MichaIng/DietPi/pull/4086

StephanStS, 2021-02-06:
Oh, sorry. Didn't check the open issue list. :-(

-> Typical PEBCAC problem. :-)

Joulinar, 2021-02-06:
well it's a RP and not an issue 🤣

MichaIng, 2021-02-06:
Sometimes I wonder how many abbreviations I still need to look up 😄: https://www.cyberdefinitions.com/definitions/PEBCAC.html

StephanStS, 2021-02-06:
See also 
- https://en.wiktionary.org/wiki/PEBCAK
- (german): https://de.wikipedia.org/wiki/Liste_von_Abk%C3%BCrzungen_(Netzjargon)#P 

:rofl:

closed 2021-02-14 by MichaIng