MichaIng/DietPi #4092: install of docker with dietpi-software fails on dietpi 6.34.3 buster on sparky armhf

seniorgod, 2021-02-07:
Dear all,

i try to install docker in my sparky with DietPi 6.34.3 on buster armhf 

The installation fails, but it seems docker is at least partly installed (i can use it with systemctl). 
I get an error during installation (i send a bug via the included bug report of dietpi-software) 
When i start docker with systemctl i get the error i attached below. It seems that docker try to start something with the "graphics" and fails. I use a sparky without any monitor in headless mode. 

    Feb 07 11:28:01 DietPi systemd[1]: Starting Docker Application Container Engine...
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.436432475+01:00" level=info msg="Starting up"
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.455004975+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.455488725+01:00" level=info msg="scheme \"unix\"     not registered, fallback to default scheme" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.456089975+01:00" level=info  msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}"module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.456679975+01:00" level=info msg="ClientConnswitching balancer to \"pick_first\"" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.480782225+01:00" level=info msg="parsed scheme:\"unix\"" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.481501725+01:00" level=info msg="scheme \"unix\"not registered, fallback to default scheme" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.482131975+01:00" level=infomsg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.482622475+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
    Feb 07 11:28:02 DietPi dockerd[13900]: time="2021-02-07T11:28:02.740165725+01:00" level=error msg="[graphdriver] prior storage driver devicemapper failed: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed"
    Feb 07 11:28:02 DietPi dockerd[13900]: failed to start daemon: error initializing graphdriver: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed
    Feb 07 11:28:02 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
    Feb 07 11:28:02 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
    Feb 07 11:28:02 DietPi systemd[1]: Failed to start Docker Application Container Engine.
    Feb 07 11:28:05 DietPi systemd[1]: docker.service: Service RestartSec=2s expired, scheduling restart.
    Feb 07 11:28:05 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 1.
    Feb 07 11:28:05 DietPi systemd[1]: Stopped Docker Application Container Engine.
    Feb 07 11:28:05 DietPi systemd[1]: Starting Docker Application Container Engine...
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.134044726+01:00" level=info msg="Starting up"
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.144081226+01:00" level=info msg="parsed scheme:\"unix\"" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.144338226+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.144662726+01:00" level=info  msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}"module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.144992476+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.157367976+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.157660476+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.157931476+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.158227226+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
    Feb 07 11:28:06 DietPi dockerd[13927]: time="2021-02-07T11:28:06.283034476+01:00" level=error msg="[graphdriver] prior storage driver devicemapper failed: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed"
    Feb 07 11:28:06 DietPi dockerd[13927]: failed to start daemon: error initializing graphdriver: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed
    Feb 07 11:28:06 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
    Feb 07 11:28:06 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
    Feb 07 11:28:06 DietPi systemd[1]: Failed to start Docker Application Container Engine.
    Feb 07 11:28:08 DietPi systemd[1]: docker.service: Service RestartSec=2s expired, scheduling restart.
    Feb 07 11:28:08 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 2.
    Feb 07 11:28:08 DietPi systemd[1]: Stopped Docker Application Container Engine.
    Feb 07 11:28:08 DietPi systemd[1]: Starting Docker Application Container Engine...
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.688373976+01:00" level=info msg="Starting up"
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.706710226+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.707492976+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.708273726+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.708848726+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.732438726+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.732940226+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.733445976+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.741911976+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
    Feb 07 11:28:09 DietPi dockerd[13947]: time="2021-02-07T11:28:09.977995226+01:00" level=error msg="[graphdriver] prior storage driver devicemapper failed: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed"
    Feb 07 11:28:09 DietPi dockerd[13947]: failed to start daemon: error initializing graphdriver: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed
    Feb 07 11:28:10 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
    Feb 07 11:28:10 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
    Feb 07 11:28:10 DietPi systemd[1]: Failed to start Docker Application Container Engine.
    Feb 07 11:28:12 DietPi systemd[1]: docker.service: Service RestartSec=2s expired, scheduling restart.
    Feb 07 11:28:12 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 3.
    Feb 07 11:28:12 DietPi systemd[1]: Stopped Docker Application Container Engine.
    Feb 07 11:28:12 DietPi systemd[1]: docker.service: Start request repeated too quickly.
    Feb 07 11:28:12 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
    Feb 07 11:28:12 DietPi systemd[1]: Failed to start Docker Application Container Engine.

MichaIng, 2021-02-07:
Many thanks for your report.

It's not about "graphics" but the "graph" driver. It looks like a typical source of such error is a corrupted device mapper meta data file: https://bodhileafy.wordpress.com/2018/04/20/docker-error-running-devicecreate-createsnapdevice-dm_task_run-failed/

Do you still have enough empty space? `df`

If so, can you run those commands:
```
systemctl stop docker
thin_check --clear-needs-check-flag /mnt/dietpi_userdata/docker-data/devicemapper/devicemapper/metadata
thin_check /mnt/dietpi_userdata/docker-data/devicemapper/devicemapper/metadata
systemctl start docker
```
If the `thin_ckeck` command is not available, please install the following package:
```
apt install thin-provisioning-tools
```

seniorgod, 2021-02-07:
that’s strange 
i have enough space but there is no docker_data directory below /mnt/dietpi_userdata: 

none              506596   13032    493564    3% /run
udev               10240       0     10240    0% /dev
tmpfs             506596       0    506596    0% /dev/shm
/dev/mmcblk0p2  14801320 2197780  12431580   16% /
tmpfs               5120       4      5116    1% /run/lock
tmpfs             506596       0    506596    0% /sys/fs/cgroup
tmpfs              51200     108     51092    1% /var/log
tmpfs            1048576       0   1048576    0% /tmp
/dev/mmcblk0p1     49038   12196     36842   25% /boot

root@DietPi:/mnt/dietpi_userdata# ls -l
insgesamt 16
drwxrwxr-x 2 dietpi dietpi 4096 Jan 31 17:12 downloads
drwxrwxr-x 2 dietpi dietpi 4096 Jan 31 17:18 Music
drwxrwxr-x 2 dietpi dietpi 4096 Jan 31 17:12 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 Jan 31 17:12 Video

How to find out what went wrong at installation of docker ?

> Am 07.02.2021 um 12:49 schrieb MichaIng <notifications@github.com>:
> 
> thin_check --clear-needs-check-flag /mnt/dietpi_userdata/docker-data/devicemapper/devicemapper/metadata

Joulinar, 2021-02-07:
Pls Run the command's as requested by @MichaIng above

seniorgod, 2021-02-07:
i Did

systemctl stop docker
thin_check --clear-needs-check-flag /mnt/dietpi_userdata/docker-data/devicemapper/devicemapper/metadata
thin_check /mnt/dietpi_userdata/docker-data/devicemapper/devicemapper/metadata
systemctl start docker

stop docker worked

but the first thin_check failed because there is no docker-data directory below dietpi_userdata

Joulinar, 2021-02-07:
It would be good to know what went wrong during docker install. Do you have the logs? Or can you rerun install?

seniorgod, 2021-02-07:
I can rerun the installation again and again with the same error.
Where do I find detailed logs of the Installation?

Joulinar, 2021-02-07:
during installation you get notified with an error message. pls can you share this one

seniorgod, 2021-02-07:
that’s the error - but it seems that this is the problem during start 
do we have any details logs from the installation?

Joulinar, 2021-02-07:
you did not share any error message. If you sent a picture it is not shipped via email. you would need to login to Github

seniorgod, 2021-02-07:
sorry
 
i jump to GitHub

seniorgod, 2021-02-07:
[image]

Joulinar, 2021-02-07:
ah ok you are failing on `apt upgrade`. Looks like your system is trying to update docker software package, which is failing due to incorrect initial setup. we could try to remove docker completly

`apt purge docker-ce`

if this complete without issues, try to run `dietpi-software` again

MichaIng, 2021-02-07:
The Docker package install implies a service start. Since this fails, the package install fails, hence the Docker installer fails, hence dietpi-software terminates before Docker data is even moved to dietpi_userdata, so that all makes sense.

I think purging `docker-ce` does not purge the container data. Before doing that, please try the following, same as above, but on original Docker data location:
```
systemctl stop docker
thin_check --clear-needs-check-flag /var/lib/docker/devicemapper/devicemapper/metadata
thin_check /var/lib/docker/devicemapper/devicemapper/metadata
systemctl start docker
```

seniorgod, 2021-02-07:
The content of that post was also send by mail. Until now it doesn't appear here. So i posted again via GitHub. Perhaps the content occur twice if the mail is processed... 
####################

i follow the steps below. It means i didn’t  purge docker-ce 
but it doesn’t improve the situation

root@DietPi:/mnt/dietpi_userdata# systemctl stop docker
root@DietPi:/mnt/dietpi_userdata# thin_check --clear-needs-check-flag /var/lib/docker/devicemapper/devicemapper/metadata
examining superblock
  superblock is corrupt
    bad checksum in superblock, wanted 1490015127
root@DietPi:/mnt/dietpi_userdata# thin_check /var/lib/docker/devicemapper/devicemapper/metadata
examining superblock
  superblock is corrupt
    bad checksum in superblock, wanted 1490015127
root@DietPi:/mnt/dietpi_userdata# systemctl start docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
root@DietPi:/mnt/dietpi_userdata# 

content of journalctl : 

Feb 07 18:56:02 DietPi systemd[1]: Starting Docker Socket for the API.
Feb 07 18:56:02 DietPi systemd[1]: Listening on Docker Socket for the API.
Feb 07 18:56:02 DietPi systemd[1]: Starting Docker Application Container Engine...
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.661732680+01:00" level=info msg="Starting up"
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.680686930+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.681144680+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.681739430+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <n
il>}] <nil> <nil>}" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.682324430+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.706793180+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.707471930+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.708022180+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <n
il>}] <nil> <nil>}" module=grpc
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.708412180+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Feb 07 18:56:03 DietPi kernel: device-mapper: table: 254:0: thin-pool: unknown target type
Feb 07 18:56:03 DietPi kernel: device-mapper: ioctl: error adding target to table
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.959894930+01:00" level=error msg="[graphdriver] prior storage driver devicemapper failed: devicemapper: Error running deviceCre
ate (CreatePool) dm_task_run failed"
Feb 07 18:56:03 DietPi dockerd[19335]: time="2021-02-07T18:56:03.970215680+01:00" level=info msg="stopping event stream following graceful shutdown" error="context canceled" module=libcontaine
rd namespace=plugins.moby
Feb 07 18:56:03 DietPi dockerd[19335]: failed to start daemon: error initializing graphdriver: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed
Feb 07 18:56:04 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 18:56:04 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Feb 07 18:56:04 DietPi systemd[1]: Failed to start Docker Application Container Engine.
Feb 07 18:56:06 DietPi systemd[1]: docker.service: Service RestartSec=2s expired, scheduling restart.
Feb 07 18:56:06 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 1.
Feb 07 18:56:06 DietPi systemd[1]: Stopped Docker Application Container Engine.
Feb 07 18:56:06 DietPi systemd[1]: Starting Docker Application Container Engine...
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.067104180+01:00" level=info msg="Starting up"
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.097611680+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.097982180+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.098444930+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <n
il>}] <nil> <nil>}" module=grpc
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.098991930+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.119760930+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.120093680+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Feb 07 18:56:07 DietPi dockerd[19358]: time="2021-02-07T18:56:07.120428930+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0 <n
[... cut ...]
Feb 07 18:56:13 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 3.
Feb 07 18:56:13 DietPi systemd[1]: Stopped Docker Application Container Engine.
Feb 07 18:56:13 DietPi systemd[1]: docker.service: Start request repeated too quickly.
Feb 07 18:56:13 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Feb 07 18:56:13 DietPi systemd[1]: Failed to start Docker Application Container Engine.
Feb 07 18:56:13 DietPi systemd[1]: docker.socket: Failed with result 'service-start-limit-hit'.

seniorgod, 2021-02-07:
root@DietPi:/mnt/dietpi_userdata# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3       
 29:    6276247    6267591    6608169    8794807       GIC  twd
 30:          0          0          0          0       GIC  owl_wdt
 32:    1581015          0          1          0       GIC  ethernet_mac
 33:          0          0          0          0       GIC  asoc_de
 43:          9          0          0          0       GIC  timer1_tick
 55:          0          0          0          0       GIC  xhci-hcd:usb1
 56:         56          0          0          0       GIC  aotg_hub_hcd:usb3
 57:    4813928          0          0          0       GIC  b0170000.i2c
 58:          0          0          0          0       GIC  b0174000.i2c
 59:          2          0          0          0       GIC  b0178000.i2c
 64:         65          0          0          0       GIC
 74:          1          0          0          0       GIC  sdcard
 76:      55961          0          0          0       GIC  emmc
 78:          0          0          0          0       GIC  hdmidev
 81:          0          0          0          0       GIC  vce_isr
 82:          0          0          0          0       GIC  vde_isr
 89:      20599          0          0          0       GIC  owl_dma0
 90:          0          0          0          0       GIC  owl_dma1
 91:          0          0          0          0       GIC  owl_dma2
 92:          0          0          0          0       GIC  owl_dma3
 93:   40053562          0          0          0       GIC  aotg_hub_hcd:usb4
215:          0          0          0          0  owl_sirq_irq  atc2603c
216:          0          0          0          0  atc2603c  atc260x_onoff
217:          0          0          0          0  atc2603c  RTC alarm
219:          0          0          0          0  atc2603c  atc260x-irkeypad
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:     196210     865331     498975     562729  Rescheduling interrupts
IPI3:        528        965        943       1058  Function call interrupts
IPI4:        480          8         11          5  Single function call interrupts
IPI5:          0          0          0          0  CPU stop interrupts
IPI6:          0          0          0          0  completion interrupts
IPI7:          0          0          0          0  CPU backtrace
Err:          0

root@DietPi:/mnt/dietpi_userdata# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temperature  |     45'C : 113'F (Optimal temperature)
 Governor     |     conservative
 Throttle up  |     80% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      504 MHz      240 MHz    1104 MHz
 CPU1         |      504 MHz      240 MHz    1104 MHz
 CPU2         |      504 MHz      240 MHz    1104 MHz
 CPU3         |      504 MHz      240 MHz    1104 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.

Joulinar, 2021-02-07:
It's posted into wrong issue atm

MichaIng, 2021-02-07:
I copied it over to the other issue. Okay so the Docker meta data indeed is corrupt.

Let's see if it's possible to repair it:
```
thin_repair /var/lib/docker/devicemapper/devicemapper/metadata
thin_check --clear-needs-check-flag /var/lib/docker/devicemapper/devicemapper/metadata
thin_check /var/lib/docker/devicemapper/devicemapper/metadata
```

Otherwise, since it's a fresh install, simply remove it, and see if the Docker then succeeds to start:
```
rm -Rf /var/lib/docker/{*,.??*}
systemctl restart docker
```
If it does, redo the whole install to have everything configured:
```
dietpi-software install 162
```

seniorgod, 2021-02-15:
I rolled bay to dietPi on stretch on the sparky device 

btw: on a raspberry pi i was able to install Docker without any problem

closed 2021-02-15 by seniorgod