MichaIng/DietPi #4093: HELP - How can i add the monted drive to external apps

GulyFMG, 2021-02-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 10.7
- Kernel version | Linux tartarus 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | Radarr and sonarr

#### Steps to reproduce
N/A

#### Expected behaviour
N/A

Hi Guys,
This is more a requested for help or a way to point me in the right direction.
I have monted my HDD to the mnt/xxxxxxxxxxxx folder via dietpi-drive_manager in sonarr you can see on the system tab the diskspace being used by the MMC but not by the HDD. I did try to ask on radarr forum on reddit, this is the [post](https://www.reddit.com/r/radarr/comments/kn20lu/how_to_add_a_mount_point_or_folder_to_disk_space) i made in the forums.
Its there any way to trick or edit maybe fstab so this would be populated in radarr/sonarr?
[image]
In theory since this is grabbing the folder /var/log/radarr, maybe we can do a workaround.

Thank you in advance.

Joulinar, 2021-02-07:
Hi,

looks like Radarr is going to grep specific stuf (probably hardcode) and not something from `/etc/fstab`. If I'm not mistaken your question was already answered on reddit

GulyFMG, 2021-02-07:
True, i was hoping for maybe a workaround... i did check the code and this line caught my attention:
var optionalRootFolders = GetFixedDisksRootPaths().Except(importantRootFolders).Distinct().ToList();

MichaIng, 2021-02-07:
It's a bit strange that they show space for /media and /mnt at all, since those are not meant to be mount points themselves, so will always be part of the root file system. In case of /var/log/radarr it makes sense to cover RAM log. Although, do you have DietPi-RAMlog enabled? If so it would even somehow get the wrong space.

What does Bakerboy448 mean with "Add it as root folder"? Mounting it to `/mnt` directly? No this is simply wrong to expect.
If it's really only this hardcoded list of directories listed there, then using /media as direct mount point could be a workaround, since at least `dietpi-drive_manager` does not use it. But /media is used by some tools as auto-mount location, i.e. mount on plug, just to note.

GulyFMG, 2021-02-07:
> It's a bit strange that they show space for /media and /mnt at all, since those are not meant to be mount points themselves, so will always be part of the root file system. In case of /var/log/radarr it makes sense to cover RAM log. Although, do you have DietPi-RAMlog enabled? If so it would even somehow get the wrong space.
> _I do have Dietpi-Ramlog option2 if im not mistaken._

> What does Bakerboy448 mean with "Add it as root folder"? Mounting it to `/mnt` directly? No this is simply wrong to expect.
> If it's really only this hardcoded list of directories listed there, then [...]
Im gonna try and move some folder and try this... for a couple days if nothing goes wrong, it could be a workaround, since this is a headless server im not expecting to connect and disconnect drives. but having this info on sonarr / radar would help me a bit to manage the server. Just to give a litttle context i use an app NZB360 to control all the media, before i would ssh the server and just DF -h but since version 14 of NZB introduced a dashboard this would be a great QOL for me.
Thank you for the input.

MichaIng, 2021-02-08:
> I do have Dietpi-Ramlog option2 if im not mistaken._

To be sure: `df /var/log/radarr`
If that's a `tmpfs`, not the root drive, then the check is currently done completely wrong 🤔.

GulyFMG, 2021-02-08:
> df /var/log/radarr
root@tartarus:~# df /var/log/radarr
Filesystem     1K-blocks  Used Available Use% Mounted on
tmpfs              51200   292     50908   1% /var/log

Looks like you are on point.

MichaIng, 2021-02-08:
Not sure how the check is done, but probably it works at least if the directory is the mount point itself, so i.e. if `/var/log/radarr` itself would be an own tmpfs (or other), so that mounting a drive directly to /media does work.

Since via shell/script it is trivial to get the actual file system where a sub directory belongs to, it looks like we found a bug to report, aside of the design choice to show this hard-coded list only.

closed 2021-02-09 by GulyFMG