MichaIng/DietPi #4097: erro dietpi-upgrade

carlosspre, 2021-02-08:
- DietPi has encounted an error, and, is unable to continue         -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket              -
---------------------------------------------------------------------

#### Details:
- Date           | Mon  8 Feb 22:50:36 GMT 2021
- Bug report     | d7a220ef-87b4-40aa-85fb-dba9d86db73f
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux NetPlexPi4 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```

ravenclaw900, 2021-02-08:
Hello,
Did you run `dpkg --configure -a`, like it suggested?

carlosspre, 2021-02-08:
yes

root@NetPlexPi4:~# dpkg --configure -a
Setting up mariadb-server-10.3 (1:10.3.27-0+deb10u1) ...
/usr/bin/deb-systemd-helper: line 1: shape:EventCategoriesMapList,: command not found
/usr/bin/deb-systemd-helper: line 2: documentation:<p>A list of EventCategoriesMap data types.</p>: No such file or directory
/usr/bin/deb-systemd-helper: line 3: syntax error near unexpected token `}'
/usr/bin/deb-systemd-helper: line 3: `        }'
dpkg: error processing package mariadb-server-10.3 (--configure):
 installed mariadb-server-10.3 package post-installation script subprocess returned error exit status 2
Setting up rpi-eeprom (11.6-1) ...
/usr/bin/deb-systemd-helper: 1: /usr/bin/deb-systemd-helper: shape:EventCategoriesMapList,: not found
/usr/bin/deb-systemd-helper: 2: /usr/bin/deb-systemd-helper: documentation:<p>A list of EventCategoriesMap data types.</p>: not found
/usr/bin/deb-systemd-helper: 3: /usr/bin/deb-systemd-helper: Syntax error: "}" unexpected
/usr/bin/deb-systemd-helper: 1: /usr/bin/deb-systemd-helper: shape:EventCategoriesMapList,: not found
/usr/bin/deb-systemd-helper: 2: /usr/bin/deb-systemd-helper: documentation:<p>A list of EventCategoriesMap data types.</p>: not found
/usr/bin/deb-systemd-helper: 3: /usr/bin/deb-systemd-helper: Syntax error: "}" unexpected
/usr/bin/deb-systemd-helper: 1: /usr/bin/deb-systemd-helper: shape:EventCategoriesMapList,: not found
/usr/bin/deb-systemd-helper: 2: /usr/bin/deb-systemd-helper: documentation:<p>A list of EventCategoriesMap data types.</p>: not found
/usr/bin/deb-systemd-helper: 3: /usr/bin/deb-systemd-helper: Syntax error: "}" unexpected
Setting up rpi-eeprom-images (11.6-1) ...
dpkg: dependency problems prevent configuration of mariadb-server:
 mariadb-server depends on mariadb-server-10.3 (>= 1:10.3.27-0+deb10u1); however:
  Package mariadb-server-10.3 is not configured yet.

dpkg: error processing package mariadb-server (--configure):
 dependency problems - leaving unconfigured
Processing triggers for mime-support (3.62) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (241-7~deb10u5+rpi1) ...
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Errors were encountered while processing:
 mariadb-server-10.3
 mariadb-server
root@NetPlexPi4:~#

carlosspre, 2021-02-08:
┤ DietPi-Update | Error handler ├──────────────────────────
Command: G_AGUG
Exit code: 100
DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:4 | HW_ARCH:2 | DISTRO:5
Image creator: DietPi Core Team
Pre-image: Raspbian Lite

 file content:
r/bin/deb-systemd-helper: 2: /usr/bin/deb-systemd-helper: documentation:<p>A lis
EventCategoriesMap data types.</p>: not found^M
r/bin/deb-systemd-helper: 3: /usr/bin/deb-systemd-helper: Syntax error: "}"
xpected^M
r/bin/deb-systemd-helper: 1: /usr/bin/deb-systemd-helper:
pe:EventCategoriesMapList,: not found^M
r/bin/deb-systemd-helper: 2: /usr/bin/deb-systemd-helper: documentation:<p>A lis

       Retry         : Re-run the last command that failed
       DietPi-Config : Edit network, APT/NTP mirror settings etc
       Send report   : Uploads bugreport containing system info to DietPi

Joulinar, 2021-02-08:
looks like MariaDB package is brocken. Can you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt install --reinstall mariadb-server-10.3
```

carlosspre, 2021-02-10:
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt install --reinstall mariadb-server-10.3

did not solve the problem

carlosspre, 2021-02-10:
[image]

Joulinar, 2021-02-10:
what was the output of `apt install --reinstall mariadb-server-10.3`

carlosspre, 2021-02-10:
root@NetPlexPi4:~# apt install --reinstall mariadb-server-10.3
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for mariadb-server-10.3:armhf
root@NetPlexPi4:~#

temtem more gave the error above

carlosspre, 2021-02-10:
[image]

MichaIng, 2021-02-10:
The package lists seem to be not updated successful. Could you run again those three commands and paste the output of all three of them here:
```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt install --reinstall mariadb-server-10.3
```
Btw, you can stick with copy&pasting the raw text from the SSH client (?) here. Wrap after pasting it, select it and hit the "Insert code" button `< >`, so it will be shown in monotype with spaces preserved, just like in the console.

If the last command still fails with this "Internal Error, No file name" error: https://askubuntu.com/a/510830
Please assure first that `/var/cache/apt/archives/mariadb-server-10.3_<something>.deb` exists:
```
ls -l /var/cache/apt/archives/mariadb-server-10.3*.deb
```
If so:
```
dpkg -i /var/cache/apt/archives/*.deb
dpkg -r --force-depends mariadb-server-10.3:armhf
dpkg --configure -a
dpkg -i /var/cache/apt/archives/*.deb
apt upgrade
dpkg -i /var/cache/apt/archives/*.deb
```

It looks strange, but APT + DPkg are great in locking up themselves in dependency issues, unable to resolve without some nasty manual back and forth.

If the last command of these fails again with `installed mariadb-server-10.3 package post-installation script subprocess returned error exit status 2`, please run and paste the following (which might produce a lot of lines):
```
bash -x /var/lib/dpkg/info/mariadb-server-10.3.postinst configure
```

MichaIng, 2021-03-15:
Is this still an issue or can it be closed?

MichaIng, 2021-03-24:
Marking as closed. Feel free to reopen when required.

closed 2021-03-24 by MichaIng