MichaIng/DietPi #4098: Bitwarden app makes error on login in Android and Ios!

cocoflan, 2021-02-09:
**On Android i get this error:**

An error has occured.
Exception message:java.security.cert.CertPathValidatorException: Trustanchor for certification path not found.

**On IOS i get this error:**

An error has occured.
Exception message:The certificate for this server is invalid. You might be connecting to a server that is pretending to be "IP" wich could put your confidentail information at risk.

So used the credential like on my pc and setup with the same ip and port.

Also it takes very long to install Bitwarden, and in the past sometimes an error on installing not this time lukely en the pc web interface works, only the phone apps have a problem.

Greeting Cocoflan

Joulinar, 2021-02-09:
Hi,

did you install the certificates on your mobile devices?

cocoflan, 2021-02-09:
I did try on my IOS device stayed the same, or  i did something wrong, any info on how to do that the good way?

Joulinar, 2021-02-09:
iOS is not accepting the certificate at the moment. We will fix this in 6.35. I will link the discussion we had  https://github.com/MichaIng/DietPi/issues/3724#issuecomment-750883187 😄 

change log: https://github.com/MichaIng/DietPi/blob/adee418c7b15f046e9b58a571202d4fd090902dd/CHANGELOG.txt#L19

cocoflan, 2021-02-09:
Super!!! Yes just tested it again and when i add the cert. "Not verified" message, good to know it wil be fixed in next version.

Greetings keep up the good work!

Joulinar, 2021-02-09:
if you agree I will mark it as duplicate and close the issue?

cocoflan, 2021-02-09:
Tested on Android 9 again still the same error afther adding the cert. Maby add it to the old issue also.

Greetings.

MichaIng, 2021-02-10:
Probably Android is affected the same way as macOS is, requiring the CA:TRUE flag in the certificate.

Here is the line in the code/certificate generation command that needs to be altered: https://github.com/MichaIng/DietPi/commit/8a5d0931fcfb764511f941f3e8782573240ed991#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5R11867

Joulinar, 2021-02-13:
@MichaIng 
should we close this one? Or keep it open until 6.35 has been released?

cocoflan, 2021-02-14:
If the next release is soon i think it's ok to close.

cocoflan, 2021-03-01:
Still the same problem on latest update, DietPi v7.0.0 <3.
Now the certificate is ok when importing it on ios, but get same errors on IOS and Android on login.

Greetings

MichaIng, 2021-03-01:
But you did add the (new) certificate to the trusted OS stores, didn't you?

We have instructions for this for macOS and Windows. I can write them for Android 8 later, but cannot help with iOS 😢: https://dietpi.com/docs/software/cloud/#bitwarden_rs

Ah, another thing, if you installed Bitwarden before the update: We did not force a reinstall or certificate re-creation with the update, since this would required everyone to re-import the cert on all their clients. Even a reinstall would not re-create it as long as the config is present already. To manually re-create the cert with the new method:
```
rm -f /mnt/dietpi_userdata/bitwarden_rs/cert.pem /mnt/dietpi_userdata/bitwarden_rs/privkey.pem
openssl req -reqexts SAN -subj '/CN=DietPi Bitwarden_RS' -config <(cat /etc/ssl/openssl.cnf <(echo -ne "[SAN]\nsubjectAltName=DNS:$(</etc/hostname),IP:$(mawk 'NR==4' /run/dietpi/.network)\nbasicConstraints=CA:TRUE,pathlen:0")) \
-x509 -days 7200 -sha256 -extensions SAN -out /mnt/dietpi_userdata/bitwarden_rs/cert.pem \
-newkey rsa:4096 -nodes -keyout /mnt/dietpi_userdata/bitwarden_rs/privkey.pem
```
The "basicConstraints=CA:TRUE,pathlen:0" options are what was missing before and required on iOS, basically a flag that this is meant to be a CA (certificate authority) cert.

Joulinar, 2021-03-01:
I still have an old iPhone SE flying around that I use for testing. Certificate implementation on iOS is explained on that blog very well. https://www.theictguy.co.uk/adding-trusted-root-certificates-to-ios14/

cocoflan, 2021-03-01:
"Joulinar" thx for your info, i did forget to activate the cert. in the about settings for certificates.

Now it is working perfectly on ios, now i have to test it on android.

Thx a lot.

Joulinar, 2021-03-01:
Before release I tested it on Android 10. There it was working as well.

cocoflan, 2021-03-01:
Yes works on android 9, so all is ok! :-)

Joulinar, 2021-03-01:
ok I will close the issue for now. Feel free to reopen if needed

closed 2021-03-01 by Joulinar