MichaIng/DietPi #4099: Radarr fail to update

Dr0bac, 2021-02-10:
rpi 3b 1gb of ram.

There was a radarr update few days ago, and when i tried to update it from radarr web view, i coudn't update it, radarr downloaded updates files and restarted it self, but after restart program version was the same. So i send them logs, and problem was in /tmp directory.

So i had around 300-350 mb of used ram memory when radarr started to update itself, and radarr download update files in /tmp directory

in logs there was a line that says: there is not enough space. in tmp/ directory i can see that radarr update files are quite large to put them in ram

````
root@DietPi:/tmp# du -hs /tmp/*
0       /tmp/clr-debug-pipe-1847-330717-in
0       /tmp/clr-debug-pipe-1847-330717-out
0       /tmp/clr-debug-pipe-5745-3864108-in
0       /tmp/clr-debug-pipe-5745-3864108-out
0       /tmp/clr-debug-pipe-6189-3958719-in
0       /tmp/clr-debug-pipe-6189-3958719-out
0       /tmp/clr-debug-pipe-665-1670-in
0       /tmp/clr-debug-pipe-665-1670-out
0       /tmp/dotnet-diagnostic-1847-330717-socket
0       /tmp/dotnet-diagnostic-5745-3864108-socket
0       /tmp/dotnet-diagnostic-6189-3958719-socket
0       /tmp/dotnet-diagnostic-665-1670-socket
0       /tmp/pms-0216841b-fdaf-4f9a-b104-d66e8c6e8cdf
1.2M    /tmp/radarr_backup
490M    /tmp/radarr_update
````

then i enabled swap file and radarr updated successfully..

Does /tmp directory need to be in ram memory, how often files are written to that directory?

MichaIng, 2021-02-10:
> Does /tmp directory need to be in ram memory

It doesn't need to be, but I highly recommend to leave it in RAM, which is the default in all modern distro. `/tmp` is meant to be for temporary volatile, often smaller files. If very larger temporary files are created, e.g. video transcoding and such, most software use a separate directory on disc for this, since /tmp is mostly in tmpfs with limited space. 500 MiB is the border of what can usually be stored in /tmp without risking that some systems run into full RAM or more likely full tmpfs file system, since the default tmpfs size is 512 MiB. On DietPi we assign 50% of physical memory + swap, so in your case with 1 GiB physical memory and without swap file it was probably slightly smaller, due to GPU memory split on RPi e.g.

In your case, a swap file looks like the best solution indeed. By default it's used only very rarely, due to `vm.swappiness=1`, to prevent exactly such OOM errors. It's much better than moving /tmp out of RAM, since now only when physical RAM is full, parts of that Radarr update files are stored on disk while everything that fits is left in RAM. If you remove the tmpfs, you'll have everything written to disk directly, including all other temporary files, reducing performance and increasing disk wear.

Dr0bac, 2021-02-10:
Thank you for advice, will leave swap file on then.

closed 2021-02-10 by Dr0bac